SpenceKonde/DxCore #290: Float addition/printing incorrect in loop

nabelekt, 2022-06-27:
I have this code on a AVR128DA:
```
void setup()
{
    Serial.begin(19200);

}

float val = 134217000;

void loop()
{
    Serial.printf("%f\n", val);

    val = val + 10;

    delay(100);
}
```
Which results in:
```
16:05:03.025 -> 134217000.000000
16:05:03.157 -> 134217010.000000
16:05:03.251 -> 134217020.000000
16:05:03.345 -> 134217020.000000
16:05:03.438 -> 134217030.000000
16:05:03.532 -> 134217040.000000
16:05:03.664 -> 134217050.000000
16:05:03.758 -> 134217060.000000
16:05:03.852 -> 134217060.000000
16:05:03.945 -> 134217070.000000
16:05:04.039 -> 134217080.000000
16:05:04.124 -> 134217090.000000
16:05:04.265 -> 134217100.000000
16:05:04.359 -> 134217100.000000
16:05:04.454 -> 134217110.000000
16:05:04.548 -> 134217120.000000
16:05:04.641 -> 134217130.000000
16:05:04.735 -> 134217140.000000
16:05:04.828 -> 134217140.000000
16:05:04.969 -> 134217150.000000
16:05:05.063 -> 134217160.000000
...
```
As you can see, some values are repeated.

If I change line 13 to `    val = val + 1;` I instead get:
```
16:06:57.677 -> 134217000.000000
16:06:57.771 -> 134217000.000000
16:06:57.865 -> 134217000.000000
16:06:57.949 -> 134217000.000000
16:06:58.089 -> 134217000.000000
16:06:58.183 -> 134217000.000000
16:06:58.277 -> 134217000.000000
16:06:58.371 -> 134217000.000000
16:06:58.455 -> 134217000.000000
16:06:58.596 -> 134217000.000000
16:06:58.690 -> 134217000.000000
16:06:58.784 -> 134217000.000000
16:06:58.878 -> 134217000.000000
16:06:58.962 -> 134217000.000000
...
```
where the value doesn't change at all.

If I then change line 13 back to `    val = val + 10;` and line 15 to `    delay(0);`, I get:
```
16:09:15.665 -> 134217010.000000
16:09:15.665 -> 134217020.000000
16:09:15.665 -> 134217020.000000
16:09:15.665 -> 134217030.000000
16:09:15.665 -> 134217040.000000
16:09:15.712 -> 134217050.000000
16:09:15.712 -> 134217060.000000
16:09:15.712 -> 134217060.000000
16:09:15.712 -> 134217070.000000
16:09:15.712 -> 134217080.000000
16:09:15.760 -> 134217090.000000
16:09:15.760 -> 134217100.000000
16:09:15.760 -> 134217100.000000
...
16:09:16.422 -> 134217720.000000
16:09:16.422 -> 134217730.000000
16:09:16.469 -> 134217740.000000
16:09:16.469 -> 134217760.000000
16:09:16.469 -> 134217780.000000
16:09:16.469 -> 134217790.000000
16:09:16.469 -> 134217810.000000
16:09:16.517 -> 134217820.000000
16:09:16.517 -> 134217840.000000
16:09:16.517 -> 134217860.000000
16:09:16.517 -> 134217870.000000
16:09:16.517 -> 134217890.000000
16:09:16.564 -> 134217900.000000
16:09:16.564 -> 134217920.000000
16:09:16.564 -> 134217940.000000
16:09:16.564 -> 134217950.000000
16:09:16.564 -> 134217970.000000
16:09:16.612 -> 134217980.000000
16:09:16.612 -> 134218000.000000
16:09:16.612 -> 134218020.000000
16:09:16.612 -> 134218030.000000
...
```
where some values are printed twice and some are skipped entirely.

In all of these cases, I don't think it matters if `val` is defined as a `float` or `double`.

Any ideas on any of this? Are these more likely issues with Microchip's printf() implementation or something else? This is with printf option set to "Full".

grandaspanna, 2022-06-27:
Read the next sentence: "That means the total number of digits, not the number to the right of the decimal point"

nabelekt, 2022-06-27:
Got it! Thanks for correcting me.

So then, I guess my qustion is: even though `double`s aren't supported on Arduino, could they be supported on AVR modules like AVR128DA by DxCore? I am not sure what inherits from what, and at what level some of these things are defined.

grandaspanna, 2022-06-27:
This may be worth exploring: http://www.gammon.com.au/forum/?id=11519

Part of me is wondering why you need such precision. In most real-world scenarios, you're already well below the noise of sensors and other devices.

nabelekt, 2022-06-28:
Thanks.
At this point it's mostly academic/for future reference. I noticed the issue when experimenting with some fairly precise radio frequencies.

grandaspanna, 2022-06-28:
Sounds fun. C has never had a great reputation for high precision computational stuff. People often resort to rolling their own based on integer computation and converting as appropriate. Combined with an 8-bit core, it's not the best option out there :-)

The bigNum library above seems a reasonable option.

grandaspanna, 2022-06-28:
It might make sense to close this issue, since it's not really a core issue.

nabelekt, 2022-06-28:
I'd like to leave it open for a little while longer to see if anyone else here is some input about supporting `double`s.

SpenceKonde, 2022-07-01:
Nope, there isn't some way that we can just pass a different option to the compiler or something to get support for 

Very few tasks require a high precision floating point value, and those few tasks are not a food fit for AVR. These parts don't have a built-in insutructin for even plain old integer division (and he multiplication routines are little awkward) 

Floats, doubles, and "long doubles"  are the same because there is no separate double support provided by avr-gcc toolchain until you get all the way to the avr-gcc 10. (but "upgrading" to newer versions of avr-gcc while all else stays equal (same size data types, and all), because architecture specific things have not kept pace with the compiler at large, so more recently added "optimizations" are performed during early passes, making assumptions that are invalid for AVR, and so the final avr-specific passes generate inefficient code. Versions after 7.3 are said to get worse on AVR (which makes sense if as I believe to be the case, that's what Microchip uses, so they have not tuned later versions to make them produce efficient code). 

(the compiler makes many passes over the code trying optimizations, but each pass is unaware of what the other passes will do, many working with abstract ideas like "store this to SRAM based on one of the pointer register pairs and use displacement, but represents the three pointer registers equivalently even though only 2 of the three can perform load/store with displacement, until it reaches one of the lowest level passes witch converts these virtual instructions into specific real ones. This is when it needs to face the fact that the X Y and Z registers aren't created equal. The mighty Z register is the most capable is the most powerful as it guiles LPM, icall and (on Dx-series devices, which streamline the writing of application data to flash, self programming) as well as being capable of displacement.  The Y pointer, which is frequently pointing to the stack registers, is capable of displacement too (this is used extensively to access data on the stack by the compiler - the stack pointer has a is almost like a 4th pointer register, except that you only get st -PC, and ld PC+ - but practical application may need to look at the data in the  stack. The Y pointer register is usually used for this since it isn't call used., though it is so useful that it;s often used for other purposes too. The X pointer on the other hand is the weakest of the pointer registers, capable of only acting on it's current address, with post increment and predecrement as alway; There's no ldd X+q (there aren't enough opcodes available for them to have that - LDD and STD takes 1/16th of 16-bit opcodes each for the the Y and Z registers variants.). But at some point in the compile process, it treats the three registers as all being callable of displacement. A later lower level pass then converts these virtual instructions into real ones. Impossible ones like ldd X+q , which then gets transformed adiw q, ld x, sbiwq. Nothing that comes after that pass in the compile process notices that this often produces situation where a 2 clock sbiw is used to restore the original value to a pointer immediately before an adiiw is used to add to it to load a different offset. This is an example of poor tuning of gcc to avr (and they are said to get more numerous in later version,. Recall that classic AVR started to become a backwater after well ARM'ed competitors beat the stuffing out of XMega in the market place (It seems to me that they had underestimated the persistence of 5v logic, as well als the "critical mass" effect that made ARM the de-facto standard for anything from more-computationally-capable microcontrollers  than the 8-bit classic AVRs, all the way to high end tablet computers and now even laptops.. While Atmel had (wisely) gotten into the ARM microcontroller game, they seemed to not recognize that extensions of AVR into that space were never gong to work! The the xMega sacrified the 5v operature range, and probably even more importantly, was byzantine and nearly incomprehensible, full of special cases, exceptions to patterns and so on, with no real, well, pattern to it, whikle pries remained stubbornkly high. That all changed when Microchip,(preusmably accepting that AVR it was better than PIC and far easier to write, , commenced reinvigorating teh AVR line. taking the best pieces of xMega and classic AVRs  to yield the current generation  of parts. 

In any event, floating point math (and indeed, anything) with 64-bit operands would be agonizingly slow and bloated on AVR. AVR only has 32 working registers, and the ABI would require that both operrands be passed on the stack (16 pushes, and then 16 pops at the end, So in addition to the theoretical greater complexity,  there is a ton of extra overhead.

closed 2022-07-01 by SpenceKonde