SpenceKonde/DxCore #295: DD support - broken in 1.4.10 and other problems in head of tree

ObviousInRetrospect, 2022-07-02:
1.4.10, build of the default sketch with empty setup and loop (avr64dd32):

```
In file included from /Users/username/Library/Arduino15/packages/DxCore/hardware/megaavr/1.4.10/cores/dxcore/Arduino.h:27:0,
                 from /private/var/folders/jp/8cvzydc904n1td9j3y4ntcgm0000gn/T/arduino-sketch-B22E3CF43EDC1B13AEE9CC8CBE0E1CBF/sketch/sketch_jul2a.ino.cpp:1:
/Users/username/Library/Arduino15/packages/DxCore/hardware/megaavr/1.4.10/cores/dxcore/core_devices.h:426:4: error: #error "Unrecognized combination of flash size and chip type"
   #error "Unrecognized combination of flash size and chip type"
    ^~~~~
In file included from /Users/username/Library/Arduino15/packages/DxCore/hardware/megaavr/1.4.10/cores/dxcore/Arduino.h:571:0,
                 from /private/var/folders/jp/8cvzydc904n1td9j3y4ntcgm0000gn/T/arduino-sketch-B22E3CF43EDC1B13AEE9CC8CBE0E1CBF/sketch/sketch_jul2a.ino.cpp:1:
/Users/username/Library/Arduino15/packages/DxCore/hardware/megaavr/1.4.10/cores/dxcore/UART.h:31:10: fatal error: pins_arduino.h: No such file or directory
 #include "pins_arduino.h"
          ^~~~~~~~~~~~~~~~
compilation terminated.

exit status 1

Compilation error: exit status 1
```

Copying head of tree into /Users/username/Arduino/hardware/DxCore in 2.0.0-rc6 spun forever on startup.

Upgrading to 2.0.0-rc8 fixed the spin, but with head of tree in path boards manager would list no boards as installed. Same result after removing head of tree, removing 1.4.10, and then re-adding head of tree.

Wondering if there is a workaround and how close we are to the next version. Not entirely sure how to debug this, I see what looks like a logs directory in ~/.arduinoIDE/logs which contains a directory per-launch timestamp but no files.

SpenceKonde, 2022-07-02:
Yeah I know I was working on it - it needed a lot more work than I thought)
but got a urgent commission from a client. I was part way through with a
bunch of committed code that broke stuff, working on fixes, then I got the
email with those magic words "additional fee" for giving his project
priority. Since he pays considerably better than this (which pays
$0.00/month for), I need to get that design done before I get back to the
core.

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2022-07-04:
And the DD's will require version Azduino5 of the toolchain, which I haven't packaged yet. I think I have a clean build now though.

ObviousInRetrospect, 2022-07-22:
No pressure but is the head of tree worth trying with your latest commits or is it still known broken?

With Melody, Microchip accomplished the amazing feat of making Atmel Start look good (Atmel Start didn't get DD support, it is what I used in the pre-megatinycore dark ages). Trying to figure out whether to keep fighting with Melody or if 1.5.0 is likely to hit before I get Melody working.

SpenceKonde, 2022-07-23:
I'm going to try to get the DxCore stuff working assuming i'm now getting clean results from megaTinyCore. 

1.5.0 versions are going to require a bunch of syncing of critical fixes from megaTinyCore though (I wish they'd submitted the PRs to the repo I wanted to work on - I have a test board built now for every size of the DD-series but DxCore doesn't compile at all! Three libraries need to be synced, 2 more need to get teh same set of fixes applied to them and changes to DxCore's serial classes to match megaTinyCore and rip out another part of the disastrous ArduinoAPI that was wasting flash with virtual functions because they wrote arduinoapi without considering the resource limitations of small AVRs.

SpenceKonde, 2022-08-13:
Nothing is working at the head of either core. for a status update.

wa2mze, 2022-09-23:
Just noticed there is no bootloader support option for DD parts (there is for DA and DB).  Any reason why not, other than you haven't got a round tuit yet?

mcuee, 2022-10-28:
Indeed as of now, simple compilation of the Blink sketech does not work and there is no optiboot_dx bootloader support yet.

```
Using board 'avrdd' from platform in folder: C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10
Using core 'dxcore' from platform in folder: C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10
Detecting libraries used...
"C:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=avr64dd28 -DF_CPU=24000000L -DCLOCK_SOURCE=0 -DCORE_ATTACH_ALL -DTWI_MORS_SINGLE -DMILLIS_USE_TIMER -DARDUINO=10607 -DARDUINO_avrdd -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.4.10\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=4UL -DDXCORE_PATCH=10UL -DDXCORE_RELEASED=1 -DMVIO_ENABLED "-IC:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.4.10\\cores\\dxcore/api/deprecated" "-IC:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.4.10\\cores\\dxcore" "-IC:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.4.10\\variants\\28pin-ddseries" "C:\\Users\\xiaof\\AppData\\Local\\Temp\\arduino-sketch-07DCBE58C31D28F5C44D1058B336EB98\\sketch\\Blink.ino.cpp" -o nul
Error while detecting libraries included by C:\Users\xiaof\AppData\Local\Temp\arduino-sketch-07DCBE58C31D28F5C44D1058B336EB98\sketch\Blink.ino.cpp
Generating function prototypes...
"C:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\DxCore\\tools\\avr-gcc\\7.3.0-atmel3.6.1-azduino4b/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=avr64dd28 -DF_CPU=24000000L -DCLOCK_SOURCE=0 -DCORE_ATTACH_ALL -DTWI_MORS_SINGLE -DMILLIS_USE_TIMER -DARDUINO=10607 -DARDUINO_avrdd -DARDUINO_ARCH_MEGAAVR "-DDXCORE=\"1.4.10\"" -DDXCORE_MAJOR=1UL -DDXCORE_MINOR=4UL -DDXCORE_PATCH=10UL -DDXCORE_RELEASED=1 -DMVIO_ENABLED "-IC:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.4.10\\cores\\dxcore/api/deprecated" "-IC:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.4.10\\cores\\dxcore" "-IC:\\Users\\xiaof\\AppData\\Local\\Arduino15\\packages\\DxCore\\hardware\\megaavr\\1.4.10\\variants\\28pin-ddseries" "C:\\Users\\xiaof\\AppData\\Local\\Temp\\arduino-sketch-07DCBE58C31D28F5C44D1058B336EB98\\sketch\\Blink.ino.cpp" -o "C:\\Users\\xiaof\\AppData\\Local\\Temp\\arduino-sketch-07DCBE58C31D28F5C44D1058B336EB98\\preproc\\ctags_target_for_gcc_minus_e.cpp"
In file included from C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\cores\dxcore/Arduino.h:27:0,
                 from C:\Users\xiaof\AppData\Local\Temp\arduino-sketch-07DCBE58C31D28F5C44D1058B336EB98\sketch\Blink.ino.cpp:1:
C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\cores\dxcore/core_devices.h:426:4: error: #error "Unrecognized combination of flash size and chip type"
   #error "Unrecognized combination of flash size and chip type"
    ^~~~~
In file included from C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\variants\28pin-ddseries/pins_arduino.h:24:0,
                 from C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\cores\dxcore/UART.h:31,
                 from C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\cores\dxcore/Arduino.h:571,
                 from C:\Users\xiaof\AppData\Local\Temp\arduino-sketch-07DCBE58C31D28F5C44D1058B336EB98\sketch\Blink.ino.cpp:1:
C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\cores\dxcore/timers.h:106:4: error: #error "Millis timer not specified, nor is millis disabled - can't happen!"
   #error "Millis timer not specified, nor is millis disabled - can't happen!"
    ^~~~~

exit status 1

Compilation error: exit status 1

```

SpenceKonde, 2022-11-09:
Optiboot issue is in another issue. 

All the double-D's compile now.

closed 2022-11-09 by SpenceKonde