SpenceKonde/DxCore #310: Unable to get code running AVR64DB64

tomdowad, 2022-08-03:
I made a custom AVR128DB32 board, and I can program it with Arduino and DxCore, using an Atmel ICE. 
I also made a custom AVR64DB64 board. I can program it with Microchip Studio and an Atmel ICE. 
But when I try to program the 2nd board with Arduino/DxCore, I am getting no output. The output test uses PF_1 on an LED (pin 45) and PF_4/PF_5 (pins 48/49) as the Serial2 UART. Its the same code as the DB32 but different pin numbers.

The sketch is very simple so I am guessing the problem is due to the particulars of this MCU in DxCore. Code below.
Thanks for creating DxCore!

//#define MCU_PINS 32
#define MCU_PINS 64

#if MCU_PINS == 32
#define PIN_LED 23 // PORTF
#define PIN_TX  24  
#define PIN_RX  25
#endif

#if MCU_PINS == 64
#define PIN_LED 45 // PORTF
#define PIN_TX  48  
#define PIN_RX  49
#endif

void setup()
{
   pinMode(PIN_LED,OUTPUT);
   Serial2.pins(PIN_TX,PIN_RX);
   Serial2.begin(57600);
}

void loop() {
digitalWrite(PIN_LED,0);
delay(500);
digitalWrite(PIN_LED,1);
delay(500);

Serial2.write('!');
if(Serial2.available())
    Serial2.write(Serial2.read());
 
}

gordonthree, 2022-08-05:
Can you share what error you are getting? I copy pasted your code into a blank sketch, selecting AVR64DB64 and it compiled clean with no errorss.

```
Sketch uses 2786 bytes (4%) of program storage space. Maximum is 65536 bytes.
Global variables use 347 bytes (4%) of dynamic memory, leaving 7845 bytes for local variables. Maximum is 8192 bytes.
```

tomdowad, 2022-08-05:
It compiles. But doesn’t do anything. Without a debugger I cannot see why. However since the 32 pin works, and the 64 pin doesn’t, and since they both use PORTF, I reckoned it is probably something about the driver for the 64 pin chip. 

I am well versed in programming this MCU in C, but I don’t have any knowledge as to how Arduino manages boards. I couldn’t find the board code on my Mac. But I would like to learn. Perhaps I can help debug it

gordonthree, 2022-08-05:
Finding where Arduino hides code is frustrating. I like to pull up the code in Github instead, much easier to navigate and search. 

There's some stuff going on with Serial towards the bottom of [this](https://github.com/SpenceKonde/DxCore/blob/master/megaavr/cores/dxcore/Arduino.h) file. In the same folder as that header are all the UART* files, might be some clues there about what you're seeing.

SpenceKonde, 2022-08-05:
Okay...

1. refer to pins with PIN_xn notation (this doesn't mean no PIN_LED. just not pin 40 or something 

2. I don't know where you'tr getting the pincount from - there should be an _AVR_PINCOUNT define

3. The released version has a bug in serial receiving that I just found

tomdowad, 2022-08-06:
OK I see. I'm sorry. I needed to RTFM. When the 32 pin worked, I assumed
that using the physical pin numbers was the way to go. I changed my pin
definitions to PIN_PF1, PIN_PF4, PIN_PF5 and it of course worked.

Thanks. Your response time is better than Microchip's!

SpenceKonde, 2022-08-07:
>I Changed my pin definitions to PIN_PF1, PIN_PF4, PIN_PF5 and it of course worked. Thanks. Your response time is better than Microchip's!

Sounds like the issue was xactly the kind of pin number confusion that the PIN_xn notation is meant to eliminate. 

Faster than microchip support? Wow, I'm honored, such a high bar. Plus my answer sounds like it was useful

closed 2022-08-21 by SpenceKonde

tomdowad, 2022-10-11:
Hey Spence

I will send you one of these boards if you like. First version wasn’t perfect, but I can make you one or you can wait for the revision. Just the AVR and a lot of JST PH connectors. A regulator and an EEPROM.

I also have a UC1609 display I’m working on but so far no luck with the Arduino library. It connects to the MCU board with a JST PH 10 pin.

The compiler complained about the swap method , so I renamed it. I set the pins for SPI…PORTE. But no go so far. I’m trying to figure it out this time. 

Regards
Tom