SpenceKonde/DxCore #321: BUG FOUND

mikrohendese, 2022-08-10:
Hello,
I am already using AVR128DB28 Microcontroller in a project. I found a bug in file "pins_arduino.h" in directory \DxCore\hardware\megaavr\1.4.10\variants\28pin-standard\
I am using SPI1 interface for communicating to W5500 Ethernet Interface IC. 
Although swapping the SPI pins with   "SPI.swap(SPI1_SWAP_DEFAULT) ;" , it doesn't work.
After my observations in library I  found a bug:  Pin definitions are incorrect so the SPI was not working. After corection, I have been able to run the application. 
The correction is as follows ;
 
// SPI 1
#define SPI1_MUX                PORTMUX_SPI1_DEFAULT_gc

//#define PIN_SPI1_MISO           PIN_PC0  // Remark 
#define PIN_SPI1_MISO           PIN_PC1    // Corrected.

//#define PIN_SPI1_SCK            PIN_PC1  // Remark
#define PIN_SPI1_SCK            PIN_PC2    // Corrected

//#define PIN_SPI1_MOSI           PIN_PC2    // Remark
#define PIN_SPI1_MOSI           PIN_PC0       // Corrected

This project is a good work. Thanks to SpenceKonde for this work. 

Mehmet HALICI
Mikrohendese Elektronik

ObviousInRetrospect, 2022-08-10:
Same issue in the 32-pin variant.

The 48 and 64 have the signal names in the same order as the data sheet and are correct.

I have a fix here: https://github.com/SpenceKonde/DxCore/pull/323

SpenceKonde, 2022-08-13:
fix merged thanks.

closed 2022-08-13 by SpenceKonde