SpenceKonde/DxCore #322: i2c slave corruption in sleep modes other than SLEEP_MODE_IDLE

ObviousInRetrospect, 2022-08-10:
I've noticed unreliable slave behavior in SLEEP_MODE_STANDBY (and SLEEP_MODE_POWER_DOWN). This seems to be more of an issue when transferring larger buffers. I thought guarding the sleep_cpu() call with !Wire.slaveTransactionOpen() was sufficient. The twi supports wake from STANDBY and even POWER_DOWN but every i2c slave I have written with DxCore and megaTinyCore has been flakey unless run in IDLE.

The code here:
https://github.com/ObviousInRetrospect/DualMode/tree/main/DualModeExample

works as expected

but if SLEEP_MODE_IDLE is changed to SLEEP_MODE_STANDBY the slave starts corrupting data: https://github.com/ObviousInRetrospect/DualMode/blob/10dfca6b5bc6c8fedf1605d2d8c5981e2e4aca34/DualModeExample/DualModeExample.ino#L319 

this is most easily seen (especially without an ina3221) by uncommenting the test pattern data fill (L184-188):
https://github.com/ObviousInRetrospect/DualMode/blob/10dfca6b5bc6c8fedf1605d2d8c5981e2e4aca34/DualModeExample/DualModeExample.ino#L184

When I raised this in https://github.com/SpenceKonde/DxCore/discussions/316 @MX682X suggested opening a separate issue with more details.

The output on the master https://github.com/ObviousInRetrospect/DualMode/tree/main/DualModeExampleClient in SLEEP_MODE_IDLE is a long string of correct reads:

(the master does a 32-byte read follow by a 25-byte read. the last 4 bytes are a crc32 covering the entire data)

```text
0000000000000000000000000C000000
000033B9A8020000000000000C000000
000033B9000B01000C0000000C0033B9
010033B90078CF670F
ch1 bv:0 sv:0 ua:0 acc0:0/12=0.00mah acc1:0/47411=0.00mah
ch2 bv:680 sv:0 ua:0 acc0:0/12=0.00mah acc1:0/47411=0.00mah
ch3 bv:2816 sv:1 ua:400 acc0:12/12=0.00mah acc1:112947/47411=5.16mah

0000000000000000000000000C000000
00003FB9A0020000000000000C000000
00003FB9000B01000C0000000C003FB9
01003FB900713F2E0F
ch1 bv:0 sv:0 ua:0 acc0:0/12=0.00mah acc1:0/47423=0.00mah
ch2 bv:672 sv:0 ua:0 acc0:0/12=0.00mah acc1:0/47423=0.00mah
ch3 bv:2816 sv:1 ua:400 acc0:12/12=0.00mah acc1:112959/47423=5.16mah
```
(ran this overnight, 99.2% of transfers were clean, 0.8% corruption due to race condition around crc calculation where data looks fine but the data changed during crc calculation)

when the slave is change to SLEEP_MODE_STANDBY, the output changes to:

[content in square brackets are comments I added]
```
[this is the slave being updi programmed and not responding]
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
ch1 bv:65535 sv:-1 ua:-400 acc0:-1/65535=-0.00mah acc1:-1/65535=-0.00mah
ch2 bv:65535 sv:-1 ua:-400 acc0:-1/65535=-0.00mah acc1:-1/65535=-0.00mah
ch3 bv:65535 sv:-1 ua:-400 acc0:-1/65535=-0.00mah acc1:-1/65535=-0.00mah

[the first read succeeds]
00000000000000000000000006000000
00000600F00200000000000006000000
00000600F80A01000600000006000600
0000060000CEC895DF
ch1 bv:0 sv:0 ua:0 acc0:0/6=0.00mah acc1:0/6=0.00mah
ch2 bv:752 sv:0 ua:0 acc0:0/6=0.00mah acc1:0/6=0.00mah
ch3 bv:2808 sv:1 ua:400 acc0:6/6=0.00mah acc1:6/6=0.00mah

[subsequent read starts failing. the second line is pretty clearly a fourth line given the CRC looking thing in the middle followed by FFFFFs which are an underrun]
rcv_crc:38ACA57E calc:A3E7FCC4
bad crc, will retry 2 times
00001300F80A01001300000013001300
00001300007EA5AC38FFFFFFFFFFFFFF
00001300F80A01001300000013001300
00001300007EA5AC38
rcv_crc:38ACA57E calc:A3E7FCC4
bad crc, will retry 1 times
00001300F80A01001300000013001300
00001300007EA5AC38FFFFFFFFFFFFFF
[... 222 lines cut ...]
ch2 bv:5396 sv:5910 ua:4704 acc0:454695192/7452=20764.41mah acc1:555753246/8994=25379.40mah
ch3 bv:9508 sv:10022 ua:11104 acc0:724183336/11564=33071.04mah acc1:825241390/13106=37686.02mah

000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244
ch1 bv:1284 sv:1798 ua:-1696 acc0:185207048/3340=8457.79mah acc1:286265102/4882=13072.77mah
ch2 bv:5396 sv:5910 ua:4704 acc0:454695192/7452=20764.41mah acc1:555753246/8994=25379.40mah
ch3 bv:9508 sv:10022 ua:11104 acc0:724183336/11564=33071.04mah acc1:825241390/13106=37686.02mah
rcv_crc:18171615 calc:E8257576
bad crc, will retry 2 times
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
000102030405060708090A0B0C0D0E0F
101112131415161718

000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244
ch1 bv:1284 sv:1798 ua:-1696 acc0:185207048/3340=8457.79mah acc1:286265102/4882=13072.77mah
ch2 bv:5396 sv:5910 ua:4704 acc0:454695192/7452=20764.41mah acc1:555753246/8994=25379.40mah
ch3 bv:9508 sv:10022 ua:11104 acc0:724183336/11564=33071.04mah acc1:825241390/13106=37686.02mah
```

MX682X, 2022-08-12:
I've started with debugging right now and I stumbled upon 2 things so far that might create problems: 
1. Do you make sure that Serial finishes before putting the CPU into sleep? - It's not impacting in IDLE, but will probably in the deeper sleep states, since USART can wake the MCU only on SOF. the MCU might go to sleep with bytes still left in the buffer or maybe a half-finished transmitted byte.
2. I noticed that a BUSERR is thrown by the TWI SLAVE in the last byte of a transaction when in STANDBY (and I expect POWER_DOWN). I think this is kind of a silicon bug. When in Standby or Power Down, all High-Frequency clocks are disabled, and thus the clock for TWI MASTER. But the TWI MASTER has to be enabled for BUSERR to work. So, the BUSERR circuitry does not register the Start condition, but only a STOP condition and thus creates the BUSERR. This leads the Wire library to virtually throw away the whole transaction. I'll have to think about how to work around that, if someone has any suggestions, like a "this interrupt has woken up the CPU flag", if something like this exists, feel free to write a comment

SpenceKonde, 2022-08-12:
But the datasheet says it can wake on a i2c start condition from deeper
modes, even classic avrs could

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

ObviousInRetrospect, 2022-08-12:
1. in my real code I have a serial.flush() prior to sleep; I should add it to the sample code.
2. given that I am already checking  Wire.slaveTransactionOpen() before sleep I have no problem either passing a flag warning that I am sleeping or similar or calling some new Wire.sleep() which waits until a slave transaction is not open and then sleeps (setting a flag). although that probably won't work with other wakeup sources. FWIW [line 309](https://github.com/ObviousInRetrospect/DualMode/blob/10dfca6b5bc6c8fedf1605d2d8c5981e2e4aca34/DualModeExample/DualModeExample.ino#L309) is
```
TCB2.CTRLA |= 1 << TCB_RUNSTDBY_bp;
```
which I think keeps the HF clock running in standby (and millis keeping time)

ObviousInRetrospect, 2022-08-12:
should have said this up-front but I am doing my testing on 1.4.10 (latest release) against an avr64db32 (which had less errata than the 128's or the da's). 

fwiw I am 80% sure the same issue exists on the latest megaTinyCore release (on a tiny1614), I can try to build a repro if it would be helpful.

in theory the chips will wake from sleep-mode-power-down on i2c slave address match. I've thought about trying to screw with the clock prescaler to make sleep mode idle suck less but I'd need to hack a revert-the-clock into all the ISRs

Appreciate you looking at it. These parts would make awesome slaves if they could do so from deep sleep. A tiny412 is cheaper than lots of special purpose i2c devices (want an extra dac? ...).

MX682X, 2022-08-13:
Since I sometimes fail to convey my thoughts properly because I might leave out some knowledge I assume other people already have:

For clarification: a TWI transaction consists of
```
START, Address, optional Data and a STOP 
              ^(Addr IRQ)   ^(Data IRQ) ^(STOP IRQ)
              ^wake-up happens here, with a little clock stretch for the oscillator
```

> But the datasheet says it can wake on a i2c start condition from deeper modes, even classic avrs could

It can and it does. (I only got to test STANDBY so far though.) The problem is a bit different. (No pun intended)
Basically, it is just the BUSERR detection that behaves unintended. Since it gets no clock in STANDBY (my theory), it misses the START condition, and it's job is 
> The BUSERR flag indicates that an illegal bus operation has occurred. Illegal bus operation is detected if a protocol
> violating the Start (S), repeated Start (Sr), or Stop (P) conditions is detected on the TWI bus lines. A Start condition
> directly followed by a Stop condition is one example of a protocol violation.

But the library is checking for the BUSERR bit and cancels any transaction that occured if it is set. I think the most straightforward solution would be to check if the SMODE in the sleep register is not IDLE. In that case we ignore the BUSERR, but still clear it

> which I think keeps the HF clock running in standby (and millis keeping time)

It does, but to conserve as much power as possible, only enabled peripherals get the clock. the TWI_MASTER has no RUNSTBY.

> given that I am already checking Wire.slaveTransactionOpen() before sleep I have no problem either passing a flag warning that I am sleeping or similar or calling some new Wire.sleep() which waits [...]

I think I'll do it by checking the selected sleepmode

MX682X, 2022-08-13:
[twi.c.pdf](https://github.com/SpenceKonde/DxCore/files/9331203/twi.c.pdf)
(too lazy to archive, just remove the .pdf after downloading)

@ObviousInRetrospect 
Try to replace your twi.c with this one. It seems to work on my end, at least it is reading the 120 byte buffer I prepared correctly. On the 128DA48 at least. (didn't use your sketch for testing)

SpenceKonde, 2022-08-13:
> But the library is checking for the BUSERR bit and cancels any transaction that occured if it is set. I think the most straightforward solution would be to check if the SMODE in the sleep register is not IDLE. In that case we ignore the BUSERR, but still clear it

I agree 1000% 

I came to the same conclusion as you did in the shower about what was causing it.

ObviousInRetrospect, 2022-08-13:
Still getting basically 100% corruption in SLEEP_MODE_STANDBY with the new twi.c. Tried deleting twi.c in the middle and verified I got a bunch of linker errors (until I dropped the new one in) so some evidence it is picking up the new code.

While testing this I was wondering if there is a way to make it error out if you call enableDualMode and have set master or slave in the menu - somehow my boards menu got reset which took me a while to track down. Perhaps just #ifdef that function if you would never call it in master or slave.

```
rcv_crc:00000002 calc:FA25D711
bad crc, will retry 2 times
00000000000000000000000008000000
00000800F00200000000000008000000
00000000000000000000000008000000
00000800F002000000
rcv_crc:00000002 calc:6E9EBE02
bad crc, will retry 1 times
00000000000000000000000009000000
00000900F00200000000000009000000
00000000000000000000000009000000
00000900F002000000

00000000000000000000000009000000
00000900F00200000000000009000000
00000000000000000000000009000000
00000900F002000000
ch1 bv:0 sv:0 ua:0 acc0:0/9=0.00mah acc1:0/9=0.00mah
ch2 bv:752 sv:0 ua:0 acc0:0/9=0.00mah acc1:0/0=0.00mah
ch3 bv:0 sv:0 ua:0 acc0:0/9=0.00mah acc1:0/9=0.00mah
rcv_crc:00000002 calc:D2E7F7AF
bad crc, will retry 2 times
00000000000000000000000015000000
00001500F00200000000000015000000
00000000000000000000000015000000
00001500F002000000
rcv_crc:00000002 calc:D2E7F7AF
bad crc, will retry 1 times
00000000000000000000000015000000
00001500F00200000000000015000000
00000000000000000000000015000000
00001500F002000000

00000000000000000000000015000000
00001500F00200000000000015000000
00000000000000000000000015000000
00001500F002000000
ch1 bv:0 sv:0 ua:0 acc0:0/21=0.00mah acc1:0/21=0.00mah
ch2 bv:752 sv:0 ua:0 acc0:0/21=0.00mah acc1:0/0=0.00mah
ch3 bv:0 sv:0 ua:0 acc0:0/21=0.00mah acc1:0/21=0.00mah
rcv_crc:00000002 calc:30E79125
bad crc, will retry 2 times
00000000000000000000000021000000
00002100F00200000000000021000000
00000000000000000000000021000000
00002100F002000000
rcv_crc:00000002 calc:30E79125
bad crc, will retry 1 times
00000000000000000000000021000000
00002100F00200000000000021000000
00000000000000000000000021000000
00002100F002000000

00000000000000000000000021000000
00002100F00200000000000021000000
00000000000000000000000021000000
00002100F002000000
ch1 bv:0 sv:0 ua:0 acc0:0/33=0.00mah acc1:0/33=0.00mah
ch2 bv:752 sv:0 ua:0 acc0:0/33=0.00mah acc1:0/0=0.00mah
ch3 bv:0 sv:0 ua:0 acc0:0/33=0.00mah acc1:0/33=0.00mah
rcv_crc:00000002 calc:E5DF0BA7
bad crc, will retry 2 times
0000000000000000000000002C000000
00002C00F0020000000000002C000000
0000000000000000000000002C000000
00002C00F002000000
rcv_crc:00000002 calc:F36675D3
bad crc, will retry 1 times
0000000000000000000000002C000000
00002C00F0020000000000002C000000
0000000000000000000000002D000000
00002D00F002000000

0000000000000000000000002C000000
00002C00F0020000000000002C000000
0000000000000000000000002D000000
00002D00F002000000
ch1 bv:0 sv:0 ua:0 acc0:0/44=0.00mah acc1:0/44=0.00mah
ch2 bv:752 sv:0 ua:0 acc0:0/44=0.00mah acc1:0/0=0.00mah
ch3 bv:0 sv:0 ua:0 acc0:0/45=0.00mah acc1:0/45=0.00mah
rcv_crc:00000002 calc:275B1F14
bad crc, will retry 2 times
00000000000000000000000038000000
00003800F00200000000000038000000
00000000000000000000000038000000
00003800F002000000
rcv_crc:00000002 calc:B3E07607
bad crc, will retry 1 times
00000000000000000000000039000000
00003900F00200000000000039000000
00000000000000000000000039000000
00003900F002000000

00000000000000000000000039000000
00003900F00200000000000039000000
00000000000000000000000039000000
00003900F002000000
ch1 bv:0 sv:0 ua:0 acc0:0/57=0.00mah acc1:0/57=0.00mah
ch2 bv:752 sv:0 ua:0 acc0:0/57=0.00mah acc1:0/0=0.00mah
ch3 bv:0 sv:0 ua:0 acc0:0/57=0.00mah acc1:0/57=0.00mah
rcv_crc:00000002 calc:6E15A9E1
bad crc, will retry 2 times
00000000000000000000000045000000
00004500F00200000000000045000000
00000000000000000000000045000000
00004500F002000000
rcv_crc:00000002 calc:6E15A9E1
bad crc, will retry 1 times
00000000000000000000000045000000
00004500F00200000000000045000000
00000000000000000000000045000000
00004500F002000000

00000000000000000000000045000000
00004500F00200000000000045000000
00000000000000000000000045000000
00004500F002000000
ch1 bv:0 sv:0 ua:0 acc0:0/69=0.00mah acc1:0/69=0.00mah
ch2 bv:752 sv:0 ua:0 acc0:0/69=0.00mah acc1:0/0=0.00mah
ch3 bv:0 sv:0 ua:0 acc0:0/69=0.00mah acc1:0/69=0.00mah
```

ObviousInRetrospect, 2022-08-13:
mind sharing the test code you have working?

I wonder if it is related to the 32-byte chunking I am doing

SpenceKonde, 2022-08-13:
Oh shit the code I merged didn't work?

ObviousInRetrospect, 2022-08-13:
In my testing it didn't resolve the problem I am seeing. It seems to work similarly to the prior version. It probably fixed at least part of the problem in @MX682X 's testing so merging it might not be a mistake. I've generally found twi slave via wire to be mega-flakey when standby+ modes are used so there might be more than one issue.

MX682X, 2022-08-13:
sender: attiny1614 with modded TWI buffer size: 
```
#include "Wire.h"

void setup() {
  pinMode(PIN_PA3, INPUT_PULLUP);
  Wire.usePullups();
  Wire.swap(0);
  Wire.begin();
  
}

void loop() {
  if (digitalRead(PIN_PA3) == LOW) {
    Wire.beginTransmission('A');
    for (uint8_t i = 0; i < 120; i++) {
      Wire.write(i);
    }
    Wire.endTransmission(true);

    while (digitalRead(PIN_PA3) == LOW);
  }
  delay(100);
}
```

Receiver: (AVR128DA48 Curiosity nano):
```
 #include "Wire.h"
#include <avr/sleep.h>

volatile uint8_t numRx = 0;
volatile uint8_t rxBuffer[128];
volatile uint8_t sleep_en = 0;

void setup() {
  TWI0.DBGCTRL = 1;
  USART1.DBGCTRL = 1;
  //set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  //set_sleep_mode(SLEEP_MODE_IDLE);
  set_sleep_mode(SLEEP_MODE_STANDBY);

  sleep_enable();
  
  pinMode(PIN_PC6, OUTPUT);
  digitalWrite(PIN_PC6, HIGH);
  
  Wire.usePullups();
  
  Wire.onReceive(srmw);
  Wire.begin('A');
  
  Serial1.begin(500000);
  while (Serial1.read() == -1) {};
  Serial1.println("Hello World");
}

void loop() {
  // put your main code here, to run repeatedly:
  while (Wire.slaveTransactionOpen() != 0) {}

  
  if (numRx > 0) {
    Serial1.print("Rx: ");
    for (uint8_t i = 0; i < numRx; i++) {
      Serial1.printHex(rxBuffer[i]);
    }
    Serial1.println();
    numRx = 0;
  }
  
  Serial1.flush();
  
  if (sleep_en) {
    digitalWrite(PIN_PC6, LOW);
    sleep_cpu();
  }
  
  digitalWrite(PIN_PC6, HIGH);
}

void srmw(int num) {
  Serial1.print("IRQ ");
  for (uint8_t i = 0; i < num; i++) {
    rxBuffer[i] = Wire.read();
  }
  numRx = num;
}
```

I was also monitoring the PIN_PC6 on my Oscilloscope, so I knew when the chip was asleep

MX682X, 2022-08-13:
> make it error out if you call enableDualMode and have set master or slave in the menu

This would be a misleading error, since you can use the dual mode with the slave only.

ObviousInRetrospect, 2022-08-13:
so fwiw it is possible that doing a single write pattern (filling the modified-size-buffer) instead of sequential reads and writes of different is masking the problem.

using my code with the test pattern data (which you can totally run without an ina3221 - the ina3221 code will silently read 0xFFs one of the most common corruption patterns I see is the read of the second half returns a duplicate set of first half data when sleeping in standby:

```
bad crc, will retry 2 times
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
000102030405060708090A0B0C0D0E0F
101112131415161718
rcv_crc:18171615 calc:E8257576
bad crc, will retry 1 times
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
000102030405060708090A0B0C0D0E0F
101112131415161718
```

MX682X, 2022-08-14:
Alright, so seems like I've introduced a bug that automatically called the onReceive function even though no data was available. I should add that I wanted to go step by step and fiirst figure out onReceive, then onRequest. That's where the problem above might have come from. Could you try this version out? should have the other parts fixed aswell as the getBytesRead() function
[src.zip](https://github.com/SpenceKonde/DxCore/files/9332968/src.zip)

ObviousInRetrospect, 2022-08-14:
It is definitely not completely fixed:
```
bad crc, will retry 2 times
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
000102030405060708090A0B0C0D0E0F
101112131415161718
CRC valid
fail: 24 pass: 23
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244
```

My perception though is it is succeeding more often than it used to. I just added the pass/fail counters and can go back and do a longer test comparing the two.

ObviousInRetrospect, 2022-08-14:
Did more testing. For these tests I sped up the reads to read at 10hz instead of 0.2hz. This is with test pattern data for which I took out the recalculation race condition. 

sleep_mode_standby: (prev >50% corruption rate)
```
CRC valid
fail: 0 pass: 10576
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244
```

also seeing perfect behavior in SLEEP_MODE_PWR_DOWN:
```
CRC valid
fail: 0 pass: 2506
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244
```

will do additional testing removing some of the various delay()'s that I used to use to make it behave slightly better.

ObviousInRetrospect, 2022-08-14:
so first off, THANK YOU, this is hugely better. The failure rate is down by a factor of at least 2500x.

Testing in prior post and this post was conducted with this version of the code:
https://github.com/ObviousInRetrospect/DualMode/commit/0499bbe852f923fd49f5aff3fac51d8ff9a7a92f

As you can see most of the competing code is IFDEF'ed out.

Slave was running on an avr64dd32 on pins c2/c3 (10mhz)
Master was running on a tiny3224 (4mhz) - not built with your new library, not entirely on purpose just didn't drop it into megaTinyCore yet.
Both boards are well-decoupled 2 layer pcbs (running both 0.1 and 1uf on all power pins on the slave). Each board has 4.7k pull-ups to 2.8v vcc, so 2.35k.

A longer test of sleep_mode_pwr_down hit two types of problem. After 2770 successes the slave stopped responding until the reset button was hit, yielding 681 failures. In this failure mode the slave's main look keeps running.  (The counts were not reset at this point). After 8179 successes another failure was hit in which the first transmission had incorrect bytes at the front and 0xFFs for about half.

This appears to be reproducible although the number of transactions in that the crash happens is variable.

```text
CRC valid
fail: 0 pass: 2770
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244
ch1 bv:1284 sv:1798 ua:-1696 acc0:185207048/3340=8457.79mah acc1:286265102/4882=13072.77mah
ch2 bv:5396 sv:5910 ua:4704 acc0:454695192/7452=20764.41mah acc1:555753246/8994=25379.40mah
ch3 bv:9508 sv:10022 ua:11104 acc0:724183336/11564=33071.04mah acc1:825241390/13106=37686.02mah
rcv_crc:FFFFFFFF calc:C530B076
bad crc, will retry 2 times
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 1 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
fail: 2 pass: 2770
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
... <omitting ~679 duplicate failures>
fail: 680 pass: 2770
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
ch1 bv:65535 sv:-1 ua:-400 acc0:-1/65535=-0.00mah acc1:-1/65535=-0.00mah
ch2 bv:65535 sv:-1 ua:-400 acc0:-1/65535=-0.00mah acc1:-1/65535=-0.00mah
ch3 bv:65535 sv:-1 ua:-400 acc0:-1/65535=-0.00mah acc1:-1/65535=-0.00mah
rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 2 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF

<RESET SLAVE via RST pin>
CRC valid
fail: 681 pass: 2771
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244

...

rcv_crc:44A2C3A5 calc:A55704A9
bad crc, will retry 2 times
[... 143 lines cut ...]
CRC valid
fail: 0 pass: 4297
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244

rcv_crc:FFFFFFFF calc:4BFBA3F3
bad crc, will retry 2 times
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 1 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
fail: 2 pass: 4297

<slave is not responding over i2c. 
slave main loop is still running>

```

ObviousInRetrospect, 2022-08-14:
running the master at 10x the speed of the slave (10mhz/1mhz), my perception is the crash rate goes up:
(this is still 16 chunk read on the master and 32 chunk write on the slave)

```
CRC valid
fail: 0 pass: 3199
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244

rcv_crc:FFFFFFFF calc:0493FDE6
bad crc, will retry 2 times
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
20FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 1 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF

...
<reset>

CRC valid
fail: 284 pass: 3441
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244

rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 2 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 1 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF

<note this was only ~200 good>

CRC valid
fail: 538 pass: 11336
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244

rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 2 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE

<but this was ~8k>

CRC valid
fail: 592 pass: 14374
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244

rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 2 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF

<another reasonably fast failure>

CRC valid
fail: 795 pass: 15507
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244

rcv_crc:FFFFFFFF calc:0493FDE6
bad crc, will retry 2 times
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
20FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
<reset>
...
RC valid
fail: 873 pass: 19928
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244

rcv_crc:FFFFFFFF calc:0493FDE6
bad crc, will retry 2 times
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
20FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 1 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
<reset>
...

CRC valid
fail: 932 pass: 21103
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244

rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 2 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 1 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
...
<reset>
CRC valid
fail: 1019 pass: 25165
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244

CRC valid
fail: 1019 pass: 26367
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244

rcv_crc:FFFFFFFF calc:0493FDE6
bad crc, will retry 2 times
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
20FFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 1 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF

```

ObviousInRetrospect, 2022-08-14:
And back to sleep_mode_standby, same slow (1mhz) slave fast (10mhz) master, no errors in 65k operations!!! that means this is totally usable in standby.

code version: https://github.com/ObviousInRetrospect/DualMode/commit/5f6f3e75b4ad620d8185442521d569b4a2592c6e:

```
CRC valid
fail: 0 pass: 65060
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244
```

SpenceKonde, 2022-08-14:
You got this shit sorted?! You da man!!!I wouldn have had no idea how to deal with and "won't fix'ed" it because i didn't have a clue where to begin.

ObviousInRetrospect, 2022-08-14:
@MX682X got it sorted, I'm just testing. Now to write an example that turns a t412 into an i2c dac while making correct use of sleep.

@SpenceKonde this fix should get a mention in the release notes after its merged ("problems with i2c slave mode in sleep_mode_standby resolved", unless @MX682X has a better note). I had given up on i2c slaves being able to sleep below idle as causing flakey behavior and stuck i2c buses, I am probably not the only one so calling out the change would be helpful.

I don't critically need sleep_mode_pwr_down if you don't feel like debugging that mode (in which case we should put a note in the documentation recommending against using that mode with i2c slaves) but I am happy to help run it down if you want to.

meanwhile the test hit 107k without error in sleep_mode_standby:

```text
CRC valid
fail: 0 pass: 107039
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
3031323334A5C3A244
```

I'll stop it for now but try to run it overnight.

SpenceKonde, 2022-08-15:
Thank you, I was clueless here. 

Does this also need to bported to mTC?

ObviousInRetrospect, 2022-08-15:
yes this also applies to mTC.

I'm not totally sold on the grand unified power management library. I think some examples and docs might cover quite a bit of it.

the biggest ROI for a library imo is a sleepy version of delay(). 

FWIW I also think Serial works less well in sleep mode standby than it should but I haven't characterized the problem yet. In theory it should support wake on serial but get a lot of corruption

MX682X, 2022-08-15:
Yes, most definetely impacts mTC. Spence, you mentioned once you know someone from Microchip. Maybe it's worth pointing out that out so they might add it to their errata? That when waking up from sleep might have the BUSERR flag set when receiving the STOP condition.

> A longer test of sleep_mode_pwr_down hit two types of problem. After 2770 successes the slave stopped responding until the reset button was hit, yielding 681 failures. In this failure mode the [...]

Are this problems reproducable? Because if they aren't it will be really hard to find them

ObviousInRetrospect, 2022-08-15:
Both problems are reproducible but variable time to reproduce. There are a few test traces above. always seems to happen within 10k cycles, usually much faster.

Of the two, the slave stopping responding entirely until reset is by far the worse bug and happens more often. Occasional data corruption can be worked around with a crc.

if you run my test code it will trigger both failures. no need for an ina3221. that code path is now mostly ignored if you #define TEST_PATTERN and the parts that aren't ignored fail harmlessly.

Meanwhile in overnight testing:

```
CRC valid
fail: 0 pass: 3476989
```

Re the errata, I have had better luck with Microchip support than I expected (as a non-entity). If you open a documentation case asking where the behavior is documented you might actually get somewhere.

MX682X, 2022-08-17:
Well... I've let your code run on a AVR128DA48 on Curiosity Nano and a Attiny1614, also connected to a debugger, and had no problems so far with SLEEP_MODE_POWER_DOWN. I tried the maximum CPU Frequency aswell as 10MHz on DA, 4 MHz on 1614.
It's 100k+ passed CRCs. I had more problems getting the correct UART BAUD rate on the 1614 at MHz then anything else.
To be fair, I was using a slightly modified version of the Wire library. Let's hope that's the reason why I can't reproduce your error, because otherwise, I don't think I will be able to find the error.

[src.zip](https://github.com/SpenceKonde/DxCore/files/9364056/src.zip)

SpenceKonde, 2022-08-17:
Is that a version of the wire library that I should pull into the core?

ObviousInRetrospect, 2022-08-18:
reverse the frequency: run the slave (DA) at 1mhz and the master (1614) at 10mhz

ObviousInRetrospect, 2022-08-18:
New code give me:

```
Users/user/Library/Arduino15/packages/DxCore/hardware/megaavr/1.4.10/libraries/Wire/src/twi_pins.c:468:68: error: 'PORTB' undeclared (first use in this function); did you mean 'PORT0'?
     PORT_t *port = (PORT_t *)((portmux == PORTMUX_TWI1_ALT2_gc) ? &PORTB : &PORTF);
                                                                    ^~~~~
                                                                    PORT0
/Users/user/Library/Arduino15/packages/DxCore/hardware/megaavr/1.4.10/libraries/Wire/src/twi_pins.c:468:68: note: each undeclared identifier is reported only once for each function it appears in
```

So repro attempt 1 (old code):

```
CRC valid
fail: 0 pass: 199

CRC valid
fail: 0 pass: 200

rcv_crc:FFFFFFFF calc:20BEDCF9
bad crc, will retry 2 times
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292A2B2C2D2E2F
30FFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 1 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
```

repro attempt 2:
```
CRC valid
fail: 0 pass: 1447

CRC valid
fail: 0 pass: 1448

CRC valid
fail: 0 pass: 1449

rcv_crc:FFFFFFFF calc:C530B076
bad crc, will retry 2 times
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 1 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
fail: 2 pass: 1449
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
```

these were running both sides at 10mhz

ObviousInRetrospect, 2022-08-18:
and with the new code with the following patch applied:
```patch
--- /Users/user/Downloads/src-2/twi_pins.c	2022-08-17 15:34:06.000000000 -0700
+++ /Users/user/Library/Arduino15/packages/DxCore/hardware/megaavr/1.4.10/libraries/Wire/src/twi_pins.c	2022-08-17 19:43:12.000000000 -0700
@@ -464,9 +464,10 @@
 
 void TWI1_usePullups() {
   #if defined(TWI1)
-    uint8_t portmux = PORTMUX.TWIROUTEA & PORTMUX_TWI1_gm;
-    PORT_t *port = (PORT_t *)((portmux == PORTMUX_TWI1_ALT2_gc) ? &PORTB : &PORTF);
-
+    //uint8_t portmux = PORTMUX.TWIROUTEA & PORTMUX_TWI1_gm;
+    //PORT_t *port = (PORT_t *)((portmux == PORTMUX_TWI1_ALT2_gc) ? &PORTB : &PORTF);
+    uint8_t temp = PORTMUX.TWIROUTEA & PORTMUX_TWI1_gm;
+    PORT_t *port = portToPortStruct(temp == 8?PB:PF);
     port->OUTCLR = 0x0C;  // bits 2 and 3
     port->PIN2CTRL |= PORT_PULLUPEN_bm;
     port->PIN3CTRL |= PORT_PULLUPEN_bm;
```

try 1:
```text 
CRC valid
fail: 0 pass: 2002

CRC valid
fail: 0 pass: 2003

CRC valid
fail: 0 pass: 2004

rcv_crc:FFFFFFFF calc:FEA3EF34
bad crc, will retry 2 times
000102030405060708FFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 1 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
```

try 2:
```text
CRC valid
fail: 0 pass: 5448

CRC valid
fail: 0 pass: 5449

rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 2 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
```

try 3: (timed at about 4 seconds)
```
CRC valid
fail: 0 pass: 481

rcv_crc:FFFFFFFF calc:8088E2CE
bad crc, will retry 2 times
000102030405060708090A0B0C0D0E0F
101112131415161718191A1BFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
```

ObviousInRetrospect, 2022-08-18:
just to verify: you applied this patch to DualModeExample.ino:

```patch
@ -326,7 +326,7 @@ void setup() {
  pinMode(P_LED, OUTPUT);
  digitalWriteFast(P_LED, 1);
  RTC_init();
-  set_sleep_mode(SLEEP_MODE_STANDBY);
+  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  sleep_enable();
#ifdef ENABLE_SCAN     // i2c scan
  byte error, address; // variable for error and I2C address

```

And are running the slave (DualModeExample.ino) in Master & Slave (dual mode) on the DA using DxCore 1.4.10 and a master on a t1614?

Have you tried it without the debugger enabled? 

Is the Curiosity Nano a DM164151? (https://www.digikey.com/en/products/detail/microchip-technology/DM164151/12177481) - if so I will pick up a copy. I'm sure the 1614 vs 3224 doesn't matter but have 1614s lying around (ironically I just desoldered the 1614 from the board acting as a master and replaced it with a 3224).

MX682X, 2022-08-18:
I did some changes in the setup function :

```
void setup() {
  memset(reg.r, 0, sizeof(ewdt_regs_t));
  memset(bak.r, 0, sizeof(ewdt_regs_t));
  TCB2.CTRLA |= 1 << TCB_RUNSTDBY_bp;
  
  Wire.usePullups();
  Wire.enableDualMode(false);

  Wire.onReceive(receiveHandler);
  Wire.onRequest(requestHandler);
  Wire.pins(PIN_PA2, PIN_PA3);
  Wire.begin();
  Wire.begin(0x42);

  Serial1.begin(115200);
  Serial1.println("boot");
  
  ina_reset();
  ina_config();
  
  pinMode(P_LED, OUTPUT);
  pinMode(PIN_PC6, OUTPUT);  (Built in LED)
  digitalWriteFast(P_LED, 1);
  digitalWriteFast(PIN_PC6, LOW);
  
  RTC_init();
  set_sleep_mode(SLEEP_MODE_PWR_DOWN);
  sleep_enable();
  
#ifdef ENABLE_SCAN     // i2c scan
  ... is the same
#endif

  #ifdef TEST_PATTERN
    rd_ina3221();
  #endif
}
```
also: 
```
while (!wake && !Wire.slaveTransactionOpen()) {
    Serial1.flush();
    digitalWriteFast(PIN_PC6, LOW);
    sleep_cpu();
    digitalWriteFast(PIN_PC6, HIGH);
    //delay(1);
  }
```

```
void setup() {
  // put your setup code here, to run once:
 Wire.usePullups();
  Wire.swap(0);
  Wire.begin();
  
  Serial.begin(115200);
  
  byte error, address; // variable for error and I2C address
  int nDevices;

  do{
  //rest is same
```

Yes, I'm using two DM164151  to debug both chips. (You'll need to change some settings in Microchip studio to debug any other chips).
DA is running at 4MHz. Core didn't want to work at 1MHz - a bug?
1614 is running at 10MHz (didn't check the actual frequency, I trust the Board settings)

Also I didn't expect any chip with TWI1 to not have a PORTB, so that's where the error came from, 

Without a debugger running, I only got CRC errors on startup, then it run error free 100k+ times. (I've used the Debugger to reset them before, didn't bother to add a pin IRQ driven software Reset.

ObviousInRetrospect, 2022-08-19:
I have a DM164151 coming tomorrow mostly to see if it reproduces on a 128DA48 - wouldn't shock me if there were differences (and a couple ICE3s if I need to watch more than one MCU). I was able to repro it with a debugger attached although I am have not a clue what I should be looking for. On AVR I mostly do printf debugging because I am usually on a Mac and mplab-x is horribly slow and crashes too often for me to use it. But if there is something you want me to inspect I can.

MX682X, 2022-08-19:
>  I was able to repro it with a debugger attached although I am have not a clue what I should be looking for.

I had my Oscilloscope connected to the data lines and a breakpoint when the CRC did not match, so I would see what was really transmitted (best way to see where the corruption originated). If you don't have a way to monitor the datalines, the best way would be to have both chips hit a break point when the CRC didn't match and compare the buffers.

ObviousInRetrospect, 2022-08-21:
So I just repro'ed it 3x on a DM164151 very quickly. So much for my theory this is DB only or something wrong with my hardware.

Have you tried running the code as-is (set to PWR_DOWN and with #define TEST_PATTERN without the debugger and without the gpio modifications just to see if you can repro the failure?

I am using dual mode and 2x master and slave (C2/C3 is where the slave that is failing is).

I could rig up a logic analyzer but I was assuming that the slave is just not responding and the master is thus clocking in 0xFF's as it would when talking to a non-existent slave. I just went ahead and set a flag in request handler and receive handler that triggers a print in the loop and can confirm that neither handler is being invoked once the failure happens.

I am not sure how you are setting a breakpoint on the slave when the master detects corruption.

```
CRC valid
fail: 0 pass: 226

rcv_crc:FFFFFFFF calc:6B399483
bad crc, will retry 2 times
000102030405060708090A0B0C0D0E0F
101112131415161718191A1B1C1D1E1F
202122232425262728292AFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 1 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
fail: 2 pass: 226
```

```
CRC valid
fail: 0 pass: 257

rcv_crc:FFFFFFFF calc:9E7FF7D1
bad crc, will retry 2 times
000102030405060708090A0B0C0D0E0F
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 1 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
```

```

CRC valid
fail: 1936 pass: 2056

rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 2 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
```

I assume you are 100% sure you are going into SLEEP_MODE_PWR_DOWN and something about the debugger setup isn't keeping it from sleeping?

ObviousInRetrospect, 2022-08-21:
[DxCore.megaavr.avrda.zip](https://github.com/SpenceKonde/DxCore/files/9388393/DxCore.megaavr.avrda.zip)
is a built copy for the DM164151 - that might be useful to figure out why mine crashes constantly and yours doesn't - if you write that does it crash? This is set to 10mhz and the slave is on PC2/PC3. Pull-ups are not enabled.

[DualModeExample.zip](https://github.com/SpenceKonde/DxCore/files/9388395/DualModeExample.zip)
is the exact inputs that generated it

MX682X, 2022-08-21:
I think I found the problem. Could you confirm it by uncomenting TWI_TIMEOUT_ENABLE in twi.h for the device that runs the DualModeExampleClient sketch?

MX682X, 2022-08-21:
The problem was actually in your code. 

```
while (!wake && !Wire.slaveTransactionOpen()) { 
    Serial.flush(); 
    sleep_cpu();
    //delay(1);
  }
```

[image]

Basically, your code prepares to sleep before onRequest is started, and sleeps after it was started. You have to move Serial.flush out of this while loop.

There are multiple solutions: 
A: use endTransmission (false) before requestFrom- this does not generate a STOP condition between receive/request and thus gives the option to completely process everything.
B: move Serial.flush outside of this while loop - works on my end right now

MX682X, 2022-08-21:
furthermore, I recommend to do check, if the expected amount of bytes were received, like this:
`if ( len != Wire.requestFrom(devaddr, (size_t)len) ) `

ObviousInRetrospect, 2022-08-21:
Really appreciate all the help with this issue. I think you found the root cause but I don't think we have a working fix.

I actually added the code you flagged:
```
if(printflag){
    Serial.print("+");
    printflag=0;
  }
```

Between the second and third repro (see my comment above "I just went ahead and set a flag in request handler and receive handler that triggers a print in the loop and can confirm that neither handler is being invoked once the failure happens."), it wasn't required for the problem or any of the earlier traces reported on previous days (but did make it happen much faster). It was added to confirm that once the slave stopped responding the handlers also were no longer being invoked.

Unfortunately, the change you proposed:
```
  while (!wake && Wire.slaveTransactionOpen()) {
    delay(1);
  }
  if(!wake){
    Serial.flush();
  }
  while (!wake && !Wire.slaveTransactionOpen()) {
    sleep_cpu();
    //delay(1);
  }
```

Still lets me repro the problem, although it cuts the incidence rate by many orders of magnitude. I would have called it fixed if I hadn't gotten (un?)lucky with the first failure happening very quickly:

```
CRC valid
fail: 0 pass: 2957

rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 2 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
```

because the second failure took 144k cycles to happen:

```
CRC valid
fail: 3840 pass: 144212

CRC valid
fail: 3840 pass: 144213

rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 2 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
rcv_crc:FFFFFFFF calc:33E718CE
bad crc, will retry 1 times
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFFFFFFFFFFFFFFFF
FFFFFFFFFFFFFFFFFF
fail: 3842 pass: 144213
```

I'll note that this new version still has a much narrower version of the same race condition in that an interrupt can fire between !Wire.slaveTransactionOpen() and the sleep_cpu() call. That was sort of bugging me when I wrote it but I have no clue how to solve that issue as it seems like the usual solution of adding cli()/sei() would just cause the mcu to sleep forever. Based on what you wrote above this is probably the source of the problem - the slave stops responding forever in this case. 

Is there a safe way to write this check? If not, would adding Wire.sleepIfSlaveTransactionNotOpen() at least shrink the race condition further? (by removing the function call return overhead between the check and the sleep). I assume an assembly version of this that did a skip-if-bit-set followed by a sleep is about as good as we can get.

If there isn't a way to remove this race condition, is there a way on the slave side to determine this has happened and fix things? I assume that any fix would require a second wake source beyond the now-defunct twi interrupt, probably PIT, which will greatly increase power use for some applications.

I think the other fixes you are proposing are on the master side and would cause the master to either detect the slave not responding or hang and don't solve the issue of once the slave hits this case it dies.

Including another zip so exactly what I am running is clear.
[DualModeExample.zip](https://github.com/SpenceKonde/DxCore/files/9389417/DualModeExample.zip)

Thanks again!

ObviousInRetrospect, 2022-08-21:
Taking a step back, I think we now know why the failure happens. The failure is bad - the slave totally stops responding and if the twi interrupt is the slave's only wakeup condition the slave hangs until reset. Unless there is a safe way to atomically sleep if slave transaction is not in progress or otherwise prevent this hang, perhaps the right thing to do is to document the fact that it is unsafe to use Wire.slave() in sleep mode pwr down. Standby is still very low power and a huge improvement over needing idle. In this case, it is debatable whether it might be more helpful to have Wire.slaveTransactionOpen assert that the sleep mode is not set to pwr_down (unless a false by default argument is set) to avoid people getting bitten by this.

MX682X, 2022-08-21:
`if(!wake){`
try it without this. always make sure that there is no ongoing serial transaction, then make sure no flag is set. 

If that won't work, you could try to replace the function with this:

```
uint8_t TwoWire::slaveTransactionOpen() {
   if (vars._module->SSTATUS & (TWI_DIR_bm | TWI_AP_bm) return 2;
   if (vars._module->SSTATUS & TWI_AP_bm return 1;
   return 0;
}
```

ObviousInRetrospect, 2022-08-21:
actually trying:
```
  while (!wake && !Wire.slaveTransactionOpen()) {
    cli();
    if(!Wire.slaveTransactionOpen()){
      sei();
      sleep_cpu();
    }
    else{
      sei();
    }
    //delay(1);
  }
```
http://ww1.microchip.com/downloads/en/devicedoc/atmel-0856-avr-instruction-set-manual.pdf says the instruction following SEI will be executed before any pending interrupts so assuming the compiler generated reasonable code this should actually be a fix!

ObviousInRetrospect, 2022-08-21:
Rather than assuming anything about avr-gcc I inspected the assembly. It looks right. This is actually a fix.

```
    171e:	f8 94       	cli
    1720:	0e 94 81 01 	call	0x302	; 0x302 <TwoWire::slaveTransactionOpen() [clone .constprop.6]>
    1724:	81 11       	cpse	r24, r1
/Users/user/Documents/GitHub/DualMode/DualModeExample/DualModeExample.ino:392
    cli();
    if(!Wire.slaveTransactionOpen()){
    1726:	03 c0       	rjmp	.+6      	; 0x172e <main+0x6e8>
    1728:	78 94       	sei
    172a:	88 95       	sleep
    172c:	fb ce       	rjmp	.-522    	; 0x1524 <main+0x4de>
```

I'm certainly happy to throw together an example and some documentation but I think it would be better if Wire provided an easy to use sleepUnlessSlaveTransactionOpen() function that encapsulated this ugliness. I2C slaves are often sleepy. They also are sufficiently Wire-centric that it doesn't feel wrong to delegate the sleep call to Wire.

MX682X, 2022-08-21:
weird. My compiler does not generate a call instruction... It inlines everything. Do you use 1x Wire or 2x Wire?

ObviousInRetrospect, 2022-08-21:
2x at the moment. Might be why I was having an easier time reproducing it.

ObviousInRetrospect, 2022-08-21:
so far I'm around 300k without a failure with the cli/sei added, I think its a fix but will obviously run it for many hours / overnight.

ObviousInRetrospect, 2022-08-21:
Do you have a version of the library that can be linked to from the docs (that handles the TWI1 without PORTB issue)? I kind of want to do a documentation update saying that in versions prior to 2.6/1.5 either use sleep_mode_idle or upgrade your library to <link>

MX682X, 2022-08-21:
Nope. I don't have a standalone library, but the TWI1/PORTB issue is already fixed locally.

ObviousInRetrospect, 2022-08-21:
Yeah I meant is it in a state where you want to attach it here as a zip to point people at (and to let @SpenceKonde merge)?

I'm 95% convinced this was the remaining problem I was having with pwr_down and you have now fixed the library to work flawlessly in all sleep modes (and overnight testing will cover the remaining 5%). Thanks again!

MX682X, 2022-08-21:
[src.zip](https://github.com/SpenceKonde/DxCore/files/9389669/src.zip)
I guess this is the release candidate then.

ObviousInRetrospect, 2022-08-21:
with 1.2m pass and no errors this is looking fixed
```
CRC valid
fail: 0 pass: 1204684
```

SpenceKonde, 2022-08-22:
I will contact my craftsmen at once to have them produce an extra mvp award today from the finest materials, as we clearly have two individuals to deliver them to. _"Cat, you hear that? We need a second trophy made for this month. Don't give me that about lead time. I see you snoozing on the job all day and I see how fast you shed, if you put that energy towards doing your job here, you would have no trouble meeting this deadline... and if you expect to get cat food next month, I will see two awards ready for the awards ceremony on the august 30! Do I make myself clear?"_

MX682X, 2022-09-11:
@ObviousInRetrospect after I managed to figure out how to optimize the slave interrupt routine (by convincing the compiler to use the Y-Register and ldd/std instead of sts/lds), I was able to change how slaveTransactionOpen() works internally without bloating the code size. I'd apreciate it if you could run a test without the cli/sei guards.

this version uses a variable to remember when an address match interrupt occurs, so slaveTrasactionOpen won't return 0 between address and stop interrupt and thus should not allow the system to sleep.  
[src.zip](https://github.com/SpenceKonde/DxCore/files/9542408/src.zip)

ObviousInRetrospect, 2022-09-11:
Happy to run the test (and will start it shortly) although I still think there is a highly improbable highly infrequent race condition without the cli/sei guard. 

The address interrupt can still fire between or during the return of slaveTransactionOpen() and the sleep instruction.

Unless I am missing something the cli/sei is correct and excludes a race condition that will cause extremely infrequent lock ups. I still am not sure I understand why the failure that happens is as bad as it is (slave fails until reset) in that I have a PIT interrupt causing non twi wake frequently - I would kind of expect the transaction the slave fell asleep on to fail but the transaction open to return true and prevent sleep. I think there might still be a missing recovery transition.

MX682X, 2022-09-11:
You are right. But I think reducing the time it might happen is desirable anyway. 

I've brainstormed a bit. What if the TWI ISR will set the sleep mode to IDLE at Address IRQ and restore it at Stop IRQ? This might even allow people to put the AVR into IDLE sleep between data IRQs further minimizing power consumption, instead of letting the CPU doing nothing...

ObviousInRetrospect, 2022-09-11:
The savings from sleep mode idle are very minor, I wouldn't jump through hoops to sit in idle while a slave transaction is open vs busy-wait. 

Or are you suggesting it not for the purpose of allowing idle sleep for that limited time but instead for the purpose of making it safe to sleep regardless of TWI state and eliminating the need to call slaveTransactionOpen() at all?

The second thing, if it works, seems like a win.

MX682X, 2022-09-11:
For making it save. the idle power saving is just a plus

ObviousInRetrospect, 2022-09-11:
it might actually make sleep safe without the check - I've tested without the call to slaveTransactionOpen in idle and it seemed to work fine

[42 more replies]

closed 2022-11-02 by SpenceKonde