SpenceKonde/DxCore #329: PlatformIO problem with setting

mikrocoder, 2022-08-19:
Hello,

do you know anything about PlatformIO? Somehow there is still one setting wrong.

I can compile and transfer a blinksketch but the led does not blink. What is wrong?

This is an AVR128DB64 with bootloader to which a FTDI232RL is connected and works with the Arduino IDE and your DxCore. It is flashed via USB > ComPort.

What is wrong set in PlatformIO?

``````cpp
#include <Arduino.h>

void setup() {
  pinMode(PIN_PB4, OUTPUT);
}

void loop() {
  digitalWrite(PIN_PB4, HIGH);
  delay(200);
  digitalWrite(PIN_PB4, LOW);
  delay(800);
}
``````

platformio.ini
``````
[platformio]
default_envs = Upload_USB

[env]
platform = atmelmegaavr
framework = arduino
board = AVR128DB64
board_build.variant = 64pin-standard
board_build.f_cpu = 16000000L

[env:Upload_USB]
upload_protocol = arduino
upload_port = COM[5]
upload_speed = 115200
upload_flags = 
``````

Output:
```````
Processing Upload_USB (platform: atmelmegaavr; framework: arduino; board: AVR128DB64)
-------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/AVR128DB64.html
PLATFORM: Atmel megaAVR (1.6.0) > AVR128DB64     
HARDWARE: AVR128DB64 16MHz, 16KB RAM, 128KB Flash
PACKAGES:
 - framework-arduino-megaavr-dxcore @ 1.4.10     
 - tool-avrdude-megaavr @ 3.60300.220118 (6.3.0) 
 - toolchain-atmelavr @ 3.70300.220127 (7.3.0)   
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 17 compatible libraries
Scanning dependencies...
Dependency Graph
|-- SPI @ 1.0.2
|-- Wire @ 2.0.6        
Building in release mode
Checking size .pio\build\Upload_USB\firmware.elf
Advanced Memory Usage is available via "PlatformIO Home > Project Inspect"
RAM:   [          ]   0.0% (used 4 bytes from 16384 bytes)   
Flash: [          ]   0.7% (used 958 bytes from 131072 bytes)
Configuring upload protocol...
AVAILABLE: arduino
CURRENT: upload_protocol = arduino
Looking for upload port...
Auto-detected: COM5
Uploading .pio\build\Upload_USB\firmware.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e970b (probably avr128db64)
avrdude: reading input file ".pio\build\Upload_USB\firmware.hex"
avrdude: writing flash (1470 bytes):

Writing | ################################################## | 100% 0.09s

avrdude: 1470 bytes of flash written
avrdude: verifying flash memory against .pio\build\Upload_USB\firmware.hex:
avrdude: load data flash data from input file .pio\build\Upload_USB\firmware.hex:
avrdude: input file .pio\build\Upload_USB\firmware.hex contains 1470 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.07s

avrdude: verifying ...
avrdude: 1470 bytes of flash verified

avrdude: safemode: Fuses OK (E:FF, H:FF, L:FF)

avrdude done.  Thank you.
``````

MCUdude, 2022-08-19:
Workout looking into it, it's probably macro that has not been set in the PlatformIO build.

DxCore's boards.txt adds various macros as build flags (-DSOMEVAL=1), and this has to be replicated in the PlatformIO build configuration. @brunob45 was the one that added DxCore support to PlatformIO, but the implementation has to be kept up to date for it to work with newer DxCore releases.

mikrocoder, 2022-08-19:
Thanks for your info. Is brunob45 here on GitHub or on the PlatformIO forum? Basically I understand you that I can't use the DxCore with PlatformIO yet, because nothing works yet.

Is there a way to flash using Atmel-ICE and still use the DxCore Package for programming? Then I also have to do without the bootloader.

SpenceKonde, 2022-08-19:
No you can use platform io and dx core (not the github head, but the last
release). I just knno nothing about platform io and I'm already working on
things 16 hours a day 7 days a week and falling further and further behind.
So I'm not going 6o take on learning a new ide. But it's totally possible
and somewhere I habe an open issue requesting help from people with the
necessary knowledge to write guides for alternative ides because I get tons
of questions about them that I can't answer. The only ide I know is arduino

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

MCUdude, 2022-08-19:
> But it's totally possible
and somewhere I habe an open issue requesting help from people with the
necessary knowledge to write guides for alternative ides because I get tons
of questions about them that I can't answer. The only ide I know is arduino

My time is quite limited, but I was _pretty_ deep into the PlatformIO build system a while ago. You don't _have_ to learn how to use it, but you should take into account _how_ it gathers all the necessary bits and pieces from the platformio.ini project file before it starts the build. I also have various macros (`-DSOMETHING`) in the MegaCoreX boards.txt file, but I always try to have a fallback state in the core, in case macros aren't explicitly defined at compile time.

brunob45, 2022-08-20:
@MCUdude Thanks for the credits, but the Platformio team has taken over the porting of DxCore to Platformio IDE.

As of right now, the version 1.4.10 is officially released: https://registry.platformio.org/tools/platformio/framework-arduino-megaavr-dxcore

I have made a blink sketch for the AVR128DA48 [here](https://github.com/brunob45/dev-AVR-Dx). It should work the same for a DB-series, but I can't test it as I only have DA-series chips.

mikrocoder, 2022-08-20:
Hello,

Maybe you can help me.

PlatformIO has only been freshly installed for a few days. The DxCore package is also up to date on 1.4.10 according to the output.

Now I had to find out how to install 'pymcuprog'. Okay, it works. Error message about this is gone. 

However, it still does not work. A socket.py is supposedly not found. But it is definitely present in the path. What do I have to do?
``````
Uploading .pio\build\Upload_USB\firmware.hex
Traceback (most recent call last):
  File "C:\Users\Worker\.platformio\python3\lib\runpy.py", line 197, in _run_module_as_main
    return _run_code(code, main_globals, None,
  file "C:\Users\Worker\.platformio\python3\lib\runpy.py", line 87, in _run_code
    exec(code, run_globals)
  File "C:\Users\Worker\AppData\Local\Programs\Python\Python310\Scripts\pymcuprog.exe\__main__.py", line 4, in <module>
  File "C:\Users\Worker\AppData\Local\Programs\Python\Python310\lib\site-packages\pymcuprog\pymcuprog.py", line 12, in <module>
    from logging.config import dictConfig
  File "C:\Users\Worker\.platformio\python3\lib\logging\config.py", line 30, in <module>
    import logging.handlers
  File "C:\Users\Worker\.platformio\python3\lib\logging\handlers.py", line 26, in <module>
    import logging, socket, os, pickle, struct, time, re
  File "C:\Users\Worker\.platformio\python3\lib\socket.py", line 51, in <module>
    import _socket
ImportError: DLL load failed while importing _socket: The specified module was not found.
*** [upload] Error 1
``````
Also, is this true for me at all? I want to flash via USB > FTDI232RL > USART bootloader > µC. Is the upload protocol the right one for me? In the Arduino IDE it all works.
``````
[platformio]
default_envs = Upload_USB

[env]
platform = atmelmegaavr
framework = arduino
board = AVR128DB64
;board_build.variant = 64pin-standard
;board_build.f_cpu = 160000L

[env:Upload_USB]
upload_protocol = custom
upload_command = pymcuprog erase && pymcuprog write -f $SOURCE
``````

Because of f_cpu. 16MHz are default. okay. But how does PlatformIO know how and with what I clock the controller? For me this is too much default and too much hidden settings.

mikrocoder, 2022-08-20:
The solution is here: https://github.com/platformio/platform-atmelmegaavr/issues/41
This setting works.

SpenceKonde, 2022-08-21:
I will note  that this is in essence a duplicate of #319. A lot of our users are on platformio, about which I know about as much as my cat knows about AVR assembly (which is to say, not very much - I mean, she can't even name a single instruction that got timing enhancements in AVRxt versus AVRe+.

They deserve better docs. But right now the head doesn't compile blink so... priorities are priorities

closed 2022-08-21 by SpenceKonde

mikrocoder, 2022-08-21:
Hallo brunob45,

can you do me a favor? What settings do I need so that I can flash like in the Arduino IDE. What is hidden somewhere default and what you have to change yourself?  I hope this is not too much to ask.

[image]

mikrocoder, 2022-08-21:
@ Spence:
I think we have different views. #319 is a documentation thread for me. It should stay as clean as possible. You wanted an information section.

This is a thread for me to ask stupid questions.

brunob45, 2022-08-22:
@mikrocoder I will answer to the best of my knowledge. I did some work to port DxCore to PlatformIO a while back, but I haven't touch this since.

The config you have can be translated as 
```ini
[env:avrdb]
platform = atmelmegaavr
board = AVR128DB64
framework = arduino

; Set clock speed & source - default is 24MHz internal
board_build.f_cpu = 16000000L
board_hardware.oscillator = crystal

; Set millis timer source - default is B2
board_hardware.millistimer = B2

; Set BOD level (fixed 32Hz) - default is Disabled
board_hardware.bod = 2.85v

; Set reset pin function - default is reset
board_hardware.rstpin = reset

; Cannot change startup time - default is 32ms

; Set bootloader serial port - default is no_bootloader
board_hardware.uart = ser2_alt

; Set MVIO - default is no
board_hardware.mvio_enable = no

build_unflags =
    -DTWI_MORS_SINGLE ; Unset 1x Wire (enabled by default)
    ; -DCORE_ATTACH_ALL ; Uncomment to select another attach method

build_flags = 
    -DTWI_MORS_BOTH ; Set 2x Wire
    ; -DCORE_ATTACH_NONE ; Uncomment to select this attach method
    ; -DCORE_ATTACH_OLD ; Uncomment to select this attach method

upload_protocol = pkobn_updi
; Or try the other protocols: https://github.com/MCUdude/MegaCoreX/blob/master/PlatformIO.md#upload_protocol
; If none of the above work, use
; upload_protocol = custom
; upload_command = pymcuprog erase && pymcuprog write -f $SOURCE

; run the following command to set fuses
; pio run -t fuses
; run the following command to set fuses + burn bootloader
; pio run -t bootloader
```

I've made a pull request (platformio/platform-atmelmegaavr#48) to improve DxCore support (added bootloader selection and millis timer parameter in the ini file), you can try by using the platform
```ini
platform = https://github.com/brunob45/platform-atmelavrdx#dev/avrea
```

mikrocoder, 2022-08-22:
Hi,

Thanks for the help. I just had to change the upload setting. I flash with USB to USART (FTDI232RL) connector onboard.

``````
build_flags = 
    -DTWI_MORS_BOTH ; Set 2x Wire
    -DUSING_OPTIBOOT
    ; -DCORE_ATTACH_NONE ; Uncomment to select this attach method
    ; -DCORE_ATTACH_OLD ; Uncomment to select this attach method

upload_protocol = arduino
upload_speed = 115200
``````
Therefore I continue with USB connection for now.

But what I already noticed and now still not working is the 16MHz crystal. When I remove the crystal, the led continues to blink. This means that the internal clock is still active. If I switch to 'external' nothing flashes. So far okay. With 'internal' and 'crystal' there must be an error in the detail.

The PlatformIO universe is very large. :wink: In which file are all these settings? Where are they defined, what is default and what is optional? For example, the clock source must somehow lead to a register setting in the end so that it works. Then I could look once myself.

brunob45, 2022-08-22:
@mikrocoder 
The `board_hardware.oscillator` setting actually defines the `CLOCK_SOURCE` define in the background. The values are:
| `board_hardware.oscillator` | `-DCLOCK_SOURCE=` |
|-------------------------------|-------------------------|
| internal | 0 |
| crystal | 1 |
| external | 2 |

You can use `build_unflags = -DCLOCK_SOURCE` and `build_flags=-DCLOCK_SOURCE=1` to select the crystal.
Otherwise, I made the correction in my branch
```ini
platform = https://github.com/brunob45/platform-atmelavrdx#dev/avrea
```

mikrocoder, 2022-08-22:
Hello,

okay, thanks, helps a little.
But with this it does not compile.
``````
C:\Users\Worker\.platformio\packages\framework-arduino-megaavr-dxcore\cores\dxcore\wiring.c:37:4: error: #error "CLOCK_SOURCE not defined. Must be 0 for internal, 1 for crystal, or 2 for external clock"
   #error "CLOCK_SOURCE not defined. Must be 0 for internal, 1 for crystal, or 2 for external clock"
    ^~~~~
*** [.pio\build\AVR128DB\FrameworkArduino\wiring.c.o] Error 1
``````

brunob45, 2022-08-22:
Do you have this in the ini file? 
```ini
build_unflags =
    -DCLOCK_SOURCE
    -DTWI_MORS_SINGLE ; Unset 1x Wire (enabled by DEFAULT) 

build_flags =
    -DUSING_OPTIBOOT
    -DCLOCK_SOURCE=1 ; crystal
    -DTWI_MORS_BOTH ; Set 2x Wire
 
```

mikrocoder, 2022-08-22:
Hello,

thank you very much. Now it works. :thumbsup: :grinning:
One must always invalidate the exact parameter before rewriting it?

brunob45, 2022-08-22:
I didn't think so, but apparently yes 😅