SpenceKonde/DxCore #330: Update to Avrdude 7.0

MCUdude, 2022-08-19:
Avrdude 7.0 is a big improvement over 6.3 and brings lots of new features and bug fixes. Arduino now provides "flavored builds" with statically linked libraries, so boards like curiosity Nano will work on _any_ OS. You can pull the binaries straight from Github.

Here are a few good ones:
* Support for SerialUPDI. It would be interesting to hear what you think of the implementation.
* Support for arbitrary board rates on macOS. Now 230400 baud is not the upper limit anymore.
* Support for lots of new and fast baud rates for the jtag2updi programmer
* WinUSB driver support for Windows. This means that there's no need for Zadig anymore. Programmers like USBasp and USBtinyISP will _just work_ with Windows.
* Support for the Micronucleus bootloader
* A bunch of new targets supported (including AVR-DD and AVR-EA)

https://github.com/arduino/avrdude-packing

MCUdude, 2022-08-20:
For reference, here's my boards manager file:

https://github.com/MCUdude/MegaCoreX/blob/1d4bebc755392f0e7fb2b5749adb5645c728ab92/package_MCUdude_MegaCoreX_index.json#L649-L698

mikrocoder, 2022-08-20:
Hello,

but here again half of the information is missing. Where should Windows users get the 7.0 binary from? :wink:  I know the new version only from the current avr-gcc from ZakKemble. From its package I have taken me the files.
https://blog.zakkemble.net/avr-gcc-builds/

SpenceKonde, 2022-08-21:
Avrdude 7 will not be in 1.5.0, but will be added in a 1.5.x version - the time and effort of testing a new version of an upload tool that I am deemphasizing because of its exemplification of the kitchenskinkism antipattern, and which I'd love to phase out entirely is impossible to justify until a working release with full DD support and the latest hardware serial enhancement is available and peoele aren't screaming that it's broken. ATTinyCore 2.0.0 however will most certainly want this, as it does not use any other upload tool.

MCUdude, 2022-08-21:
> but here again half of the information is missing. Where should Windows users get the 7.0 binary from? 😉

Eh, no? I provided a link to the repository where all major OSes' binaries are hosted.

But here you go:
https://github.com/arduino/avrdude-packing/releases/tag/7.0-arduino.3

mikrocoder, 2022-08-21:
In the first link it does not go on for me.
The further path /release/tag/... I do not see. Your last link works. Thank you.

mcuee, 2022-10-30:
For those who are more adventurous, you can try [the Windows binary](https://github.com/arduino/avrdude-packing/files/9889835/avrdude-main-test-libusb-compat.zip) mentioned in this thread. I have tested it with DxCore, megaTinyCore and MegaCoreX. It is based on avrdude git main (with many fixes over 7.0 release).
* https://github.com/arduino/avrdude-packing/issues/15#issuecomment-1295147283

I have also built Linux and macOS binaries here (32/64bit Intel/AMD Linux, 32bit/64bit ARM Linux, macOS x86_64).
https://github.com/mcuee/avrdude-packing/actions/runs/3354027563

mcuee, 2022-10-30:
You can also try the binaries here. I need to use the 64bit Linux binary under 64bit Linux (no 32bit compatibility layer installed).
https://github.com/mcuee/avrdude-packing/actions/runs/3354027563

More discussions here.
* https://github.com/avrdudes/avrdude/issues/1162

mcuee, 2022-11-25:
I have published avrdude git main snapshot binaries here for those who are interested to try. git main will lead to 7.1 release.
https://github.com/mcuee/avrdude-packing/releases

pcfreak1201, 2023-01-10:
Now avrdude 7.1 is released (https://github.com/avrdudes/avrdude/releases/tag/v7.1) and on linux (Mint/Ubuntu 64bit) it compiles flawlessly.

mcuee, 2023-01-10:
For DxCore, it is probably better to wait a bit for the release from https://github.com/arduino/avrdude-packing project.

mcuee, 2023-01-10:
> For DxCore, it is probably better to wait a bit for the release from https://github.com/arduino/avrdude-packing project.

@MCUdude and @SpenceKonde 

Here it is.
https://github.com/arduino/avrdude-packing/releases/tag/7.1-arduino.1

mcuee, 2023-01-10:
@SpenceKonde 

Some changes may be required before you switched to avrdude 7.1 release.

Reference:
https://github.com/MCUdude/MegaCoreX/commit/09f3a6e04f6ab9b83063260bb43053c122c7e512

SpenceKonde, 2023-01-14:
Note - my current plan here is to use AVRdude 7.x for non-SerialUPDI uploads only, leaving me with control over the serialUPDI portion. We need to keep the python environment *anyway* so that several future planned tools i'm working on can be supported, namely a cleanup pass on the exported assembly listings to get rid of the avr-objdump's propensity to express locatioms in the data space as offsets from landmarks in the program space and vice-versa (which is at best useless), and to reformat the map to make it human readable (truncating long function names is the big one here - because the long synthetic function names that end up in the memory map deform the column structure and make the whole mess unreadable to humans, and correct the inconsistent column break marks, which made it unreadable to most generic parsers, making it hard to import into any civilized tool to further analyze.

mikrocoder, 2023-01-15:
> > For DxCore, it is probably better to wait a bit for the release from https://github.com/arduino/avrdude-packing project.
> 
> @MCUdude and @SpenceKonde
> 
> Here it is. https://github.com/arduino/avrdude-packing/releases/tag/7.1-arduino.1

Hi,

What is different about the "Arduino package" again compared to the Official Release? 
https://github.com/avrdudes/avrdude/releases

MCUdude, 2023-01-15:
> What is different about the "Arduino package" again compared to the Official Release?
https://github.com/avrdudes/avrdude/releases

"Official" binaries are built with dynamically linked libraries. This means that the computer needs to have a set of libraries installed for Avrdude to work. The Arduino release uses static linkage, libraries are bundled and built into the Avrdude binary. Great if you just want Avrdude to work on a computer, and you don't really care about libraries, like for instance when a user installed a 3rd party Arduino core that uses Avrdude.

mcuee, 2023-03-16:
Binary release from Arduino avrdude-packing project.
https://github.com/arduino/avrdude-packing/releases/tag/7.1-arduino.1

As for platform defintions, minor changes will be required, for example, Curiosity Nano programmer is now called pkobn_updi.
Ref: https://github.com/MCUdude/MegaCoreX/blob/master/megaavr/programmers.txt

SpenceKonde, 2023-06-24:
As soon as a 7.2 release is available we will upgrade to v 7.2.

mcuee, 2023-07-20:
avrdude 7.2 has been released.
https://github.com/avrdudes/avrdude/releases/tag/v7.2

For Windows you can use the official binary above.

For Linux and macOS, you may want to test using my build.
https://github.com/mcuee/avrdude/releases/tag/v7.2

Or you may want to wait for the Arduino-packing project.
https://github.com/arduino/avrdude-packing

mcuee, 2023-10-05:
Arduino avrdude-packing project has released their version of Arduino 7.2, which should be suitable for the usage here.
https://github.com/arduino/avrdude-packing/releases/tag/7.2-arduino.1

mcuee, 2024-02-16:
avrdude 7.3 has been released.
https://github.com/avrdudes/avrdude/releases/tag/v7.3

MCUdude, 2024-10-10:
Avrdude v8.0 has been released, and Arduino has been kind enough to provide statically built binaries for us, which makes it a breeze to use with the boards manager.

MCUdude, 2026-03-26:
Now that there is some activity in this repo, you should consider updating Avrdude in your next boards manager release. Avrdude 8.1 is the latest and greatest, and statically linked binaries for all platforms are available here:
https://github.com/avrdudes/avrdude/releases/tag/v8.1

With a few tweaks in your platform.txt you can ditch your custom avrdude.conf file, and you can also get rid of your custom SerialUPDI uploading tool that where you also need Python. Avrdude is self-contained, has excellent SerialUPDI support, is under active development and has very few known bugs. There is absolutely no reason to stick with 6.3 at this point.

MCUdude, 2026-04-08:
@SpenceKonde It's great to see that you'll be using Avrdude 8.1 in the next DxCore release! Here's a few tips:

* I'd recommend you to remove this repo's avrdude.conf and instead use the one that's bundled with Avrdude 8.1, just like I do with MegaCoreX. Then you don't need to make sure the avrdude.conf file is up to date if you ever update to a newer Avrdude version.
* You should consider going through the programmers.txt file and add new UPDI programmers that Avrdude 8.1 supports. Most noticeably the PICkit 5, PICkit Basic and SerialUPDI. I don't see a reason why it would be necessary to bundle Python and use `prog.py` when Avrdude can do everything prog.py can and more.
  - The Xplained Mini UPDI can be removed, as it doesn't support any of the Dx, Ex or Sx chips.
* FYI, the next version of Avrdude will be released sometime this summer, and will have support for the new AVR-LA series.

MCUdude, 2026-04-18:
SerialUPDI. I don't get why you won't to use Avrdude for this. If Avrdude doesn't have what takes to be usable for SerialUPDI programming, we'd like to know what should be improved.