SpenceKonde/DxCore #352: Error Compiling for board AVR DB Series (Optiboot)

cchimick, 2022-10-01:
Designed my project around the arduino mega. Switched to the AVR128DB64 due to supply issues. I believe I successfully uploaded the bootloader. When compiling the sketch in the Arduino IDE, I'm getting the error message:

In file included from C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2Card.h:26:0,
                 from C:\Program Files (x86)\Arduino\libraries\SD\src/utility/SdFat.h:29,
                 from C:\Program Files (x86)\Arduino\libraries\SD\src/SD.h:20,
                 from C:\Users\chris\Documents\GitHub\PipeMonitor\Temp_Logger\Temp_Logger.ino:4:
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:438:5: error: 'DDRD' was not declared in this scope
   {&DDRD, &PIND, &PORTD, 0},  // D0  0
     ^~~~
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:438:5: note: suggested alternative: 'VDD'
   {&DDRD, &PIND, &PORTD, 0},  // D0  0
     ^~~~
     VDD
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:438:12: error: 'PIND' was not declared in this scope
   {&DDRD, &PIND, &PORTD, 0},  // D0  0
            ^~~~
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:438:12: note: suggested alternative: 'PIN0'
   {&DDRD, &PIND, &PORTD, 0},  // D0  0
            ^~~~
            PIN0
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:439:5: error: 'DDRD' was not declared in this scope
   {&DDRD, &PIND, &PORTD, 1},  // D1  1
     ^~~~
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:439:5: note: suggested alternative: 'VDD'
   {&DDRD, &PIND, &PORTD, 1},  // D1  1
     ^~~~
     VDD
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:439:12: error: 'PIND' was not declared in this scope
   {&DDRD, &PIND, &PORTD, 1},  // D1  1
            ^~~~
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:439:12: note: suggested alternative: 'PIN0'
   {&DDRD, &PIND, &PORTD, 1},  // D1  1
            ^~~~
            PIN0
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:440:5: error: 'DDRD' was not declared in this scope
   {&DDRD, &PIND, &PORTD, 2},  // D2  2
     ^~~~
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:440:5: note: suggested alternative: 'VDD'
   {&DDRD, &PIND, &PORTD, 2},  // D2  2
     ^~~~
     VDD
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:440:12: error: 'PIND' was not declared in this scope
   {&DDRD, &PIND, &PORTD, 2},  // D2  2
            ^~~~
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:440:12: note: suggested alternative: 'PIN0'
   {&DDRD, &PIND, &PORTD, 2},  // D2  2
            ^~~~
            PIN0
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:441:5: error: 'DDRD' was not declared in this scope
   {&DDRD, &PIND, &PORTD, 3},  // D3  3
     ^~~~
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:441:5: note: suggested alternative: 'VDD'
   {&DDRD, &PIND, &PORTD, 3},  // D3  3
     ^~~~
     VDD
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:441:12: error: 'PIND' was not declared in this scope
   {&DDRD, &PIND, &PORTD, 3},  // D3  3
            ^~~~
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:441:12: note: suggested alternative: 'PIN0'
   {&DDRD, &PIND, &PORTD, 3},  // D3  3
            ^~~~
            PIN0
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:442:5: error: 'DDRD' was not declared in this scope
   {&DDRD, &PIND, &PORTD, 4},  // D4  4
     ^~~~
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:442:5: note: suggested alternative: 'VDD'
   {&DDRD, &PIND, &PORTD, 4},  // D4  4
     ^~~~
     VDD
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:442:12: error: 'PIND' was not declared in this scope
[... 193 lines cut ...]
     ^~~~
     DD4
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:456:12: error: 'PINC' was not declared in this scope
   {&DDRC, &PINC, &PORTC, 4},  // C4 18
            ^~~~
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:456:12: note: suggested alternative: 'PIN0'
   {&DDRC, &PINC, &PORTC, 4},  // C4 18
            ^~~~
            PIN0
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:457:5: error: 'DDRC' was not declared in this scope
   {&DDRC, &PINC, &PORTC, 5}   // C5 19
     ^~~~
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:457:5: note: suggested alternative: 'DD4'
   {&DDRC, &PINC, &PORTC, 5}   // C5 19
     ^~~~
     DD4
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:457:12: error: 'PINC' was not declared in this scope
   {&DDRC, &PINC, &PORTC, 5}   // C5 19
            ^~~~
C:\Program Files (x86)\Arduino\libraries\SD\src/utility/Sd2PinMap.h:457:12: note: suggested alternative: 'PIN0'
   {&DDRC, &PINC, &PORTC, 5}   // C5 19
            ^~~~
            PIN0
exit status 1
Error compiling for board AVR DB-series (Optiboot).

pcfreak1201, 2022-10-01:
Well, the cores cover/translate the internal I/O to the Arduino-Std-Functions. But you use a SD-Lib, which directly uses the internal registers of the - in this case - classic AVRs. That doesn't work. How the new generation works do you find here: https://github.com/SpenceKonde/DxCore/blob/master/megaavr/extras/Ref_Digital.md
So, you can change the lib by your own, or serarch for newer libs. Here are some more information:
https://github.com/SpenceKonde/megaTinyCore/blob/571da180506039903a7a3a2852cd78ec3b09a7c5/megaavr/extras/LibraryCompatibility.md

pcfreak1201, 2022-10-01:
Have a look at https://github.com/greiman/SdFat - looks like it uses fastIO, that could work...

SpenceKonde, 2022-10-01:
SD library doesn't have the fix (which is very straightforward) in the version you have installed - I thought I put in a PR to fix this - the fix is trivial. They specifically test for the 4809 (because they make official boards based on it), when they should test if __AVR_ARCH__==102 || ___AVR_ARCH__==103 || __AVR_ARCH == 104

Ugh, do I have to also add duplicated copies of the libraries as SD_megaTinyCore and SD_DxCore like I already do on servo,so that people can use my servo supplied with the core (which, again works) instead of the version that gets installed through the library manager (which does not work).

I do don't I....

annoying

pcfreak1201, 2022-10-01:
Well, I tried yor way first, by adding 
`#elif defined(__AVR_DA__) || defined(__AVR_DB__) || defined(__AVR_DD__)`
but doesn't work (may be too much if elif else ...?) on my system - because I saw the changes / PR from @MCUdude
but his updates from commit [9bb121c](https://github.com/arduino-libraries/SD/commit/9bb121c689b0980f5fbb71eea4c6e865e8822b07) aren't still in the actual version?!
So I tried your way from the MegaTinyCore and copied the folder and that worked. So as first aid for the OP, looks OK for me.

cchimick, 2022-10-01:
Sorry for the trouble guys.  Everything worked with the mega but since ive switched to the AVR128db64, I'm way beyond my depth

pcfreak1201, 2022-10-01:
That's normal when you change the controller ;-)
Not in your Arduino/library-folder but in 
`.arduino15/packages/DxCore/hardware/megaavr/1.4.10/libraries`

SpenceKonde, 2022-10-28:
Closing this as SD library is now in the master branch (granted the master branch is totally hosed in numerous ways)

In order to use it, though, I think you wont be able to have an SD library installed in libraries. I can't do the trick I did with Servo, because a million other libraries include SD.h, including the other ones you likely want to use

closed 2022-11-01 by SpenceKonde