SpenceKonde/DxCore #354: Optiboot for AVR-DD

MCUdude, 2022-10-24:
It's probably better to create a separate issue, so we can keep the discussion on point.

When helping out with the [Avrdude project](https://github.com/avrdudes/avrdude) I found out that the existing, pre-compiled bootloaders bundled to DxCore didn't work properly with the new AVR-DD series. I'm able to establish a connection with the target, but I can't read and write to EEPROM or flash. The bootloader I'm using is `optiboot_dx64_ser0_alt_8sec.hex`

The errors you're seeing below are outputted from the development version of Avrdude, but I'm getting the same result when using Avrdude 6.3.

Have you @SpenceKonde tried to get Optiboot to work on the AVR-DD series?

The pre-compiled bootloader hex file I'm using is generated by a command line tool called `srec_cat`, and is not a "real" program, but is only used as a test to see if the bootloader is actually able to do its job. It's a file containing 32256 0xAA's, starting from address 512.

`srec_cat -generate 0x200 0x8000 -repeat_data 0xaa -o 0xaa_32256b.hex -Intel`

Here are the fuse settings:
```
$ ./avrdude -cpkobn_updi -pavr64dd32 -t

         Vtarget                      : 5.02 V
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.01s

avrdude: device signature = 0x1e961a (probably avr64dd32)

avrdude> read wdtcfg
>>> read wdtcfg 
0000  00                                                |.               |

avrdude> read bodcfg
>>> read bodcfg 
0000  00                                                |.               |

avrdude> read osccfg
>>> read osccfg 
0000  00                                                |.               |

avrdude> read tcd0cfg
>>> read tcd0cfg 
0000  00                                                |.               |

avrdude> read syscfg0
>>> read syscfg0 
0000  d8                                                |.               |

avrdude> read syscfg1
>>> read syscfg1 
0000  0c                                                |                |

avrdude> read codesize
>>> read codesize 
0000  00                                                |.               |

avrdude> read bootsize
>>> read bootsize 
0000  01                                                |.               |
```

I can establish communication with the bootloader:
```
$ ./avrdude -carduino -pavr64dd32 

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: device signature = 0x1e961a (probably avr64dd32)

avrdude done.  Thank you.
```

But I can't write anything to memory.

EEPROM:

```
$ ./avrdude -carduino -pavr64dd32 -Ueeprom:w:0x55_256b.hex

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: device signature = 0x1e961a (probably avr64dd32)
avrdude: reading input file 0x55_256b.hex for eeprom
         with 256 bytes in 1 section within [0, 0xff]
avrdude: writing 256 bytes eeprom ...

Writing |                                                    | 0% 0.00s ***failed;
 ***failed;
 ***failed;
Writing | #                                                  | 1% 0.00s ***failed;
 ***failed;
 ***failed;
Writing | #                                                  | 2% 0.00s ***failed;
 ***failed;
Writing | ##                                                 | 3% 0.00s ***failed;
 ***failed;
 ***failed;
Writing | ##                                                 | 4% 0.00s ***failed;
 ***failed;
Writing | ###                                                | 5% 0.00s ***failed;
 ***failed;
 ***failed;

...

Writing | ################################################   | 95% 0.00s ***failed;
 ***failed;
Writing | ################################################   | 96% 0.00s ***failed;
 ***failed;
 ***failed;
Writing | #################################################  | 97% 0.00s ***failed;
 ***failed;
Writing | #################################################  | 98% 0.00s ***failed;
 ***failed;
 ***failed;
Writing | ################################################## | 99% 0.00s ***failed;
 ***failed;
Writing | ################################################## | 100% 0.00s

avrdude: 256 bytes of eeprom written
avrdude: verifying eeprom memory against 0x55_256b.hex

Reading |                                                    | 0% 0.00savrdude error: unable to read byte at address 0x0000
avrdude error: read operation not supported for memory eeprom
Reading | ################################################## | 100% 0.00s

avrdude error: unable to read all of eeprom memory, rc=-2

avrdude done.  Thank you.
```

Flash:
```
$  ./avrdude -carduino -pavr64dd32 -P /dev/cu.usbserial-A50285BI -D -Uflash:w:blink_4808_0x200.hex:i

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: device signature = 0x1e961a (probably avr64dd32)
avrdude: reading input file blink_4808_0x200.hex for flash
         with 768 bytes in 1 section within [0x200, 0x4ff]
         using 2 pages and 256 pad bytes
avrdude: writing 768 bytes flash ...

Writing | #########################                          | 50% 0.00savrdude error: programmer is not responding
Writing | ################################################## | 100% 5.06s

 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;

...

 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
 ***failed;
Writing | ################################################## | 100% 0.15s

avrdude: 32256 bytes of flash written
avrdude: verifying flash memory against 0xaa_32256b.hex

Reading | ################################################## | 100% 3.09s

avrdude error: verification mismatch, first encountered at addr 0x0200
        device 0xff != input 0xaa
avrdude error: verification mismatch

avrdude done.  Thank you.

mcuee, 2022-10-28:
I tried the bootloader and it does not work either.

Then I tried the option without bootloader and now the board seems to be bricked. I will see how I can revive the board using PICKit 4 (HV UPDI).

It is my fault though as it seems to me both two options requires HV UPDI programmer.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude_git -c pkobn_updi -p avr64dd32 -v

avrdude_git: Version 7.0-20221026 (a101899)
             Copyright (c) Brian Dean, http://www.bdmicro.com/
             Copyright (c) Joerg Wunsch

             System wide configuration file is C:\work\avr\avrdude_test\avrdude_bin\avrdude.conf

             Using Port                    : usb
             Using Programmer              : pkobn_updi
avrdude_git: found CMSIS-DAP compliant device, using EDBG protocol
             AVR Part                      : AVR64DD32
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             Serial program mode           : yes
             Parallel program mode         : yes
             Memory Detail                 :

                                               Block Poll               Page                       Polled
               Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               fuse0       wdtcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse1       bodcfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse2       osccfg      0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse4       tcd0cfg     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse5       syscfg0     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse6       syscfg1     0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse7       codesize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuse8       bootsize    0     0     0    0 no          1    1      0     0     0 0x00 0x00
               fuses                   0     0     0    0 no          9   16      0     0     0 0x00 0x00
               lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
               tempsense               0     0     0    0 no          2    1      0     0     0 0x00 0x00
               signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
               prodsig                 0     0     0    0 no        125  125      0     0     0 0x00 0x00
               sernum                  0     0     0    0 no         16    1      0     0     0 0x00 0x00
               userrow     usersig     0     0     0    0 no         32   32      0     0     0 0x00 0x00
               data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
               eeprom                  0     0     0    0 no        256    1      0     0     0 0x00 0x00
               flash                   0     0     0    0 no      65536  512      0     0     0 0x00 0x00

             Programmer Type : JTAGICE3_UPDI
             Description     : Curiosity nano (nEDBG) in UPDI mode
             ICE HW version  : 0
             ICE FW version  : 1.23 (rel. 91)
             Serial number   : MC020019502HIP000484
             Vtarget         : 3.31 V
             PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude_git: bad response to AVR sign-on command: 0xa0
avrdude_git: retrying with external reset applied
avrdude_git: bad response to AVR sign-on command: 0xa0
avrdude_git: retrying with external reset applied
avrdude_git main() error: initialization failed, rc=-1
            - double check the connections and try again
            - use -B to set lower ISP clock frequency, e.g. -B 200kHz
            - use -F to override this check

avrdude_git done.  Thank you.
```

[image]

technoblogy, 2022-11-17:
I'm having a problem that may be related.

I've been using the bootloader to upload a program to an AVR128DB48 with the **AVR DB-series (Optiboot)** Board option (DxCore 1.4.10), and I've encountered the avrdude verification error after making minor modifications to the program. The previous version of the program still works fine.

I tried adjusting the program size by gradually increasing the size of a PROGMEM string and found the critical point was as follows:

    Sketch uses 64512 bytes (49%) of program storage space. Maximum is 130560 bytes.

uploads fine, but adding one character to the PROGMEM string seems to increase the size of the program by 2 bytes (!) and gives:

    Sketch uses 64514 bytes (49%) of program storage space. Maximum is 130560 bytes.
     ...
    avrdude: verification error, first mismatch at byte 0x0200
             0xff != 0x0c
    avrdude: verification error; content mismatch
    avrdude: verification error; content mismatch

SpenceKonde, 2022-11-18:
I found one bit of buried treasure.... 

I was testing for RAMPZ to decide whether to build with the support for the extended address, and whether to use lpm or elpm to access flash. **the 64k DA/DB parts define RAMPZ** The 64k DD's do not. 

This is wholly responsible for the difference in size between the 64k and 32k versions of optiboot. setting it to write the data by word makes virtually no difference.

SpenceKonde, 2022-11-19:
You guys are doing better than I am. I can't even get it enable the UART at all.  And another optiboot bug just came in where any write in the last page of the first half of memory on a 128k part triggers a verification error.

SpenceKonde, 2022-11-25:
I have this working correctly now. 

The last broken part was a misspelling in a preprocessor conditional and poor compile warning settings that caused me to not realize I was testing an undefined (hence 0) variable. But this never made it into the field unless using github versions. I think misuse of SPM Z+ may have been a factor, along with beliving the headers when it said it had a RAMPZ on 64 k DA/DB.

SpenceKonde, 2022-11-25:
fixed by
08536302aeb4f35cbad20d5ddc63192a579bb1a3

closed 2022-11-25 by SpenceKonde

MCUdude, 2022-11-25:
Great! I'll give it a try just to confirm that it will work properly with the soon-to-be-released Avrdude 7.1

mcuee, 2022-12-25:
This seems to work.

```
C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude -CC:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10/avrdude.conf -v -pavr64dd32 -ccuriosity_updi -Pusb -e -Ufuse0:w:0b00000000:m -Ufuse1:w:0x00:m -Ufuse5:w:0b11010000:m -Ufuse6:w:0b00001100:m -Ufuse8:w:0x01:m -Ufuse7:w:0x00:m -Uflash:w:C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10/bootloaders/hex/optiboot_64dd_ser0_alt3_all_8sec.hex:i 

avrdude: Version avrdude-main-test-libusb-compat
         Copyright (c) Brian Dean, http://www.bdmicro.com/
         Copyright (c) Joerg Wunsch

         System wide configuration file is C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10\avrdude.conf
         Using Port                    : usb
         Using Programmer              : curiosity_updi
avrdude: found CMSIS-DAP compliant device, using EDBG protocol
         AVR Part                      : AVR64DD32
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         Serial program mode           : yes
         Parallel program mode         : yes
         Memory Detail                 :

                                           Block Poll               Page                       Polled
           Memory Type Alias    Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- -------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature               0     0     0    0 no          3    1      0     0     0 0x00 0x00
           prodsig                 0     0     0    0 no        125  125      0     0     0 0x00 0x00
           fuses                   0     0     0    0 no          9   16      0     0     0 0x00 0x00
           fuse0                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse1                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse2                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse4                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse5                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse6                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse7                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           fuse8                   0     0     0    0 no          1    1      0     0     0 0x00 0x00
           lock                    0     0     0    0 no          4    1      0     0     0 0x00 0x00
           data                    0     0     0    0 no          0    1      0     0     0 0x00 0x00
           flash                   0     0     0    0 no      65536  512      0     0     0 0x00 0x00
           eeprom                  0     0     0    0 no        256   32      0     0     0 0x00 0x00

         Programmer Type : JTAGICE3_UPDI
         Description     : Microchip Curiosity in UPDI mode
         ICE HW version  : 0
         ICE FW version  : 1.25 (rel. 116)
         Serial number   : MC020019502HIP000639
         Vtarget         : 3.31 V
         PDI/UPDI clock Xmega/megaAVR : 100 kHz
avrdude: partial Family_ID returned: "AVR "
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e961a (probably avr64dd32)
erasing chip
avrdude: reading input file 0b00000000 for fuse0
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse0 ...
avrdude: 1 byte of fuse0 written
avrdude: verifying fuse0 memory against 0b00000000
avrdude: 1 byte of fuse0 verified
avrdude: reading input file 0x00 for fuse1
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse1 ...
avrdude: 1 byte of fuse1 written
avrdude: verifying fuse1 memory against 0x00
avrdude: 1 byte of fuse1 verified
avrdude: reading input file 0b11010000 for fuse5
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse5 ...
avrdude: 1 byte of fuse5 written
avrdude: verifying fuse5 memory against 0b11010000
avrdude: 1 byte of fuse5 verified
avrdude: reading input file 0b00001100 for fuse6
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse6 ...
avrdude: 1 byte of fuse6 written
avrdude: verifying fuse6 memory against 0b00001100
avrdude: 1 byte of fuse6 verified
avrdude: reading input file 0x01 for fuse8
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse8 ...
avrdude: 1 byte of fuse8 written
avrdude: verifying fuse8 memory against 0x01
avrdude: 1 byte of fuse8 verified
avrdude: reading input file 0x00 for fuse7
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse7 ...
avrdude: 1 byte of fuse7 written
avrdude: verifying fuse7 memory against 0x00
avrdude: 1 byte of fuse7 verified
avrdude: reading input file C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10/bootloaders/hex/optiboot_64dd_ser0_alt3_all_8sec.hex for flash
         with 488 bytes in 2 sections within [0, 0x1ff]
         using 1 page and 24 pad bytes
avrdude: writing 488 bytes flash ...

Writing | ################################################## | 100% 0.16s

avrdude: 488 bytes of flash written
avrdude: verifying flash memory against C:\Users\xiaof\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.4.10/bootloaders/hex/optiboot_64dd_ser0_alt3_all_8sec.hex

Reading | ################################################## | 100% 0.09s

avrdude: 488 bytes of flash verified

avrdude done.  Thank you.
```

mcuee, 2022-12-25:
Testing:
```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c arduino -P COM5 -p avr64dd32

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e961a (probably avr64dd32)

avrdude done.  Thank you.

```

mcuee, 2022-12-25:
Using git main version of avrdude. I need to unplug and plug to get the board into bootloader mode.

```
PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM5 -p avr64dd32  -xbootsize=512 -U '.\Blink_64dd32.ino.avr64dd32opti.24c0.m{build.highestcb}.wO.v151.hex'

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e961a (probably avr64dd32)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: reading input file .\Blink_64dd32.ino.avr64dd32opti.24c0.m{build.highestcb}.wO.v151.hex for flash
         with 816 bytes in 1 section within [0x200, 0x52f]
         using 2 pages and 208 pad bytes
avrdude: preparing flash input for device bootloader
avrdude: writing 816 bytes flash ...

Writing | ################################################## | 100% 0.16 s

avrdude: 816 bytes of flash written
avrdude: verifying flash memory against .\Blink_64dd32.ino.avr64dd32opti.24c0.m{build.highestcb}.wO.v151.hex

Reading | ################################################## | 100% 0.11 s

avrdude: 816 bytes of flash verified

avrdude done.  Thank you.

PS C:\work\avr\avrdude_test\avrdude_bin> .\avrdude -c urclock -P COM5 -p avr64dd32 -xshowall -xbootsize=512 -xstrict

avrdude: AVR device initialized and ready to accept instructions
0 2022-12-25 20.13 Blink_64dd32.ino.avr64dd32opti.24c0.m{build.highestcb}.wO.v151.hex 1328 store 64130 meta 78 boot 512 x0.0 ......... vector 0 (RESET) AVR64DD32

```