SpenceKonde/DxCore #355: reading fuses with prog.py

coburnw, 2022-10-26:
On tag 1.4.10, i needed to change one line of prog.py to allow reading fuses via updi.

inside the `if args.fuses_print` section of prog.py, change line [238](https://github.com/SpenceKonde/DxCore/blob/b0bb30f8cb703991d3fe97a3241091a221eab09b/megaavr/tools/prog.py#L238)  to

```
        run_pymcu_action(pymcu._action_read, backend,
                         memory=pymcu.MemoryNames.FUSES,
                         offset=0,
	                 bytes=0,
                         filename=None,
                         max_read_chunk=None if args.read_chunk <= 0 else args.read_chunk)
```
adding `max_read_chunk` to the list of existing `_action_read` parameters.

SpenceKonde, 2022-10-27:
@qbolsee I do believe this is your department.

SpenceKonde, 2022-11-09:
Simpler than that, we don't even need to worry about read chunks when reading just a few bytes at a time for fuses. That's for working around a bug in a native USB microcontroller used as a serial adapter that had problems with 64 byte packets of characters.

commit c36b2522 (2022-11-09): fix #355
closed 2022-11-09 by SpenceKonde