SpenceKonde/DxCore #360: Menu options wrong for AVR DD parts

mcuee, 2022-11-01:
From here:
* https://github.com/avrdudes/avrdude/pull/1140#issuecomment-1297214715

@SpenceKonde mentioned the following.
> Neither of those options is the one you should be using. The one you should be using is the Reset and UPDI option. That said option is not present is clearly a bug. I didn't name the menu options distinctly.
>
> I belive is what that section of board.txt should be.
``` 
#----------------------------------------#
# Reset pin as INPUT, UPDI as I/O        #
#________________________________________#
avrddopti.menu.resetupdipinsopti.resetupdi=Reset & UPDI
avrddopti.menu.resetupdipinsopti.resetupdi.bootloader.resetpinbits=11
avrddopti.menu.resetupdipinsopti.inputupdi=Input & UPDI (no reset to enter optiboot)
avrddopti.menu.resetupdipinsopti.inputupdi.bootloader.resetpinbits=10
avrddopti.menu.resetupdipinsopti.resetgpio=Reset & GPIO (HV pulse RST for UPDI)
avrddopti.menu.resetupdipinsopti.resetgpio.bootloader.resetpinbits=01
avrddopti.menu.resetupdipinsopti.inputgpio=Input & GPIO (no reset, HV pulse for UPDI)
avrddopti.menu.resetupdipinsopti.inputgpio.bootloader.resetpinbits=00
```

mcuee, 2022-11-01:
I understand that there is no currently working version of DxCore for the AVR DD. So this is documented here so that it can be fixed in the future.

SpenceKonde, 2022-11-01:
It already has been corrected

closed 2022-11-01 by SpenceKonde