SpenceKonde/DxCore #362: Serial wrong null terminator?

mikrocoder, 2022-11-07:
Hello,

there is a problem with serial reading. Something is wrong with the null terminator. I have prepared the following small test code. On all other Arduinos available to me I get exactly one line output when I type anything or even just press empty enter. With DxCore an additional line is output afterwards. Apparently there is still something in the ringbuffer which does not belong there. Is maybe the zero terminator set wrong?

```cpp
constexpr uint8_t BUFFER_SIZE {10};

struct ReceiveData {
  char buffer [BUFFER_SIZE + 1]; // + Null-Terminator
  size_t index {0};
};
ReceiveData rx;

void setup (void)
{
  Serial2.swap(1);    // PF4 TXD2 / PF5 RXD2
  Serial2.begin(9600); 
  Serial2.println(F("\nuC Reset ### ###"));
} 

void loop (void)
{
  if (readSerial(Serial2, rx, BUFFER_SIZE)) {
    Serial2.print(F("eingelesen: ")); Serial2.println(rx.buffer);
  }
}

// ****** Funktionen ****************************** //

bool readSerial (Stream &stream, auto &data, const uint8_t LENGTH)
{
  bool status {false};
 
  char c = stream.read();

  if (c == '\r' || c == '\n') {             // wenn CR oder LF eingelesen
    data.buffer[ data.index] = '\0';        // Terminierung
    data.index = 0;
    status = true;                          // Character fertig eingelesen
  }
  else if ( (c >= 32) && ( data.index < LENGTH) ) { // wenn noch Platz im Buffer ist
    data.buffer[data.index++] = c;          // Zeichen speichern und Index inkrementieren
  }

  return status;
}
```

closed 2022-11-07 by mikrocoder

SpenceKonde, 2022-11-09:
It's astounding that we managed to end up with two characters for newline... which are used in nearly every permutation by different devices. 

(as an aside, that has nothing to do with a null terminator - the null terminator is the 0x00 that marks the end of a string - there are only 255 characters that can be in a c string - a 0 instead marks the end of the string, which may be stored in a buffer larger than the string (sometimes much larger, so things like strlen checks the charaters, counting them as it does until it reaches a null (0) and that count up to there is what it returns. Nothing to do with the cr lf sequence that println() adds.)

mikrocoder, 2022-11-09:
How Serial works is clear to me. The problem was solved shortly after the question was asked. I had different terminal settings in 2 open IDE and had not noticed it. That's why I wondered about different behavior. That was all. Had cost me 2 days of cause search.  :-)