SpenceKonde/DxCore #366: ZeroCross zcd3()' redeclared as different kind of symbol (Help me out here!!)

SpenceKonde, 2022-11-12:
I have no idea what I did wrong. Only the AVRDD's have this problem. I don't know what I am doing that is triggering that error, can anyone who is better at C++ take a look and explain this to me?

MCUdude, 2022-11-15:
Could you post the actual error?

IMO, the way you've handled the DD series in the ZCD library is kind of messy...

SpenceKonde, 2022-11-16:
What the hell... it errors in the CI.... but it compiles fine locally. 
```
 /home/runner/.arduino15/packages/DxCore/hardware/megaavr/1.4.10/libraries/ZCD/src/ZCD.cpp:14:18: error: 'ZeroCross zcd3()' redeclared as different kind of symbol
     ZeroCross zcd3();
                    ^
  In file included from /home/runner/.arduino15/packages/DxCore/hardware/megaavr/1.4.10/libraries/ZCD/src/ZCD.cpp:1:0:
  /home/runner/.arduino15/packages/DxCore/hardware/megaavr/1.4.10/libraries/ZCD/src/ZCD.h:67:20: note: previous declaration 'ZeroCross zcd3'
     extern ZeroCross zcd3;
                      ^~~~
  
  ```
(note: the 1.4.10 is from the way the CI tests make sure they have the right toolchain version - they install the previous version of the core, then replace that with the current version from github)

I don't claim that the code doesn't look messy - it's a nasty knot of ifdefs. The objective was to minimize the flash and ram footprint. We know exactly what parts it will be used on; there's no point in passing all those arguments around and storing their values in ram, when they're known to be constant because the part family only has one ZCD... 

That's a real weakness of C++ on embedded systems - the optimizer can't optimize those away. myPeripheral -> STATUS is going to be lds, lds, ld, (5 words 8 clocks) not lds (2 words 3 clocks) or ldi ldi ld (3 words 4 clocks), even when there's only one of that peripheral and only one instance of the class. You can make the code compile to a smaller binary by removing constants passed to class constructors and instead directly specifying constants in the code. That sucks. In the case of ZCD, it was a simple enough library that it was practical to manually eliminate them and make the binary smaller and the code uglier. It's less feasible for something like the ACs. They are only slightly more complicated, but using ifdefs to reduce the number of member variables would be much uglier (unless you fix the pin settings there too).

mosqu-ito, 2022-11-17:
My 1.4.10 installation won't even compile an empty sketch for DD so it's hard for me give any real help here.  All I can do is gaze and see if anything very general rings a bell.  So this is just another of my wild stabs in the dark, likely worthless, but I do recall having problems using identifiers associated with classes whose synonymous constructors had empty parameter signatures.  Counterintuitively, those identifiers needed empty parentheses.  So, for what it's worth, maybe try extern ZeroCross zcd3();
???

Floessie, 2022-11-17:
... or try removing the `()` in `ZDC.cpp`.

mosqu-ito, 2022-11-17:
@Floessie makes more sense.  My approach makes it look like you're trying to export the constructor.

mosqu-ito, 2022-11-17:
I shouldn't speak for Floessie but I believe we're talking about line 14.

mosqu-ito, 2022-11-18:
Is silence your kinder, gentler way of letting us know that didn't help?

SpenceKonde, 2022-11-18:
It would be my way of saying I hadn't seen the responses. 

That appears to fix it! Thank you! I don't really understand the syntax, but I'm not arguing.

closed 2022-11-18 by SpenceKonde

Floessie, 2022-11-19:
@SpenceKonde Glad, it helped. This problem is called [the "most vexing parse"](https://en.wikipedia.org/wiki/Most_vexing_parse). Nowadays in C++11 this is usually solved by using `{}` instead of `()`, e.g.

```c++
MyClass my_object{};
```

But if you're like me and a fan of consistency, then it's better to leave out the `()` for classes with a [default constructor](https://en.cppreference.com/w/cpp/language/default_constructor). After all, one doesn't write `std::string some_string();` but simply `std::string some_string;`.

HTH,
Flössie