SpenceKonde/DxCore #374: optiboot_dx build instruction for Linux

mcuee, 2022-12-11:
The build instruction below works well for optiboot_dx (as well as optibootx for megaTinyCore and optiboot for ATTinyCore)
https://github.com/SpenceKonde/DxCore/blob/master/megaavr/bootloaders/optiboot_dx/README.md

It only talks about Windows. But it is not so difficult to adapt to Linux.

mcuee, 2022-12-11:
Here are simple steps I did to get it done under Ubuntu 20.04 Linux.

Take note I am not a programmer and you can probably improve the shell scripts to make them more robust.

1) Download Arduino 1.0.6 Linux 64 bit and extract it to the home directory `~/arduino-1.0.6`

2) Download the toolchain and extract the archive to the correct directory, merge the files if you use GUI file manager. The compiler binaries are now in `~/arduino-1.0.6/hardware/tools/avr/bin`
https://spencekondetoolchains.s3.amazonaws.com/avr-gcc-7.3.0-atmel3.6.1-azduino6-x86_64-pc-linux-gnu.tar.bz2
(other versions available as well from the parent directory which is an xml listing).

```
mcuee@UbuntuSwift3:~/arduino-1.0.6/hardware/tools/avr/bin$ avr-gcc -v
Using built-in specs.
Reading specs from /home/mcuee/avr_gcc/bin/../lib/gcc/avr/7.3.0/device-specs/specs-avr2
COLLECT_GCC=avr-gcc
COLLECT_LTO_WRAPPER=/home/mcuee/avr_gcc/bin/../libexec/gcc/avr/7.3.0/lto-wrapper
Target: avr
Configured with: ../gcc/configure --enable-fixed-point --enable-languages=c,c++ --prefix=/home/jenkins/workspace/avr-gcc-staging/label/debian7-x86_64/objdir --disable-nls --disable-libssp --disable-libada --disable-shared --with-avrlibc=yes --with-dwarf2 --disable-doc --target=avr
Thread model: single
gcc version 7.3.0 (GCC)
```

3) copy the optiboot_dx directory to the correct directory, `~/arduino-1.0.6/hardware/arduino/bootloaders/optiboot_dx`

4) modify the file omake accordingly
```
mcuee@UbuntuSwift3:~/arduino-1.0.6/hardware/arduino/bootloaders/optiboot_dx$ cat omake
#!/bin/sh
echo make OS=linux ENV=arduino $*
make OS=linux ENV=arduino $*
```

5) create he following two shell scripts based on the two batch file, basically replacing `call omake` with `./omake`.

```
mcuee@UbuntuSwift3:~/arduino-1.0.6/hardware/arduino/bootloaders/optiboot_dx$ head build_all_dd.sh 
#!/bin/sh
echo "Beginning AVR64DD (42 files). Start with ser0 (30 files)..."
./omake 64dd_ser0_extr
./omake 64dd_ser0_all_8sec
./omake 64dd_ser0_poronly_8sec
./omake 64dd_ser0_extronly
./omake 64dd_ser0_swronly_8sec
./omake 64dd_ser0_extr_8sec
./omake 64dd_ser0_alt1_extr
./omake 64dd_ser0_alt1_all_8sec

mcuee@UbuntuSwift3:~/arduino-1.0.6/hardware/arduino/bootloaders/optiboot_dx$ head build_all_dx.sh 
#!/bin/sh
echo "Beginning AVR128DA/AVR128DB with default serial pins (36 files)..."
./omake 128dx_ser0_extr
./omake 128dx_ser1_extr
./omake 128dx_ser2_extr
./omake 128dx_ser3_extr
./omake 128dx_ser4_extr
./omake 128dx_ser5_extr
./omake 128dx_ser0_all_8sec
./omake 128dx_ser1_all_8sec
```

6) Now you are ready to build optoboot_dx.

Example: single hex file bu8ild using omake.
```
mcuee@UbuntuSwift3:~/arduino-1.0.6/hardware/arduino/bootloaders/optiboot_dx$ ./omake 128dx_ser0_extr
make OS=linux ENV=arduino 128dx_ser0_extr
Using Compiler at: /home/mcuee/arduino-1.0.6/hardware/tools/avr/bin/avr-gcc
make -f  Makefile optiboot_128dx_ser0_extr.hex UARTTX=A0 TIMEOUT=1 LED=A7 ENTRYCOND_REQUIRE=0x34
make[1]: Entering directory '/home/mcuee/arduino-1.0.6/hardware/arduino/bootloaders/optiboot_dx'
Using Compiler at: /home/mcuee/arduino-1.0.6/hardware/tools/avr/bin/avr-gcc
../../../tools/avr/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax  -DWDTTIME=1  -DLED_START_FLASHES=3  -DLED=A7   -DUARTTX=A0 -DBAUD_RATE=115200        -Wl,-section-start=.text=0 -Wl,--section-start=.application=0x200 -Wl,--section-start=.spmtarg=0x1fa -Wl,--section-start=.version=0x1fe -Wl,--relax -nostartfiles -nostdlib  -mmcu=avr128da64 -o optiboot_128dx_ser0_extr.elf optiboot_dx.c
../../../tools/avr/bin/avr-size -A optiboot_128dx_ser0_extr.elf
optiboot_128dx_ser0_extr.elf  :
section          size      addr
.data               0   8404992
.text             494         0
.application       42       512
.spmtarg            4       506
.version            2       510
.comment           17         0
.debug_aranges     48         0
.debug_info      3102         0
.debug_abbrev     763         0
.debug_line       942         0
.debug_frame      156         0
.debug_str       1469         0
.debug_loc        739         0
.debug_ranges      80         0
Total            7858

"../../../tools/avr/bin/avr-objdump" -S optiboot_128dx_ser0_extr.elf > optiboot_128dx_ser0_extr.lst
../../../tools/avr/bin/avr-objcopy -j .text -j .data -j .version -j .spmtarg --set-section-flags .version=alloc,load --set-section-flags .spmtarg=alloc,load -O ihex optiboot_128dx_ser0_extr.elf optiboot_128dx_ser0_extr.hex
make[1]: Leaving directory '/home/mcuee/arduino-1.0.6/hardware/arduino/bootloaders/optiboot_dx'
```

Or you can use the two shell scripts in Step 5 to build all the hex files.

mcuee, 2022-12-11:
Final results.

`hexls` is a nice utility from urboot project.
https://github.com/stefanrueger/urboot/blob/main/src/hexls

```
mcuee@UbuntuSwift3:~/arduino-1.0.6/hardware/arduino/bootloaders/optiboot_dx$ ./hexls 
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser0_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser0_alt_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser0_alt_extr.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser0_alt_extr_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser0_alt_extronly.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser0_alt_poronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser0_alt_swronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser0_extr.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser0_extr_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser0_extronly.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser0_poronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser0_swronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser1_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser1_alt_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser1_alt_extr.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser1_alt_extr_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser1_alt_extronly.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser1_alt_poronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser1_alt_swronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser1_extr.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser1_extr_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser1_extronly.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser1_poronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser1_swronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser2_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser2_alt_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser2_alt_extr.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser2_alt_extr_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser2_alt_extronly.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser2_alt_poronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser2_alt_swronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser2_extr.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser2_extr_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser2_extronly.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser2_poronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser2_swronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser3_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser3_alt_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser3_alt_extr.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser3_alt_extr_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser3_alt_extronly.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser3_alt_poronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser3_alt_swronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser3_extr.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser3_extr_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser3_extronly.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser3_poronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser3_swronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser4_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser4_alt_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser4_alt_extr.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser4_alt_extr_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser4_alt_extronly.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser4_alt_poronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser4_alt_swronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser4_extr.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser4_extr_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser4_extronly.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser4_poronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser4_swronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser5_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser5_alt_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_128dx_ser5_alt_extr.hex
[... 244 lines cut ...]
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser4_alt_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser4_alt_extr.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser4_alt_extr_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser4_alt_extronly.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser4_alt_poronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser4_alt_swronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser4_extr.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser4_extr_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser4_extronly.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser4_poronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser4_swronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser5_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser5_alt_all_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser5_alt_extr.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser5_alt_extr_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser5_alt_extronly.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser5_alt_poronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser5_alt_swronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser5_extr.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser5_extr_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser5_extronly.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser5_poronly_8sec.hex
496 496 o26.1 -.s-.-r-- optiboot_64dx_ser5_swronly_8sec.hex

```

SpenceKonde, 2022-12-13:
Added some notes as I understand hotw it w ould work on linux.Closed wit 53d4af71ddf250784276e58cea150bd49b2513c0

Feel; free to create news if there are specific inaccuracuies or inadquacies, I do not perform any testing there.

mcuee, 2022-12-13:
Great. The latest version works perfectly.

Please help to create similar scripts for megaTinyCore and ATTinyCore when you got time. Thanks.

```
mcuee@UbuntuSwift3:~/arduino-1.0.6/hardware/arduino/bootloaders/optiboot_dx$ sh build_all_dx.sh 
...
...
Done with AVR DA/DB-series.
36 + 36 + 36 + 36 = 144 128 and 64k versions, and 30 + 24 = 54 for 32k
Hence there should b 144 +  54 = 198 hex files for AVR Dx.

mcuee@UbuntuSwift3:~/arduino-1.0.6/hardware/arduino/bootloaders/optiboot_dx$ sh build_all_dd.sh 
...
...
Done with AVR32DD14/AVR16DD14 There should be 48 files for 14-pin DD's 
Done with DD-series. There should be 42 + 42 + 24 + 24 = 132 hex files for DD-series parts
Combined with 198 from DA/DB the total should be 330 hex files.

```