SpenceKonde/DxCore #390: Architecture Incompatible with Avr Output

PierreSFU, 2023-01-30:
Hi Spence, thanks for the amazing repository's!

I am using your DxCore repository with the megaTinyCore, also using a makefile to compile the ATtiny1614 with the Blinky test. There is SPI.cpp being compiled alongside the rest of the .c code, for future use.

/home/ubuntu/.arduino15/packages/DxCore/tools/avr-gcc/7.3.0-atmel3.6.1-azduino6/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld: avr:103 architecture of input file `/tmp/cc6eQdBE.ltrans0.ltrans.o' is incompatible with avr output

I was hoping if you had any idea if this was on your end, so I could narrow it down. Sorry if I'm misusing the issues page, I'm fairly new to GitHub and this process in general.

SpenceKonde, 2023-01-31:
I think there's a problem with your makefile - is the command that invokes the linker specifying the right part?

PierreSFU, 2023-02-01:
Hi Spence, I'll assume it's on my end! If you have not encountered an issue like this before, it likely is. Thank you for your input!

closed 2023-02-01 by PierreSFU