SpenceKonde/DxCore #397: Fails to compile a LED blinking demo

unebonnevie, 2023-02-13:
Hi,

I am pretty sure that I am missing something, but when I tried build a simple led blinking sketch after installing DxCore, I get the below error.  The DxCore uses avr-g++ by default.  Attached is my DxCore config via the IDE's Tools menu.  Please advise!  Thanks!

`avr-g++: error: device-specs/specs-avr128db28: No such file or directory
exit status 1
Error compiling for board AVR DB-series (no bootloader).
`
[image]

SpenceKonde, 2023-02-13:
You somehow have a recent version of the core but an old version of the compiler toolchain. 

Is it a linux system twhere the IDE was installed through a package manager? They are known to have a huge number of problems depending on what package manager, and that configuration will never work, uninstall the package manager version and download the IDE from arduino.cc and install. Package manager maintainers have no clue about Arduino or embedded development (for example, they think that any version ofthe toolchain can compile for any chip of a given architecture, this is not true - you get devicespec errors. 

Is that a manual installation of DxCore? If so you did not follow the (complicated, unpleasant) process to update the toolchain. 

That error should not be possible on a working IDE where the core was installed through boardmanager. That's why we recommend board manager installs - they will automatically install the required toolchain version and any other tools.

SpenceKonde, 2023-02-13:
Closing (but will still see responses; this is not a defect in the core, it is a configuration problem or a bogus IDE. Since the issues list serves as a list of action items for me, I try to keep straight support inquiries off it.

closed 2023-02-13 by SpenceKonde

unebonnevie, 2023-02-13:
Thank you!  I went ahead and reinstalled the DxCore, and it went through with the compilation.