SpenceKonde/DxCore #398: Failing to flash the LED blink program to the AVR128DB28 (PDIP)

unebonnevie, 2023-02-13:
Hi,

So, I managed to install DxCore and compiled the LED blink sketch successfully, but when I try to flash, I get the below error.  I follow the wiring of the USB serial TX/RX, per the instructions at https://community.openenergymonitor.org/t/getting-started-with-the-avr-db-on-a-breadboard/20255.  (Image https://community.openenergymonitor.org/uploads/default/original/3X/3/b/3b6b6360d5c47b5b551207a84925fb1e16a311ee.jpeg).

I know that my board exists, because COM3 is active when I plug in my board.  Board config is attached at the bottom of this message.

Greatly appreciate any tips.   Thank you!

`Global variables use 4 bytes (0%) of dynamic memory, leaving 16380 bytes for local variables. Maximum is 16384 bytes.
C:\Users\auser\AppData\Local\Arduino15\packages\megaTinyCore\tools\python3\3.7.2-post1/python3 -u C:\Users\auser\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4/tools/prog.py -t uart -u COM3 -b 57600 -d avr128db28 --fuses 6:0b00001100 7:0x00 8:0x00 -fC:\Users\auser\AppData\Local\Temp\arduino_build_76367/AVR128DB_blink.ino.hex -a write -v 
SerialUPDI
UPDI programming for Arduino using a serial adapter
Based on pymcuprog, with significant modifications
By Quentin Bolsee and Spence Konde
Version 1.2.3 - Jan 2022
Using serial port COM3 at 57600 baud.
Target: avr128db28
Set fuses: ['6:0b00001100', '7:0x00', '8:0x00']
Action: write
File: C:\Users\auser\AppData\Local\Temp\arduino_build_76367/AVR128DB_blink.ino.hex
pymcuprog.programmer - INFO - Setting up programming session for 'avr128db28'
pymcuprog.deviceinfo.deviceinfo - INFO - Looking for device avr128db28
pymcuprog.serialupdi.physical - INFO - Opening port 'COM3' at '57600' baud
pymcuprog.serialupdi.link - INFO - STCS 08 to 0x03
pymcuprog.serialupdi.link - INFO - STCS 06 to 0x02
pymcuprog.serialupdi.link - INFO - LDCS from 0x00
pymcuprog.serialupdi.link - WARNING - UPDI init failed: Can't read CS register.
pymcuprog.serialupdi.physical - INFO - Sending double break
pymcuprog.serialupdi.physical - INFO - Double-break sent. Retrying.
pymcuprog.serialupdi.physical - INFO - Opening port 'COM3' at '57600' baud
pymcuprog.serialupdi.link - INFO - STCS 08 to 0x03
pymcuprog.serialupdi.link - INFO - STCS 06 to 0x02
pymcuprog.serialupdi.link - INFO - LDCS from 0x00
pymcuprog.serialupdi.link - WARNING - UPDI init failed: Can't read CS register.
Traceback (most recent call last):
  File "C:\Users\auser\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4/tools/prog.py", line 286, in <module>
    main()
  File "C:\Users\auser\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4/tools/prog.py", line 128, in main
    return_code = pymcuprog_basic(args, fuses_dict)
  File "C:\Users\auser\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4/tools/prog.py", line 201, in pymcuprog_basic
    args_start)
  File "C:\Users\auser\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4\tools\libs\pymcuprog\pymcuprog_main.py", line 545, in _start_session
    backend.start_session(sessionconfig)
  File "C:\Users\auser\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4\tools\libs\pymcuprog\backend.py", line 362, in start_session
    sessionconfig.interface_speed)
  File "C:\Users\auser\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4\tools\libs\pymcuprog\programmer.py", line 83, in setup_device
    options=self.options)
  File "C:\Users\auser\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4\tools\libs\pymcuprog\nvm.py", line 42, in get_nvm_access_provider
    accessprovider = NvmAccessProviderSerial(transport, device_info, baud=frequency, options=options)
  File "C:\Users\auser\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4\tools\libs\pymcuprog\nvmserialupdi.py", line 54, in __init__
    self.avr = UpdiApplication(port, baud, self.dut)
  File "C:\Users\auser\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4\tools\libs\pymcuprog\serialupdi\application.py", line 79, in __init__
    datalink.init_datalink()
  File "C:\Users\auser\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4\tools\libs\pymcuprog\serialupdi\link.py", line 44, in init_datalink
    raise PymcuprogError("UPDI initialisation failed")
pymcuprog.pymcuprog_errors.PymcuprogError: UPDI initialisation failed
pymcuprog.serialupdi.physical - INFO - Closing port 'COM3'
the selected serial port pymcuprog.serialupdi.physical - INFO - Closing port 'COM3'
 does not exist or your board is not connected
`

[image]

SpenceKonde, 2023-02-14:
I don't know what that error means either, that's a @qbolsee question, not a @SpenceKonde question.

The error messages in SerialUPDI are so bad they are embarrassing

mcuee, 2023-02-19:
@unebonnevie 

If you can not get the python code to work, you may want to give avrdude 7.1 a try.
https://github.com/avrdudes/avrdude/releases/tag/v7.1 (Windows binary)

You can also use Arduino version here (Windows, Linux and macOS binary)
https://github.com/arduino/avrdude-packing/releases

unebonnevie, 2023-02-20:
[quote]
If you can not get the python code to work, you may want to give avrdude 7.1 a try now that you have the binary file for the avr chip generated by DxCore.
[/quote]
I see flash directly to AVR128DB28.  I thought the AVR128DB28 could only be programmed via UPDI not ISP.  Still very new with this MCU,

Thank you.

mcuee, 2023-02-20:
> I see flash directly to AVR128DB28. I thought the AVR128DB28 could only be programmed via UPDI not ISP. Still very new with this MCU,

avrdude 7.1 supports UPDI.

unebonnevie, 2023-02-21:
Thanks @mcuee !  I ended up following the instructions at the below link and used jtag2updi as the programmer.

For the below instructions, I downloaded the jtag2udpi from the github link in the article.  I couldn't program my Arduino UNO using the provided JTAG2UPDI.hex file.  So, I basically I created a jtag2updi Arduino UNO project and copied all the source files from the github to the Arduino project.  Boot up the Arduino IDE, and choose my Ardunio Uno board.  Open the jtag2updi.ino file, thus the project.  Build it and upload to my Arduino Uno.  Now my Arduino Uno is a jtag2updi programmer.

Then I follow the rest of the instructions in the below article and created a blink program for my AVR128DB28.  Built it and uploaded it!  And it works!!!  All I needed is 4.7k resister in series with the AVR128DB128's UPDI pin.  And the Arduino UNO's digital pin #6 connected to the AVR128DB28's UPDI pin in series with the 4.7k.

https://www.edaboard.com/threads/programming-the-new-dx-series-of-avr-microcontrollers-using-the-updi-isp-interface.402221/

mcuee, 2023-02-21:
@unebonnevie 

Yes jtag2updi is a good option as well. Its support was added in avrdude 7.0 release. 

The jtag2updi FW had some problems before. Now it is quite a bit better. Still you should be able to get serialupdi working as well. Please refer the following instruction for serialupdi.
https://github.com/SpenceKonde/AVR-Guidance/blob/master/UPDI/jtag2updi.md

SpenceKonde, 2023-02-21:
IMHO, what someone really needs to do is turn jtag2updi into a much more arduino style sketch, and translate all the baffling C++ nonsense into something more readable. And from there, separate the UPDI-out portion from the JTAG-in portion. Thus creating two libraries, one that would take in a program over jtag and, and one that would program over UPDI. 

Give them each a pointer to the other one and have a simple API exposed by the UPDI object to read and write memory, and configure the target, and likewise a system to report errors. 

It could be made as an arduino sketch then, like ArduinoAsISP. And if you used a modern AVR, you wouldn't need to bitbang the updi protocol either, so it wouldn't need to monopolize the CPU. The UPDI-out and JTAG-in could be subclasses of an abstract Output and Input class, so you could just adapt ArduinoAsISP to be a an ISP-out, or make an optiboot-out. Input subclasses might be JTAG-in or STK500-in, or HyperUPDIin (this is an idea I've been thinking about - you take like a 128DB32 - shedloads of RAM. Take data in like 3-4k chunks to minimize usb latency - you could keep like a giant 8k+ ring buffer while doing your upload - and with the UPDI actually done with a UART with lbme and odme setyou could both write and receive at the same time.  or maybe you have a standalone programer with UPDI-out and W25Flash-in and so on - the potential here is HUGE

unebonnevie, 2023-02-21:
> Still you should be able to get serialupdi working as well. Please refer the following instruction for serialupdi.
> https://github.com/SpenceKonde/AVR-Guidance/blob/master/UPDI/jtag2updi.md

@mcuee / @SpenceKonde -- Yeah, I spent days on trying to get the SerialUPDI to work, as posted by my very original post of this issue.

Let me just ask one question here:  With SerialUPDI, I don't have to go through an Arduino UNO, for example, as a SerialUPDI programmer to program my AVR128DB28, because I don't see anything that would tell me to make an Arduino UNO into a SerialUPDI programmer.

Literally, I followed the instructions from https://community.openenergymonitor.org/t/getting-started-with-the-avr-db-on-a-breadboard/20255, wiring up the AVR128DB28's UPDI with a diode (N4148) and a 4k7 resistors to the RXD and TXD pins, respectively, of a serial USB board.  I made sure to use 3V3 as the supply voltage for the AVR128DB28.

pcfreak1201, 2023-02-21:
Uh, why only 3V3 Vcc? I'd also problems to build my own adapter, but this is working, if you remove the LEDs / resistors: 
[image]

unebonnevie, 2023-02-21:
@pcfreak1201 -- Thanks for the image of you make the USBSerial thing to work.

>Uh, why only 3V3 Vcc? I'd also problems to build my own adapter, but this is working, if you remove the LEDs / resistors:

The 3V3 Vcc is shown from the instructions of the original post.  And I assume that your removed the TX/RX's LED indicators with their current limiting resistors (in series with the LEDs) from your USBSerial device in order to have serialupdi working?

pcfreak1201, 2023-02-21:
Yes, I removed/cutted the LEDs only, because it is easier to identify them :-) and than in works (5V)
And Spence always recomment Schottky diodes (faster switching and only .3V loss) instead of 1N4xxx Si diodes

unebonnevie, 2023-02-21:
As an experiment, AFTER I have successfully flashed my blink program on the AVR128DB28 via the Arduino UNO with jtagupdi programmer, I went ahead and flashed the bootloader on the AVR128DB28 as well.  Then I change my board config (via the Arduino IDE) to the AVR128DB28 *with* Optiboot.  I should be able now to a program (the same blink program) via the bootloader this using JUST Arduino IDE with the USBSerial's TX/RX connected to the AVR128DB28's RX0/TX0, respectively.  But it does not work! And I keep getting the below.  Thoughts anyone?

`avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\asle\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xac
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xac
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0xac
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0xac
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0xac
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0xac
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0xac
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0xac
`

[image]

SpenceKonde, 2023-02-21:
Do you have the AVR's Reset pin pulled up to Vcc with a 10k resistor and capacitively coupled the DTR line of the serial adapter in order to reset into the bootloader> The bootloader only runs after a reset (by default either reset pin reset or software reset), the standard autoreset circuit is appropriate provided you had reset pin enabled when you burned tyhe botloaderr

pcfreak1201, 2023-02-21:
Not holding, just triggering before the serial communication starts ... not sure how much ms ...

unebonnevie, 2023-02-21:
>On the screenshot you selected PA4 + PA5 for the bootloader!?

Great catch.  I also tried PA0 and PA1, and it still does not work.

Reflashed the bootloader successfully, per the below output, with the new config attached.  But still fails to flash the blink program via the bootloader.  Will look into it more.

I see: "**avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description**" in the below output when burning the bootloader.

If anyone has any idea, please let me know.  Thank you!

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\me\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : jtag2updi
JTAG ICE mkII sign-on message:
Communications protocol version: 1
M_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
S_MCU:
  boot-loader FW version:        1
  firmware version:              6.00
  hardware version:              1
Serial number:                   00:00:00:00:00:00
Device ID:                       JTAGICE mkII
         AVR Part                      : AVR128DB28
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no        125  125      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9   16      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          4    1      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           flash          0     0     0    0 no     131072  512      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        512   32      0     0     0 0x00 0x00

         Programmer Type : JTAGMKII_PDI
         Description     : JTAGv2 to UPDI bridge
         M_MCU hardware version: 1
         M_MCU firmware version: 6.00
[... 96 lines cut ...]
avrdude: reading on-chip fuse7 data:

Reading | ################################################## | 100% 0.01s

avrdude: verifying ...
avrdude: 1 bytes of fuse7 verified
avrdude: reading input file "C:\Users\me\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4/bootloaders/hex/optiboot_128dx_ser1_extr.hex"
avrdude: writing flash (512 bytes):

Writing | ################################################## | 100% 0.08s

avrdude: 512 bytes of flash written
avrdude: verifying flash memory against C:\Users\me\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4/bootloaders/hex/optiboot_128dx_ser1_extr.hex:
avrdude: load data flash data from input file C:\Users\me\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4/bootloaders/hex/optiboot_128dx_ser1_extr.hex:
avrdude: input file C:\Users\me\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4/bootloaders/hex/optiboot_128dx_ser1_extr.hex contains 512 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.08s

avrdude: verifying ...
avrdude: 512 bytes of flash verified

avrdude done.  Thank you.

[image]

unebonnevie, 2023-02-22:
**SUCCESS!!!!!**  Programmed the blink demo via the bootloader AFTER having flashed the bootloader FIRST via the Arduino UNO acting as a jtagupdi programmer!!!! (See above!)

Below is the programming output from the Arduino IDE, programming the blink demo via the bootloader.

Attached is the image of the FINAL config used!!!  Make sure to use the exact config when flashing the bootloader via jtagupdi programming!

C:\Users\me\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude -CC:\Users\me\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4/avrdude.conf -v -pavr128db28 -carduino -D -PCOM3 -b115200 -Uflash:w:C:\Users\me\AppData\Local\Temp\arduino_build_216385/AVR128DB_blink.ino.hex:i 

avrdude: Version 6.3-20201216
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\me\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.4/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
         AVR Part                      : AVR128DB28
         Chip Erase delay              : 0 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 0
         StabDelay                     : 0
         CmdexeDelay                   : 0
         SyncLoops                     : 0
         ByteDelay                     : 0
         PollIndex                     : 0
         PollValue                     : 0x00
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           prodsig        0     0     0    0 no        125  125      0     0     0 0x00 0x00
           fuses          0     0     0    0 no          9   16      0     0     0 0x00 0x00
           fuse0          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse1          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse2          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse4          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse5          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse6          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse7          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           fuse8          0     0     0    0 no          1    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          4    1      0     0     0 0x00 0x00
           data           0     0     0    0 no          0    0      0     0     0 0x00 0x00
           flash          0     0     0    0 no     131072  512      0     0     0 0x00 0x00
           eeprom         0     0     0    0 no        512   32      0     0     0 0x00 0x00

         Programmer Type : Arduino
         Description     : Arduino
         Hardware Version: 3
         Firmware Version: 26.1
         Vtarget         : 0.3 V
         Varef           : 0.3 V
         Oscillator      : 28.800 kHz
         SCK period      : 3.3 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e970e (probably avr128db28)
avrdude: reading input file "C:\Users\me\AppData\Local\Temp\arduino_build_216385/AVR128DB_blink.ino.hex"
avrdude: writing flash (1366 bytes):

Writing | ################################################## | 100% 0.08s

avrdude: 1366 bytes of flash written
avrdude: verifying flash memory against C:\Users\me\AppData\Local\Temp\arduino_build_216385/AVR128DB_blink.ino.hex:
avrdude: load data flash data from input file C:\Users\me\AppData\Local\Temp\arduino_build_216385/AVR128DB_blink.ino.hex:
avrdude: input file C:\Users\me\AppData\Local\Temp\arduino_build_216385/AVR128DB_blink.ino.hex contains 1366 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 0.05s

avrdude: verifying ...
avrdude: 1366 bytes of flash verified

avrdude done.  Thank you.

[image]

closed 2023-02-22 by unebonnevie