SpenceKonde/DxCore #399: USERSIG fails to write on AVR32DD14

KWolfe81, 2023-02-13:
Code below.  boot_cnt is always 4294967295 (-1) on reset.

```
#include <USERSIG.h>
#include <EEPROM.h>

typedef struct
{
  uint32_t boot_cnt;
  uint32_t flashed_param_1;
  uint32_t flashed_param_2;
} flashed_config_t;

static flashed_config_t flashed_config = { 0 };

//----------------------------------------------------------------
void setup()
{ 
  Serial.pins(PIN_PA0, PIN_PA1);
  Serial.begin(115200);

  // wait until serial port opens for native USB devices
  while ( !Serial )
  {
    delay(1);
  }
  
  USERSIG.get( 0, flashed_config );

  Serial.println(flashed_config.boot_cnt);
  flashed_config.boot_cnt++;
  Serial.println(flashed_config.boot_cnt);

  USERSIG.erase();
  delay(100);
  
  USERSIG.put( 0, flashed_config );
  USERSIG.flush();
  delay(100);
```

SpenceKonde, 2023-02-14:
well that's no good...

SpenceKonde, 2023-02-14:
I'd love to test it but for some reason I don't understand, my AVR64DD14 test board is reporting that it's a different chip ID. It worked a few days ago....

KWolfe81, 2023-02-14:
Huh.  I should be able to solder up an AVR64DD14 in the next day or two to
test.

KWolfe81, 2023-02-15:
Same behavior on the AVR64DD14.  Further, maybe this should be another git issue, but EEPROM retention on arduino upload isn't working either (it's wiping the EEPROM):

[image]

SpenceKonde, 2023-02-15:
Usersig library for Dx is setting the nvm command incorrectly, it's setting it to 1 (NOCMD) instead of 2 (Flash Write Enable)

SpenceKonde, 2023-02-15:
Oh wait, that's not it, it's just doing the write process all wrong.

KWolfe81, 2023-02-15:
Thanks for looking into that.  EDIT as well: ignore the comment above about the EEPROM getting wiped.  I didn't have the bootloader burned on that device.

SpenceKonde, 2023-02-15:
Okay, I see why it's being set to 1, that's valid... but the indexes i'm writing sequentially are coming out wrong

Basically nothing looks like it should here.... I don't understand what's going on. 

I'm storing flags into the GPIOR's to export data from inside the library.... but the stuff that it reports makes no sense. 

And to add to the frustration, i fixed a bug that I was sure was the problem, but... it didn't fix it.

AAAh! Got it!

I was using the wrong kind of protected write - it needs a _PROTECTED_WRITE_SPM() not a normal _PROTECTED_WRITE

closed 2023-02-15 by SpenceKonde

KWolfe81, 2023-02-18:
@SpenceKonde It was a bit of a pain in the ass switching from Board Manager installed DxCore to manually installed core, but this fix works!  Thanks again.  You rock.

commit 08cfa142 (2023-02-15): Fix #399