SpenceKonde/DxCore #403: analogChannelToDigitalPin macro on ddseries has errant parenthesis

coburnw, 2023-02-20:
It seems that the forth conditional has an extra parenthesis which incorrectly groups the condition with the true expression:
https://github.com/SpenceKonde/DxCore/blob/6a4634510d24911ee11ed31bd79e3f67b79a96b8/megaavr/variants/32pin-ddseries/pins_arduino.h#L99

The same appears to be true for its compliment on line 103 also.  The db variants file compiles just fine.

Just as a fun aside, that very line causes my emacs to lock up when it attempts show brace completion...  Even after repair. 

Thanks for a great core!

SpenceKonde, 2023-02-21:
Thanks you just barely squeaked into 1.5.5

SpenceKonde, 2023-02-21:
Buddy those like 20 lines of bad code contained more buried treasure than captain blackbeard's lost hoard. 

Probably in all DD variants.

SpenceKonde, 2023-02-21:
Actually not widespread in DD-series. Found some dubious but no syntactically invalid code in other DD variants.

coburnw, 2023-02-21:
Kinda like buying a gold mine and only finding a couple nuggets near the entrance...

commit 1cf3eb42 (2023-02-24): Fix #403 - there were a bunch of issues around these
commit 47548bd3 (2023-02-24): Fix #403
closed 2023-02-24 by SpenceKonde