SpenceKonde/DxCore #406: An example of SDCard arduino sketch for the AVR128DB28?

unebonnevie, 2023-02-23:
Hi,

Now I've got things going with my AVR128DB28, I have searched high and low for working example of Arduino SDCard interfacing with the AVR128DB28 or any AVR238DBxx.

I looked at Arduino's SD.h/cpp source code and it has the begin() overloaded methods that either take a chip select pin or a CLK and chip select pin.  The AVR128DB28 has more than one SPI modules, e.g., SPI0, SPI1, etc.

How would the Arduino SD library know which SPI module to use?

Thanks.

SpenceKonde, 2023-02-23:
https://github.com/SpenceKonde/DxCore/blob/master/megaavr/libraries/SPI/README.md

SpenceKonde, 2023-02-23:
Closing issue as this is a clearly documented limitation of the core and considerable thought was devoted to tryingto find a better way without success, so this is not an action item, and I use the issue list for that purpose

SpenceKonde, 2023-02-23:
To be clear I'll still see and respond.

commit e91e84ae (2023-02-24): Fix #406
closed 2023-02-23 by SpenceKonde