SpenceKonde/DxCore #410: analogWrite() and DAC

SpenceKonde, 2023-02-27:
### Discussed in https://github.com/SpenceKonde/DxCore/discussions/409

<div type='discussions-op-text'>

<sup>Originally posted by **ikkei3** February 26, 2023</sup>
1.5.5 has been released.
I have AVR64DB32 and AVR64DD32.
analogWrite() works on AVR64DB32 but not on AVR64DD32.
DAC does not work on AVR64DB32 and AVR64DD32.
do i need to configure anything?</div>

SpenceKonde, 2023-02-27:
I told people tocheck analog write and the dac in 1.5.0, but only hear 5 versions later!

Okay, more information please - analogWrite broken everywhere or just on the DAC pin and you say the dac doesn't work anywhere?

SpenceKonde, 2023-02-28:
@ikkei3  which pins are you using for analogWrite that don't work? There are 23 PWM capable pins on each of those two parts, of which 11 can be active simultaneously. Ya gotta help narrow tthis down for me

How have you configured the PORTMUX? 

Have you read the part specific documentation? 

https://github.com/SpenceKonde/DxCore/blob/master/megaavr/extras/DB32.md
https://github.com/SpenceKonde/DxCore/blob/master/megaavr/extras/DD32.md - PWM pins that the core selects at startup are in bold, though for TCA and TCD, you can switch  to other rows of the table just by setting the PORTMUX.TCAROUTEA or PORTMUX.TCDROUTEA (DD only due to errata on the

ikkei3, 2023-03-01:
I'm only using PIN_PA7 for analogWrite() for PWM.
and PIN_PD6 for analogWrite() for DAC.

I am not using PORTMUX.
Leave as default.
What are the necessary PORTMUX settings for errata?

Best regards,
-------------------------------------------------------------
***@***.***
Kazuhiro Yamanishi
-------------------------------------------------------------

SpenceKonde, 2023-03-01:
The necessary settings for the portmux errata on AVR DA and AVR DB are that the it be left at the default setting. 
Otherwise enabling PWM on pin 0 (CMPAEN) on that TCD0 pinset enables PWM on pins 1, 2, and 3, and CMPBEN, CMPCEN, and CMPDEN are non-functional. (note that the timer can only generate 2 PWM signals (though a third 8-bit pwm signal can be bodged up with  PROGEV, 2 event channels and an even numbered CCL LUT (or 3 event channels and and odd numbered LUT), but only under certain combinations of prescaling settings.

SpenceKonde, 2023-03-01:
In the meantime, you can easily workaround the bug. The bug caused the wrong pin to be set output... and it didn't sem to be setting it at all for TCD, which is why I need to investigate more closely (on my device, PWM definitely comes out on PA7 even before my fixes - the TCD PWM pins worked correctly, only the TCA0 PWM pins (Px0-Px5 on any port, core defaults to PORTD on AVR-DB, since despite losing the first pin on the DBs, it's still better than encouraging the MVIO pins to be used for PWM (the most common configuration I think is Vdd = 5V, Vddio2 = 3.3V, and the most common applications of PWM would prefer the 5v vs the 3.3v), so since - the default was miserable (TCA0 defaults to PORTA... where of the six pwm capable pins, the first two pins are your crystal pins, the next two are your non-MVIO-pin I2C host pins, and the last two are the only option left for USART0 on the DB) - and hence we didn't default the core to the chip default, we chose default for the most common use cases. 
```

ikkei3, 2023-03-02:
OK. OK.
I know very well that the defaults are the best.
So why is this code able to output PWM on AVR64DB32 but not on AVR64DD32?
Neither DAC comes out.

void setup() {
  // do nothing
}

void loop() {
  static uint8_t a = 0;
  analogWrite( PIN_PA7, a);
  analogWrite( PIN_PD6, a * 4);
  a++;
  delay(10);
}

I'm sure you know why.

Best regards,

-------------------------------------------------------------
***@***.***
Kazuhiro Yamanishi
-------------------------------------------------------------

SpenceKonde, 2023-03-02:
I know sort of why there is the DAC issue (only sort of though). I have no f'ing clue whats's up with the PWM on the DD, and will note that you got lucky in picking a pin that would do PWM at all on the DB as TCA PWM was busted completely.). I am fixing the PWM issue. 
Then I will look into why there an issue with the DD PWM.

ikkei3, 2023-03-02:
DAC issues
This is not a big deal as setting the register directly works just fine.

If DD PWM is fixed, I'm happy with that.

Best regards,
-------------------------------------------------------------
***@***.***
Kazuhiro Yamanishi
-------------------------------------------------------------

SpenceKonde, 2023-03-02:
It's not fixed yet, and I've been working on it pretty much straight out. 
As it was released, absolutely, **analogWrite() should not work anywhere on any pin in the release** on **any part**. And yet it does.

I think the errata listings are missing a few entries.. I'm currently trying to assemble a DB32 that works
I fried the first one when I stupidly connected the MVIO power *backwards* and am now building one on generic protoboard. 

Both the DAC and PWM issues are considered critical and as soon as I have verified code I am going to do a release, and contact like half a dozen VIPs to let them know.

SpenceKonde, 2023-03-03:
I am able to reproduce your observation that PWM does not work from TCD. Additionally, I notice that the mux is not configured to match the docuementation 
```
PWM selftest
Testing TCA0 mux number 3 channel 0
Pin is 12
Pin does not exist on this part
Testing TCA0 mux number 3 channel 1
Pin is 13
One second togglecount: 2940
Testing TCA0 mux number 3 channel 2
Pin is 14
One second togglecount: 2939
Testing TCA0 mux number 3 channel 3
Pin is 15
One second togglecount: 2938
Testing TCA0 mux number 3 channel 4
Pin is 16
One second togglecount: 2939
Testing TCA0 mux number 3 channel 5
Pin is 17
One second togglecount: 2937
Testing TCB0
AltPin 24
One second togglecount: 1
Testing TCB1
AltPin 25
One second togglecount: 0
Millis is using TCB2
The current configuration does not permit PWM from this timer channel.
Testing TCD0 mux number 2
Pin is 20
One second togglecount: 1
Testing TCD0 mux number 2
Pin is 21
One second togglecount: 0
Testing TCD0 mux number 2
Pin is 22
One second togglecount: 0
Testing TCD0 mux number 2
Pin is 23
One second togglecount: 0
The test has completed
Pass: 5/11
```

ikkei3, 2023-03-04:
I checked analogWrite().
It gets worse and worse as the version goes up.
Early versions were better.

Best regards,
-------------------------------------------------------------
***@***.***
Kazuhiro Yamanishi
-------------------------------------------------------------

SpenceKonde, 2023-03-04:
> I checked analogWrite(). It gets worse and worse as the version goes up. Early versions were better. Best regards, --------------

That was a non-constructive (had you pointed out specific versions and what regression occurred there, that might have been useful but you didnt). I don't know if you meant it that way, but I found that comment rude and insulting, especially when I've spent the better part of the last week working on fixing it (though a cabinet of drawers in my room collapsed and I've had to devote time to repairing it). Yes, a regression occurred this is software, it happens sometimes.

I've built a test board and written a rigorous PWM test sketch that all pins that are expected to have PWM available do have PWM and that digital write does turn off that PWM. 

During this process I found a case of several semi-user facing macros (unlikely to be called from user code, but very likely to be called from libraries,  that, while they didn't cause problems with this in particular, could cause almost arbitrary undefined behavior if called using a signed integer as a pin number instead of an unsigned one (because of the fact that macros don't check or implicitly convert types), and I'm rather surprised that I haven't gotten bug reports about them already (though they may be the underlying cause of some library compatibility issues). I consider those issues to be serious by I think I have those fixed. 

As for the problem at you have there is something something **VERY** strange going on. It's trivial to reproduce, but only on the DD, yet what appears to be be causing it should cause it to happen on the DA and DB as well. And yet it doesn't. Preliminary evidence suggests that the issue had gone unnoticed on the DB and DA parts because of an undocumented silicon bug relating to the port overrides. I need to shutdown and restart my PC as it's got an unrelated hung process dragging performance down, meaning I have to close and figure out what to do with over thirty unsaved documents. Once I've completed the reboot, I will resume work on this issue. The way you talk, one would think you were paying me or something. I'm doing this for free, and I'm making it such a priority because PWM is widely used and very important to all users.

ikkei3, 2023-03-04:
What I checked about analogWrite() is fact.
I believe that this fact will be a hint to solve the problem.
I have no other thoughts.

-------------------------------------------------------------
***@***.***
Kazuhiro Yamanishi
-------------------------------------------------------------

pcfreak1201, 2023-03-04:
BTW: Spence, you earlier talked about setting up some kind of sponsoring. Do you have something else than tindie? (Because I want to sponsor you and not the US post service - and I have all chips I want to play with ;-)

SpenceKonde, 2023-03-05:
I'm going to have patreon if I ever get a break from these issues long enough to do the writeup. I am bad at social media, and the person I used to ask about things like that vanished, which sucks

SpenceKonde, 2023-03-05:
So this appears to work now on the DD32 after fixing bugs all over the  place. 

Then I removed the direction set macro call. 

The test results were unchanged. The direction setting was a red herring **and all kinds of timer have a peripheral override of PORTx.DIR!!!!!**

SpenceKonde, 2023-03-05:
In any event I've checked in what I belive to be a fix for this on the DD32. Some of the problems were in the variant file itself, which was not consistent with the part-specific documentation, which said we run with TCD0 on the high half of PORTA and TCA0 on PORTF, and the TCBs on PA2 and 3. all of those were swapped... in some places.... but not others... which was part of the problem, but not all of it; there were also bugs throughout analogWrite()

SpenceKonde, 2023-03-11:
Should be fixed. Finally

commit 85159c1e (2023-03-05): Okay, this may fix #410.

There were many bugs involved.
closed 2023-03-11 by SpenceKonde