SpenceKonde/DxCore #411: Use of GPIOR3 by xCore

alx2009, 2023-03-05:
Up to release 1.4.10

SpenceKonde, 2023-03-05:
Wtf....

alx2009, 2023-03-05:
Forgot to mention, I ran this sketch on a AVR128DB48. With or without optiboot (Serial on PA0 and PA1). Internal clock 24 MHz. Also tried different optimizatins, printf settings and more.

MX682X, 2023-03-06:
SREG is bit-accessible. Through BSET and BCLR as well as ALL branch instructions. What do you thing branch if Carry is set is doing? 
And cli is an alias for clearing the Interrupt enable bit in sreg. And the reason we are not just doing cli (atomic) sei is that when we do cli, we don't know if it was even set before. that's we we save the sreg. 

And the compiler treats only some of the Intermediate translated functions as sreg affecting. (Any function that has a "set CC" in the avr.md file, iirc). Anything assembly is not checked. and If I recall correctly cli() is an assembly wrapper, so no check there.

(I hope I've understand everything correctly of your previous post, Spence)

P.S. If you send us the .lst files, (Or spence makes a test compile on his own macbine, don't have time right now) we might be able to look for the GPIOR address and maybe find a third place that would explain the behaviour...

SpenceKonde, 2023-03-06:
His behavior is entirely explained by the USART issue - every time a character was sent or received by the USART it was clearing the LSB of GPIO3. 

What I was trying to ask about was the fact that cli() has the memory clobber. That is potentially extremely expensive, and my understanding of it from some discussion threads is that this is needed only because otherwise, the compiler could reorder
```
uint8_t oldsreg = SREG;
cli_with_no_memory_clobber();
must_use_atomic_access |= 1;
SREG=oldsreg
```
to

```
uint8_t oldsreg = SREG;
must_use_atomic_access |= 1; //No longer atomic!
cli_with_no_memory_clobber();
SREG=oldsreg
```
But it would not be able to perform such perverse reordering on:
```
uint8_t oldsreg = SREG;
SREG &= ~0x80; 
must_use_atomic_access |= 1;
SREG=oldsreg;
```

Interesting - in comparing the output from these, sorts of constructions in trivial examples, it seems like by touching the SREG we get the same "memory barrier" effect, leading to a lot of storing and reloading of things that are not changed nor declared volatile. Why is that needed? I had assumed that the memory clobber was used because it had the sideffect of preventing reordering, but it looks like the memory barrier may, in the compilers way of thinking, be necessary?

Re: SREG and bit access:

**I** know SREG is bit accessible, but unlike the low I/O space, **the compiler** doesn't seem to:
```
SREG &= ~0x80;
```
doesn't compile to:
```
cli
```
it compiles to this:
```
in	r24, 0x3f	; 63
andi	r24, 0x7F	; 127
out	0x3f, r24	; 63
```
`GPIOR0 &= ~0x80` in contrast compiles to `cbi 0x1C, 7`

MX682X, 2023-03-07:
I see. Well, avr-gcc has some extra instructions to cover cli and sei (see https://github.com/MX682X/gcc/blob/master/gcc/config/avr/avr.md#L5742-L5774) but it does not seem to know about that SREG is bit accessable like the I/O registers. With enough time it might be patchable, but it seems I won't have the time anytime soon...
Spence, have you considered opening a Problem-Report on the Bugtracker of gcc? Maybe there are people willing to help...

SpenceKonde, 2023-03-10:
Dear god... I cannot even tell what language that code is written in. It is a fortress of letters and punctuation marks to ,me

commit 78089da9 (2023-03-10): Fix #411, #410

Also, files in which I was working got a once-over for formatting and style things (we don't check them in the core because it sucks horribly and gives "failures" that are interpreted as hard failures.
closed 2023-03-10 by SpenceKonde