SpenceKonde/DxCore #419: Upload error:"A programmer is required to upload" for AVR64DD32 on Arduino IDE 2.0.3

HonestQiao, 2023-03-17:
OS: macOS 11.7.4
Arduino IDE: 2.0.3
DxCore: 1.5.6
Board: AVR64DD32
[image]

Problem:
When I upload a sketch , Arduino IDE throw an error:
A programmer is required to upload
[image]

If I use Arduino IDE 1.8.19, everything is ok.

pcfreak1201, 2023-03-17:
Well: 
[IDE 2.0.x is still unsupported](https://github.com/SpenceKonde/DxCore#ide-20x-unsupported-if-you-use-it-use-the-release-not-an-old-rc-all-versions-prior-to-200-rc92-known-to-have-critical-regressions)

SpenceKonde, 2023-03-18:
Yeah I don't know what we're supposed to do in Arduino 2.x for parts that do not  have a n upload method except whatever the user has selected as the programmer.

SpenceKonde, 2023-03-18:
Closing until there is an actionable change that I can make to resolve this issue.

closed 2023-03-18 by SpenceKonde