SpenceKonde/DxCore #421: tinyNeoPixel

cybermatrix84, 2023-03-27:
Hello thankyou so much for such a great core

i had flickering issue with [sk6812mini](https://akizukidenshi.com/download/ds/opscoled/sk6812-mini_rev09_cn.pdf)  which is no problem with common ws2812, maybe its timing issue?

hope there is any solution, thankyou so much

SpenceKonde, 2023-03-27:
Interesting. I think there may be something wrong with the 24 MHz implementation - it seems to be unique to 24 MHz. 

Which kind of LEDs exactly are you using? (if they're the "fairy light" style ones, a picture of which one would be helpful - as there are four different kinds and at least 2 of them and likely 3 of them use different types of LED. The part number used in listings for assembled (and less often, unassembled component) is frequently not what you're getting. There are a very large number of "compatible" WS2812 clones (and they generally are compatible), even though the only names you'll ever see mentioned in a listing are WS2812 and SK6812, there are at least half a dozen, not counting the low current versions that some of them exist in. I have samples (and in some cases far more than "samples"). 

I have a huge lighting project under way (though unfortunately it hasn't gotten to the point where I can light up any LEDs with the new controller boards. All the original controller boards wereburned out due to what turned out to be a combination of a flawed wiring topology (I do power injection by running 19v through thicker wire wrapped around the wires in the christmas light string type ones, and at the head of each string, a mini560 buck converter (like $1 on aliexpress) gives me the 5v to power that string. All the 5v rails are tied together (in case of a failure of one buck converter) and to reduce the difficulty of setup, I'd clipped off the 3 pin JST SM connectors and installed 5 pin ones. Unfortunately I hadn't considered 2 factors: 1. JST-SM connectors are utter shit connectors. They are poorly designed, and the fact that they're all chinese knockoffs doesn't help either (JST tried to save a bit of money by reusing one of the terminals from another connector). 2. When a buck converter is not connected to ground, the voltage on the positive input and output terminals is approximately equal. And the 5v from the head end was used to power the controller. So certain patterns of briefly interrupted connections due tothe shit connectors would putr 19v on the 5v rail, The LEDs would survive this, at least for a very short time. The ATmega328PB's I was using were not so fortunate. And I didn't put 2 and 2 together until I'd burned out my entire stock of the older designs, and was then working with an AVR DA-based on Aaah the lights, worked, the party cheered, and then as I stepped off the chair to put the controller down, suddenly the animation stopped, and I felt teh chip getting hot underneath my finger. At that point the penny dropped - It could only be caused by an intermittant connection, and it quickly became obvious from my schematics of the wiring which one it was. JST-SM was declared unfit for all purposes (still haven't decided what to do with the HUGE numberof JST-SM connectors I have - I got every size housing, buncha terminals. plus a bunch of prewired connectors. No way in hell am I using that garbage again. And so of the remaining working light strings (somewhere between 2 and 4 black wire ones failed, as did every single white wired one (the heat from the previous buck converter that I used generated enough heat that, in combination with the pressure from the ziptie and the squidgy low quality insulation on the cheap chinese wire, led to strands of copper working their way through the wire and shorting to another. (those converters were terribad - for the same pattern, they used a third more current than the mini560s (and cost 4x as much too), So running 255,255,255, I couldn;t run 4 strings from one 3.5A laptop power brick, I had to limit total brightness in software, which is no longer a problem).  That failure mode was in evidence on all 100% of white wire strings (with shitty white zipcord) but occurred in 0% of the black ones, which used (apparently) better black zipcord. And so all the connectors were once more chopped off and replaced with MX3.0 conectors. The two ground pins were connected more rigorously as well. I still had failures from some sort of power excursion though. I moved to an AVR128DB48E instead (E's overclock better, and the faster I can generate the pattern, the more LEDs I can have for a given target frame rate. I'm hoping I can hit my targets for string length and frame rate in all animation modes at 32 MHz, but if not, there's the option of 40 or even 48 MHz (though I suspect not all parts, even E-spec ones, are stable at 48; I think at room temperature they'll all do 40 stably, (most I-spec ones will too, and virtually all I-spec parts can do 32).  But, the firmware is in the middle of a giant overhaul.

cybermatrix84, 2023-03-28:
thankyou for quick reply.  i'm using this one
[image]
after doing some research on local web shop there is different in series which is sk6812mini and ws2812mini, i will try to buy it and see if there is any difference on 24Mhz, also i will try on higher clock.

thankyou so much

SpenceKonde, 2023-03-29:
Very interested to hear results. Here's an even more interesting comparison to make between a SK6812mini and WS2812mini in strip like that (where you can see the LED and controller dies): 
Put a camera on a tripod on the floor, and aim it straight down, up against a desk, so it's looking down on the surface of the desk. With one of them under the camera, mark the spot on the, adjust the zoom and height of camera  to enlarge the chip as much as possible, and take the picture. Then without changing the the height or zoom, swap in the other strip, and takea picture.. Then in any photo editing program you can confirm two things:
a) Which kind of LED has a larger die? 
b) If the dies are exactly the same size, compare the images very carefully. Can you spot any differences whatsoever? If not, they are likely actually the identical part and one of the sellers (probably the one hawking the WS2812mini) was lying,

cybermatrix84, 2023-04-03:
i got the led and i found its same chip inside so i think LED is sk6812mini, after i playing with cpu clock, somehow higher clock will drift timing, and flickering not appear by reducing clock to 12Mhz.

SpenceKonde, 2023-04-18:
Was looking over this code....
20, 24, 28 and 32 are getting corrections. 

Were you by any chance potentially calling .show() more than once per 250us? There are several versions of the pixels, frequently misbranded as eachother. Some of them latch in 50us, some latch in 6us, and some latch in 250us. we use the 50us figure in the same way that the Adafruit library always did.

cybermatrix84, 2023-04-18:
thankyou so much for keep looking to this issue..

i call .show() after update data every 30mS which means 30fps to refresh led. so you think i must adjust latch time in library to match the pixels?

SpenceKonde, 2023-04-21:
The timing issue is fixed in current github, or it should be. The timings were off a cycle or two for 24-48 MHz speeds

Latch time means that you may need to wait as many as 250us (0.25ms) before sending the next frame to be assured that is will be treated as a new frame depending on the make of your led controllers. We enforce that up to 50us as per the original library. With slow latching clones. you'd need to make sure you never called show() less than 250us after show() last returned.

SpenceKonde, 2023-04-21:
Note that I will be exercising this heavily in the coming weeks too.

closed 2023-04-21 by SpenceKonde

SpenceKonde, 2023-04-27:
It definitely had issues at 20 as well. One of the delays was off by several clock cycles.... Library generally got some cleanup for readability too.