SpenceKonde/DxCore #422: MILLIS_TIMER" is not defined (Core v1.5.6)

mikrocoder, 2023-03-30:
Hallo,

there is no small mistake that could have a big impact.

```
C:\Arduino IDE Portable\AVR128DB\arduino-1.8.19\portable\packages\DxCore\hardware\megaavr\1.5.6\cores\dxcore\wiring.c: In function 'nudge_millis':
C:\Arduino IDE Portable\AVR128DB\arduino-1.8.19\portable\packages\DxCore\hardware\megaavr\1.5.6\cores\dxcore\wiring.c:1510:8: warning: "MILLIS_TIMER" is not defined, evaluates to 0 [-Wundef]
 1510 |   #if (MILLIS_TIMER & 0x78) /* 0x40 matches TCDs, 0x20 matches TCBs, 0x10 matches TCA0 0x08 matches TCA1, so OR them together and AND it with the timer to make sure it's not RTC, disabled, etc.  */
      |        ^~~~~~~~~~~~
```

SpenceKonde, 2023-03-31:
Hmm, I will have to investigate what that define is supposed to be named

mikrocoder, 2023-03-31:
Hallo,

If it helps you, I previously had DxCore v1.5.1 and updated to 1.5.6. I have also already looked up where MILLIS_TIMER is defined. I think I found the error. You have changed the timers.h? Here you have deleted the line with TIMER4. I use millis with timer 4, that's why I get the warning.

timer.h DxCore v1.5.1 line 98 - #define MILLIS_TIMER TIMERB4
timer.h DxCore v1.5.6 line 98 - something is missing here for timer 4.