SpenceKonde/DxCore #423: Bootloader file specified but missing - shows up on all versions

defavltt, 2023-04-03:
First of all really appreciate the time and effort that you put into this core, i'm truly thankful.

All worked as expected, until I tried to select AVR128DB64(for the first time, 99% rest of the time AVR128DB48 is selected).
After that tried to upload bootloader through UPDI, and then this appears:

Bootloader file specified but missing: C:\Users\defavlttt\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.6\bootloaders\C:\Users\defavlttt\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.6\bootloaders\hex\optiboot_128dx_ser0_extr.hex

I have to say, when I tried to do all this version 1.3.2 was used. The error message provided here was from 1.5.6, a similar message appeared with 1.3.2:

Bootloader file specified but missing: C:\Users\defavlttt\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.3.2\bootloaders\optiboot_dx128

Deleted the \Local and program files directories, and reinstalled arduino and the core. Still doesn't fix it.

SpenceKonde, 2023-04-03:
You have got to be kidding me, you're telling me that while I have an urgent client project, BOTH of my cores can't "burn bootloader" on some or all parts"?

SpenceKonde, 2023-04-03:
Besides 
```
avrdbopti.bootloader.file={runtime.platform.path}/bootloaders/hex/{bootloader.class}{bootloader.port}{bootloader.entrycond}.hex

avrdbopti.menu.chip.avr128db48.build.mcu=avr128db48
avrdbopti.menu.chip.avr128db48.bootloader.class=optiboot_128dx
avrdbopti.menu.chip.avr128db48.upload.maximum_size=130560
avrdbopti.menu.chip.avr128db48.upload.maximum_data_size=16384
avrdbopti.menu.chip.avr128db48.build.variant=48pin-standard
avrdbopti.menu.chip.avr128db64=AVR128DB64
avrdbopti.menu.chip.avr128db64.build.mcu=avr128db64
avrdbopti.menu.chip.avr128db64.bootloader.class=optiboot_128dx
avrdbopti.menu.chip.avr128db64.upload.maximum_size=130560
avrdbopti.menu.chip.avr128db64.upload.maximum_data_size=16384
avrdbopti.menu.chip.avr128db64.build.variant=64pin-standard

avrdbopti.menu.bootloaderusart.ser0_alt=USART0 (alt): TX PA4, RX PA5
avrdbopti.menu.bootloaderusart.ser0_alt.bootloader.port=_ser0_alt
avrdbopti.menu.bootloaderusart.ser0_alt.bootloader.portname=Serial
avrdbopti.menu.bootloaderusart.ser0_alt.bootloader.portswap=1
avrdbopti.menu.bootloaderusart.ser0=USART0 (default pins): TX PA0, RX PA1 (int. clock only)
avrdbopti.menu.bootloaderusart.ser0.bootloader.port=_ser0
avrdbopti.menu.bootloaderusart.ser0.bootloader.portname=Serial
avrdbopti.menu.bootloaderusart.ser0.bootloader.portswap=0
avrdbopti.menu.bootloaderusart.ser1=USART1: TX PC0, RX PC1
avrdbopti.menu.bootloaderusart.ser1.bootloader.port=_ser1
avrdbopti.menu.bootloaderusart.ser1.bootloader.portname=Serial1
avrdbopti.menu.bootloaderusart.ser1.bootloader.portswap=0
avrdbopti.menu.bootloaderusart.ser1_alt=USART1 (alt): TX PC4, RX PC5 (48/64-pin only)
avrdbopti.menu.bootloaderusart.ser1_alt.bootloader.port=_ser1_alt
avrdbopti.menu.bootloaderusart.ser1_alt.bootloader.portname=Serial1
avrdbopti.menu.bootloaderusart.ser1_alt.bootloader.portswap=1
avrdbopti.menu.bootloaderusart.ser2=USART2: TX PF0, RX PF1
avrdbopti.menu.bootloaderusart.ser2.bootloader.port=_ser2
avrdbopti.menu.bootloaderusart.ser2.bootloader.portname=Serial2
avrdbopti.menu.bootloaderusart.ser2.bootloader.portswap=0
avrdbopti.menu.bootloaderusart.ser2_alt=USART2 (alt): TX PF4, RX PF5 (>28 pin only)
avrdbopti.menu.bootloaderusart.ser2_alt.bootloader.port=_ser2_alt
avrdbopti.menu.bootloaderusart.ser2_alt.bootloader.portname=Serial2
avrdbopti.menu.bootloaderusart.ser2_alt.bootloader.portswap=1
avrdbopti.menu.bootloaderusart.ser3=USART3: TX PB0, RX PB1 (48/64-pin only)
avrdbopti.menu.bootloaderusart.ser3.bootloader.port=_ser3
avrdbopti.menu.bootloaderusart.ser3.bootloader.portname=Serial3
avrdbopti.menu.bootloaderusart.ser3.bootloader.portswap=0
avrdbopti.menu.bootloaderusart.ser3_alt=USART3 (alt): TX PB4, RX PB5 (48/64-pin only)
avrdbopti.menu.bootloaderusart.ser3_alt.bootloader.port=_ser3_alt
avrdbopti.menu.bootloaderusart.ser3_alt.bootloader.portname=Serial3
avrdbopti.menu.bootloaderusart.ser3_alt.bootloader.portswap=1
avrdbopti.menu.bootloaderusart.ser4=USART4: TX PE0, RX PE1 (48/64-pin only)
avrdbopti.menu.bootloaderusart.ser4.bootloader.port=_ser4
avrdbopti.menu.bootloaderusart.ser4.bootloader.portname=Serial4
avrdbopti.menu.bootloaderusart.ser4.bootloader.portswap=0
avrdbopti.menu.bootloaderusart.ser4_alt=USART4 (alt): TX PE4, RX PE5 (64-pin only)
avrdbopti.menu.bootloaderusart.ser4_alt.bootloader.port=_ser4_alt
avrdbopti.menu.bootloaderusart.ser4_alt.bootloader.portname=Serial4
avrdbopti.menu.bootloaderusart.ser4_alt.bootloader.portswap=1
avrdbopti.menu.bootloaderusart.ser5=USART5: TX PG0, RX PG1 (64-pin only)
avrdbopti.menu.bootloaderusart.ser5.bootloader.port=_ser5
avrdbopti.menu.bootloaderusart.ser5.bootloader.portname=Serial5
avrdbopti.menu.bootloaderusart.ser5.bootloader.portswap=0
avrdbopti.menu.bootloaderusart.ser5_alt=USART5 (alt): TX PG4, RX PG5 (64-pin only)
avrdbopti.menu.bootloaderusart.ser5_alt.bootloader.port=_ser5_alt
avrdbopti.menu.bootloaderusart.ser5_alt.bootloader.portname=Serial5
avrdbopti.menu.bootloaderusart.ser5_alt.bootloader.portswap=1
```

SpenceKonde, 2023-04-03:
64 and 48k resolve to the same string and use the same binary!

note that 2.5.0 changed the directory structure to separate source from binaries because it was impossible to find the source burried under the >100 bootloader hex files we ship with.

defavltt, 2023-04-03:
Reverted back to my arduino and cores backup, after reinstalling it. This time I uploaded a sketch on a board with a bootloader already on it, despite the 'Bootloader file specified but missing' message. That went successful, after that, I went to burn a bootloader to a different board. that was successful too.
To me it seems like regular uploading of a sketch through serial fixed it?
This cause of this case was very hard to track down.

First time I saw that message, it prevented me from burning a bootloader. Unfortunately I didn't copy the error that it spewed out then.

Bootloader file specified but missing - still appears, but it doesn't prevent me anymore from burning a bootloader.

Sorry for opening an issue about this so fast.

SpenceKonde, 2023-04-04:
What version of the IDE is this? 

If it's 2.x.x, then it's a 2.x.x IDE bug that should be reported to Arduino.

defavltt, 2023-04-06:
it's 1.8.8 (I get it, I get it)

SpenceKonde, 2023-04-18:
I suspect that this is some arduino bug - as you can see the first part of the path is appearing twice in the current release version for you.

closed 2023-04-18 by SpenceKonde