SpenceKonde/DxCore #425: Issue with pinout for TXD4 and RXD4 on AVR128Dx64

freaklabs, 2023-04-15:
Hi. Found a discrepancy in UART4 for the AVR128Dx64 chips. Here is the description from the image posted here:
[image]
PE0 corresponds to RXD4
PE1 corresponds to TXD4

Here is the description from the datasheet:
[image]
PE0 corresponds to TXD4
PE1 corresponds to RXD4

I haven't tested yet so I don't know if it's a datasheet error or an error in the image but it's probably best to give the datasheet the benefit of the doubt. That said, it wouldn't be the first time I've run into a datasheet error. 

Thought it'd be good to point the issue out.

SpenceKonde, 2023-04-15:
Just an image error, I'll go fire up MS paint

SpenceKonde, 2023-04-15:
Uhm that's not what I see on my copy in github,

SpenceKonde, 2023-04-15:
Can you please link me to exactly where you're finding the bad file? I do not know where the file that needs to be fixed is.

freaklabs, 2023-04-18:
It's the image on the Tindie site:
https://www.tindie.com/products/drazzy/avr128da-development-board-arduino-compatible/
Is there an image on github?

closed 2023-04-30 by SpenceKonde