SpenceKonde/DxCore #430: unable to burn bootloader for AVR128DA32

uliano, 2023-04-23:
I'm quite unexperienced in the arduino world, however using the same tools and procedures, I didn't have any problem in burning a bootloader for an ATmega4808.

Initially I was using ATMELICE as a programmer and platformio but when receiving the error `Error: invalid `bootloader.class` in board config!` I tried Arduino 1.8.18 where "Uploading using Programmer gave me: 

```java.io.IOException: Cannot run program "ERROR: Upload using programmer is not supported on optiboot boards\"\"C:\Users\uliano\AppData\Local\Arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18/bin/avrdude"```

therefore I quickly prepared a cable with a diode between rx and tx in order to use SerialUPDI, the cable works as it is perfectly able to load sketches in both Platformio.

Apparently with this the somwthing was burnt 

```
SerialUPDI
UPDI programming for Arduino using a serial adapter
Based on pymcuprog, with significant modifications
By Quentin Bolsee and Spence Konde
Version 1.2.3 - Jan 2022
Using serial port COM4 at 57600 baud.
Target: avr128da32
Set fuses: ['0:0b00000000', '1:0x00', '2:0x00', '5:0b11001000', '6:0x04', '7:0x00', '8:0x01']
Action: write
File: C:\Users\uliano\AppData\Local\Arduino15\packages\DxCore\hardware\megaavr\1.5.6/bootloaders/hex/optiboot_128dx_ser2_extr.hex
Pinging device...
Ping response: 1E9709
Setting fuse 0x0=0x0
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x1=0x0
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x2=0x0
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x5=0xc8
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x6=0x4
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x7=0x0
Writing literal values...
Verifying literal values...
Action took 0.16s
Setting fuse 0x8=0x1
Writing literal values...
Verifying literal values...
Action took 0.16s
Finished writing fuses.
Chip/Bulk erase,
Memory type eeprom is conditionally erased (depending upon EESAVE fuse setting)
Memory type flash is always erased
Memory type lockbits is always erased
...
Erased.
Action took 0.06s
Writing from hex file...
Writing flash...
Writing flash...
Action took 0.46s
Verifying...
Verify successful. Data in flash matches data in specified hex-file
Action took 0.29s
```
however the led started immediately to blink in a freaky way and when tryin to upload the sketch using the bootloader I get:

```
Sketch uses 2178 bytes (1%) of program storage space. Maximum is 130560 bytes.
Global variables use 263 bytes (1%) of dynamic memory, leaving 16121 bytes for local variables. Maximum is 16384 bytes.
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x20
An error occurred while uploading the sketch
```

For sake of completeness I can report that neither Platformio is able to burn The bootloder with SerialUPDI sticking with the original error message

```
> pio run -e fuses_bootloader -t bootloader -v
Processing fuses_bootloader (upload_protocol: serialupdi; upload_flags: ; board_hardware.bod: 2.7v; board_hardware.eesave: yes; board_hardware.uart: ser2; board_hardware.rstpin: reset; platform: atmelmegaavr; framework: arduino; board: AVR128DA32; board_build.variant: 32pin-standard; build_unflags: ; build_flags: ; monitor_speed: 115200; monitor_port: COM3; monitor_dtr: 0)
-------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/AVR128DA32.html
PLATFORM: Atmel megaAVR (1.8.0) > AVR128DA32
HARDWARE: AVR128DA32 24MHz, 16KB RAM, 128KB Flash
PACKAGES:
 - framework-arduino-megaavr-dxcore @ 1.5.6
 - tool-avrdude @ 1.70100.0 (7.1.0)
 - toolchain-atmelavr @ 3.70300.220127 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf       
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 18 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Error: invalid `bootloader.class` in board config!
====================== [FAILED] Took 0.47 seconds ====================== 

Environment       Status    Duration
----------------  --------  ------------
Upload_UPDI       IGNORED
Upload_UART       IGNORED
fuses_bootloader  FAILED    00:00:00.471
================= 1 failed, 0 succeeded in 00:00:00.471 =================
C:\Users\uliano\stuff\Platformio\avr128da32-breakout
> pio run -e fuses_bootloader -t bootloader -v
Processing fuses_bootloader (upload_protocol: serialupdi; upload_port: COM4; upload_flags: ; board_hardware.bod: 2.7v; board_hardware.eesave: yes; board_hardware.uart: ser2; board_hardware.rstpin: reset; platform: atmelmegaavr; framework: arduino; board: AVR128DA32; board_build.variant: 32pin-standard; build_unflags: ; build_flags: ; monitor_speed: 115200; monitor_port: COM3; monitor_dtr: 0)
-------------------------------------------------------------------------
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/AVR128DA32.html
PLATFORM: Atmel megaAVR (1.8.0) > AVR128DA32
HARDWARE: AVR128DA32 24MHz, 16KB RAM, 128KB Flash
PACKAGES:
 - framework-arduino-megaavr-dxcore @ 1.5.6
 - tool-avrdude @ 1.70100.0 (7.1.0)
 - toolchain-atmelavr @ 3.70300.220127 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 18 compatible libraries
Scanning dependencies...
No dependencies
Building in release mode
Error: invalid `bootloader.class` in board config!
====================== [FAILED] Took 0.48 seconds ======================

Environment       Status    Duration
----------------  --------  ------------
Upload_UPDI       IGNORED
Upload_UART       IGNORED
fuses_bootloader  FAILED    00:00:00.484
================= 1 failed, 0 succeeded in 00:00:00.484 =================
C:\Users\uliano\stuff\Platformio\avr128da32-breakout
```

SpenceKonde, 2023-04-30:
Edited title to clarify that this is a PIO specific issue and added the "help needed" and "third party IDE" labels. 

I don't know much about platform IO other than the fact that it ignores parts of my platform.txt and boards.txt files that are supposed to prevent, wel, problems kinda like this

SpenceKonde, 2023-04-30:
You do have serial wired up to PF0 and PF1 right, not PA0 and PA1? And you did connectan autoreset circuit?

uliano, 2023-04-30:
Yes, I connected PF0 and PF1.

The autoreset is a 100n capacitor between DTR* and PF6 and a 10k pull-up on PF6 (the same is working for ATmega4808).

It is not only a Platformio thing, as stated in the OP also Arduino 1.8.18 fails to burn a *FUNCTIONAL* bootloader.

SpenceKonde, 2023-04-30:
Can you get any more information on that latter issue for me and put it in a new issue? This is a critical issue, especially since a version of the core is being released in a few days. 

Does that problem occur on the "good" version of the Arduino IDE, 1.8.13?

uliano, 2023-05-01:
There is no issue with Arduino IDE, it was just a FT232 clone not working as expected. I edited the OP to report only the Platformio problem (which is still there)

uliano, 2023-05-01:
In the file `.platformio/platforms/atmelmegavr/builder/bootloader.py` in line 47 and following we read

```
def get_bootloader_dxcore(framework_dir, board_config):
    btld = board_config.get("bootloader.class", "")
    port = board_config.get("bootloader.port", "")
    entry = board_config.get("bootloader.entrycond", "")

    if not btld:
        sys.stderr.write("Error: invalid `bootloader.class` in board config!\n")
        env.Exit(1)
    if not port:
        sys.stderr.write("Error: invalid `bootloader.port` in board config!\n")
        env.Exit(1)

    bootloader_file = f"{btld}_{port}_{entry}.hex" if entry else f"{btld}_{port}.hex"
```

therefore I modified the last part of the platformio.ini as following
```
[env:fuses_bootloader]
; Upload protocol for used to set fuses/bootloader
upload_protocol = ${env:Upload_UPDI.upload_protocol}
upload_port = ${env:Upload_UPDI.upload_port}
upload_flags =
; Hardware settings
board_hardware.bod = 2.7v
board_hardware.eesave = yes
;board_hardware.uart = ser2 ;this is megacoreX parameter
board_hardware.rstpin = reset
; bootloader selection
board_bootloader.class = optiboot_128dx
board_bootloader.port = ser2
board_bootloader.entrycond = extr
```
and was able to burn *some* bootloader and got the usual triple blinking.

However trying to upload a sketch on the selected serial port I get:

```
Configuring upload protocol...
AVAILABLE: arduino
CURRENT: upload_protocol = arduino
Looking for upload port...
Using manually specified: COM3
Uploading .pio\build\Upload_UART\firmware.hex

avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e970a (probably avr128da28)
avrdude: reading input file .pio\build\Upload_UART\firmware.hex for flash
         with 2174 bytes in 1 section within [0x200, 0xa7d]
         using 5 pages and 386 pad bytes
avrdude: writing 2174 bytes flash ...

Writing | ################################################## | 100% 0.24s

avrdude: 2174 bytes of flash written
avrdude: verifying flash memory against .pio\build\Upload_UART\firmware.hex

Reading | ################################################## | 100% 0.18s

avrdude warning: verification mismatch
        device 0xff != input 0x55 at addr 0x0200 (error)
avrdude error: verification mismatch

avrdude done.  Thank you.

*** [upload] Error 1
========================================================= [FAILED] Took 3.74 seconds =========================================================

Environment    Status    Duration
-------------  --------  ------------
Upload_UART    FAILED    00:00:03.739
=================================================== 1 failed, 0 succeeded in 00:00:03.739 =================================================== 
```

the fact that the bootloader is nonfunctional is testified also by Arduino IDE which reports:

```
Sketch uses 2150 bytes (1%) of program storage space. Maximum is 130560 bytes.
Global variables use 251 bytes (1%) of dynamic memory, leaving 16133 bytes for local variables. Maximum is 16384 bytes.
avrdude: verification error, first mismatch at byte 0x0200
         0xff != 0x55
avrdude: verification error; content mismatch
avrdude: verification error; content mismatch
```

Now, funny thing is: if I burn the bootloader with Arduino then *apparently* platformio uploads the sketch correctly but, alas, the sketch doesn't start (I still see the bootloader's triple blink).

SpenceKonde, 2023-05-02:
Can you post the first line of the .hex file you're trying to upload?

uliano, 2023-05-03:
Here it is

```
:1002000055C0000086C0000084C0000082C000000D
```

SpenceKonde, 2023-05-20:
Hmmph, that looks valid though.

Ugh, did you sort this out? If not, full flash dump from the working and non-working devices including values of all fuses.

SpenceKonde, 2023-06-24:
No response in over a month, closing.

closed 2023-06-24 by SpenceKonde