SpenceKonde/DxCore #440: Invalid signature on Optiboot AVR32DB28?

nomakewan, 2023-06-06:
I'm back again.

Using UPDI with my AVR32DB28 works perfectly fine. I can upload programs and change fuses no problem, code always verifies, all sketches work perfectly, all's right with the world. But I was getting annoyed having to switch wires around to get serial monitor access after programming, so I thought to throw Optiboot on. I selected AVR32DB28 (Optiboot), left all the non-optiboot settings identical to my UPDI version, changed only the Optiboot serial port (I changed it to USART0 default pins as I already had them wired for the serial monitor), and burned bootloader. That much works perfectly.

When I try to upload a sketch, however, the IDE does its serial handshake then prints this error:

"Expected signature for AVR32DB28 is 1E 95 37. Double check chip, or use -F to override this check."

The AVR32DB28 was bought brand new from Digikey in March 2022, so I wouldn't expect a bootleg (are there even bootlegs of these chips floating around yet?).

Am I misunderstanding something again, or is something wonky with the signature check for the Optiboot AVR32DB28?

nomakewan, 2023-06-08:
I have done a little more testing. I now have Optiboot "working" on my AVR32DB28 by changing the signature definition in avrdude.conf so that it's 1E 94 37 instead, and have changed the bootloader entry point to be 'always' as setting it to any other causes my code to run from where it left off in loop() rather than restarting from the beginning as expected. With the entry vector set to "always" it now works as expected, restarting when powered back up.

I have attached two verbose logs. One shows burning the bootloader, which is successful and also returns the expected signature of 1E 95 37. The other is a failed write via serial, showing the incorrect returned signature.

[burnbootloadersuccess.txt](https://github.com/SpenceKonde/DxCore/files/11682903/burnbootloadersuccess.txt)
[badsignature.txt](https://github.com/SpenceKonde/DxCore/files/11682905/badsignature.txt)

SpenceKonde, 2023-06-11:
Fixed. 

If we can save an LDS (3 clocks 2 words, and use an LDI instead (1 clock 1 word), which we can do on the first two bytes of signature always because we know the size of the overall signature space and don;rt reuse binaries for 16 and 32k parts anyway. That's what I did when I had rewritten a large chunk in asm and saved like 3 words - saving two words just like that? If that wasn't what pushed me over the threshold to getting it to fit in min size bootloader (which was an extremely high priority, because unlike classic AVRs, if the compiler thinks the bootloader starts at 0x0400 and the bootloader starts at 0x200, or vise versa, the sketch will go off the rails the first time an interrupt (likely millis) fires. So if you make the bootloadercompile to 512b instead of 1024b, you want that as few versions to have gotten the 1024b version as possible, because all of those people need to rebootload all their boards with the 512b bootloader ro use them with the new core version). it got me with in 2-4 words of it. I was *desperate* fora few more bytes of flas at that point Absolutely desperate

So, hey I know that I have to compile a different binary if it's 16/32k ,vs 64 vs 128k, and since we know that when compiling, we returned 0x95, 96, or 97. However, when I extended it to work on DD-series parts, I broke that section for 32/16k parts because I was testing for a #define that was provided only in the context of compiling a sketch

commit 953ebc8b (2023-06-11): Fix #440
closed 2023-06-11 by SpenceKonde