SpenceKonde/DxCore #448: Docs: Microchip has blocked our link checker.

SpenceKonde, 2023-06-18:
This leaves me little choice but to *remove the relevant links* because since microchip is constantly moving resources around, and you  cant have a core where half the links are broken and you're not even aware of it and expect anyone to take you seriously. So I'm left thinking that the only approach is likely to be eliminating all clickable links. Anyone would take one look at the core and when the second or third informational link they clicked was broken , they'll be like "Okay, this one's a no go.: and close the browser tab. As long as the links are known not broken, removing clickable links It makes the core docs less useful, reduces traffic to Microchip's website. - bad for everyone. 
But the moment a site reorg moves all the docs (has happened multiple times), without a link checker, (because microchip now blocks it) I don't even know there's problem, but the  the repo docs would suddenly be full of dead links, which makes it look unmaintained. Some would see the commit dates, some wouldn't and those people would be like "okay, damn, no maintained AVR DB core, I guess I'll have to use some other company's microcontroller. Maybe STM? I hear about those a lot." and lord knows that's not in any of our interests.  And because broken links are so universally recongized as a sign of neglect, people tend not to report them when the find trhem and just leave, on the assumption that why bother reporting when they aren't maintaining it anymore? (again, the fact that their impression is incorrect is really the problem - but that's the impression that people get when they see broken links, and why we have a check-links test.

WestfW, 2023-06-18:
Many "clickable" links end up being "search for xxx at site yyy"; I used to think this was just lazy, but perhaps it's an attempt to make the usefulness of the link more permanent.

MX682X, 2023-06-19:
Make a local copy of the PDFs and put them into a seperate repository. The links in the docs will point to the repository with the pdfs. Archiving public documents should fall under fair use, otherwise I can't imagine how the wayback machine is hosting a lot of PDFs (like the I2C specification)

SpenceKonde, 2023-07-15:
Yaaaargh. 

Now Arduino is blocking that, so all my libraries now fail too. Because a library fails the test if thereisnt a valid url. But there is a valid url - for any IP except the github actions ones.

nomakewan, 2023-07-15:
> Yaaaargh.
> 
> Now Arduino is blocking that, so all my libraries now fail too. Because a library fails the test if thereisnt a valid url. But there is a valid url - for any IP except the github actions ones.

I noticed actually that the links in this repo and the actual links for Arduino are different. Yes, on a human being endpoint the link automatically redirects, but what about updating all this repo's links to the newer format? Maybe without the redirect they'll go through?

As an example:

This repo: `http://www.arduino.cc/en/Reference/EEPROM`
Real URL: `https://docs.arduino.cc/learn/built-in-libraries/eeprom`

SpenceKonde, 2024-09-08:
Thanks, this was fixed by that change.

closed 2024-09-08 by SpenceKonde