SpenceKonde/DxCore #453: Is there a way to change the optimization options for the compiler?

ProxyPlayerHD, 2023-07-14:
Because the chips have a lot of flash, being able to squeeze out a bit more speed at the cost of some flash seem worth it.
And i could've sworn the Tools menu in the IDE had options to change the Compiler flags (`-Os`, `-O3`, etc) but i can't find them.

looking into the boards.txt file i can see the options there but they're commented out. why? did changing the flags break something?

SpenceKonde, 2023-07-14:
The tools compiler option was only in form a single version, when it was
discovered that any optimizaion option other than -Os would fail to compile
any sketch with serial and I couldn't see a way around it without removing
the biggest optimization ever added to serial. Since the optimization menu
worked out to a "compile successfully" vs "fail to compile" menu, it was
removed

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2023-07-14:
Closing. No plans to change, we tried options to change optimization, and it didn't work. 

The compiler couldn't optimize it's way out of a wet paper bag anyway though so you aren't missing much. Really when you get to the point that you care about the optimization level.... you're usually past the point where the compiler is smart enough to help. GCC is brilliant, clever, and incredibly stubborn, and as intended, it converts code into an intermediate and equivalent internal form (which is why almost all attempts to give the compiler a push in the right direction are doomed to failure). This representation eventually gets passed off to the avr- half of avr-gcc, which proceeds to do a poor job of rendering it as avr machine code.

closed 2023-07-14 by SpenceKonde

ProxyPlayerHD, 2023-07-14:
yea ok, fair enough.
i was mainly just wondering why it was removed, and i got my answer... because it breaks stuff.