SpenceKonde/DxCore #471: Arduino IDE 2.x features like Suggestions and hover text stop working when using DxCore Boards

ProxyPlayerHD, 2023-08-12:
title says it.
for some reason none of those more advanced features of the IDE function correctly or at all when i have a Dx board selected.

for example, i'm using a circular buffer library which has tons of functions, when i type the name of one of my buffer variables followed by a dot i should be seeing suggestions for the different functions i can choose.

when the selected board is any of the DxCore ones it doesn't work and just shows random crossed out constants, variable, and keywords from my program:
[image]

but when i select any other board (built-in or not) like Arduino Mega, generic STM32F0, raspberry pi pico, etc. it works perfectly fine:
[image]

same with the hover text (ie when you hover the cursor over a variable or function it shows you it's declaration, type, input parameters, etc.)

i thought these were universal IDE features regardless of the board selected... so why would this be happening?
or does each board need to implement or enable these features seperately?
either way i'm pretty confused

SpenceKonde, 2023-08-12:
Beats me. Apparently yes they do need to be enabled by the core since it
doesn't work out of the box. I have not downloaded any 2.x.x versions and
do not know what is needed to fix this

____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore: Arduino support for almost every ATTiny microcontroller
Contact: ***@***.***

SpenceKonde, 2023-10-06:
Please retest in latest version of the IDE once a latest version of the IDE is available which works (2.2.1 is no good) and reopen if this issue is still there.

closed 2023-10-06 by SpenceKonde

ProxyPlayerHD, 2023-11-01:
sorry for the late reply, i just tried it in 2.2.2 (nightly build 01.11.2023) and it still won't function. :(