SpenceKonde/DxCore #474: Can't compile for AVR128DA48 - cannot resolve FQBN

grandaspanna, 2023-09-03:
Arduino 1.8.19 and DxCore 1.5.6

Can't seem to compile for my target AVR128DA48 (no bootloader):

```
Error resolving FQBN: getting

Error compiling for board AVR DA-series (no bootloader).

```

I'm assuming this might be an installation issue, but where do I start?

grandaspanna, 2023-09-04:
Fixed it - I had both 1.5.6 and 1.5.8 in the megaavr folder. No idea how that happened. I deleted the 1.5.6 directory tree and things appear ok for now.

grandaspanna, 2023-09-04:
Yay - compiles and uploads goodly.

closed 2023-09-04 by grandaspanna

SpenceKonde, 2023-09-05:
What a fucked up whackjob IDE we have. 

Yeah I don't bother trying to unpick the mess when board manager screws up, I just blow away arduino15 or yjthe subdirs of portable.