SpenceKonde/DxCore #477: USING TIMERA0 WITh AVRXXDD20 get error

LaurentR59, 2023-09-17:
Hi

Suspicious bug when using AVR32DD20 and AVR64dd20 with MILLIS_USE_TIMERA0 option

Issue between wiring.h and timer.h (only for 20PINS ITEMS)

It look like CPU speed is ignored to set the right value. (divide = 64) but move to 8  So why?

Tested under platformio with latest DX_CORE FILES in the env package 1.5.8

unbuild_flags = -D MILLIS_USE_TIMERB1
build_flags = -D MILLIS_USE_TIMERA0

Thats only for 20PINS ITEMS
No issue on larger PIN parts items

Unable to compile for 20PINS items. OK with largest pins parts.

Laurent

LaurentR59, 2023-09-17:
Hi
A describes in the relativ section issue forum for MAGATINYCORE the behavior is similar on ATTINY SERIES1 parts BUT it s OK with SERIES 2 using MILLIS_USE_TIMERA0

#error "wiring.c and timers.h want to set millis timer TCA0 to different divider"

https://github.com/SpenceKonde/megaTinyCore/issues/1002

So it may heltp to find the root cause?

Laurent

LaurentR59, 2023-09-17:
Hi
 I found a way using MILLIS_USER_TIMERD0 on available parts that got it. (3217 & 3216)

But may in link with https://github.com/SpenceKonde/DxCore/blob/master/megaavr/extras/Ref_Timers.md and timers usages

Laurent

LaurentR59, 2023-09-17:
Hi

Workaround is to use set MILLIS_USET_IMERD0 for 20 PINS PARTS to make it OK

SpenceKonde, 2023-09-17:
Hm, I cannot reproduce this problem, can you post code that demonstrates this problem, and all the settings you're using?

I'm wondering if you're using an IDE that isn't Arduino, hence don't have the submenus and relies on the user to enter all the options manually, and ensure that no mutually exclusive options are being passed. the core is designed for the Arduino IDE and as such the UX is designed to be optimal on the IDE. No testing is performed on third party IDEs, but I understand that most problems come from the fact that Arduino IDE has tools submenus, and I make extensive use of them to set core options. Other IDEs do not emulate this feature. The result is that it is possible to pass unsupported combinations of options accidentally using other IDEs - like, for options we don;t have an entry in the menu for and don;'t handle in the core. Setting the millis timer TCA0 on a 3217 and a 32DD20 works without issue for me with an empty sketch, and not knowing what you were trying to do I don;'t know where to go from there. 

My verbose compile output (in case it helps you spot missing/errant define is
```
C:\arduino-1.8.13_AZD5test\arduino-builder -dump-prefs -logger=machine -hardware C:\arduino-1.8.13_AZD5test\hardware -hardware C:\Users\Spence\AppData\Local\Arduino15\packages -hardware C:\Users\Spence\Documents\Arduino\hardware -tools C:\arduino-1.8.13_AZD5test\tools-builder -tools C:\arduino-1.8.13_AZD5test\hardware\tools\avr -tools C:\Users\Spence\AppData\Local\Arduino15\packages -built-in-libraries C:\arduino-1.8.13_AZD5test\libraries -libraries C:\Users\Spence\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy7:chip=3217,clock=16internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=timera,resetpin=UPDI,startuptime=8,wiremode=mors,printf=default,PWMmux=A_default,attach=allenabled,WDTtimeout=disabled,WDTwindow=disabled -ide-version=10813 -build-path C:\Users\Spence\AppData\Local\Temp\arduino_build_834720 -warnings=all -prefs=build.warn_data_percentage=75 -verbose C:\Users\Spence\AppData\Local\Temp\untitled1396808297.tmp\sketch_sep17a\sketch_sep17a.ino
C:\arduino-1.8.13_AZD5test\arduino-builder -compile -logger=machine -hardware C:\arduino-1.8.13_AZD5test\hardware -hardware C:\Users\Spence\AppData\Local\Arduino15\packages -hardware C:\Users\Spence\Documents\Arduino\hardware -tools C:\arduino-1.8.13_AZD5test\tools-builder -tools C:\arduino-1.8.13_AZD5test\hardware\tools\avr -tools C:\Users\Spence\AppData\Local\Arduino15\packages -built-in-libraries C:\arduino-1.8.13_AZD5test\libraries -libraries C:\Users\Spence\Documents\Arduino\libraries -fqbn=megaTinyCore:megaavr:atxy7:chip=3217,clock=16internal,bodvoltage=1v8,bodmode=disabled,eesave=enable,millis=timera,resetpin=UPDI,startuptime=8,wiremode=mors,printf=default,PWMmux=A_default,attach=allenabled,WDTtimeout=disabled,WDTwindow=disabled -ide-version=10813 -build-path C:\Users\Spence\AppData\Local\Temp\arduino_build_834720 -warnings=all -prefs=build.warn_data_percentage=75 -verbose C:\Users\Spence\AppData\Local\Temp\untitled1396808297.tmp\sketch_sep17a\sketch_sep17a.ino
Using board 'atxy7' from platform in folder: C:\Users\Spence\Documents\Arduino\hardware\megaTinyCore\megaavr
Using core 'megatinycore' from platform in folder: C:\Users\Spence\Documents\Arduino\hardware\megaTinyCore\megaavr
Detecting libraries used...
"C:\\arduino-1.8.13_AZD5test\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny3217 -DF_CPU=16000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERA0 -DCORE_ATTACH_ALL -DUSE_TIMERD0_PWM -DARDUINO=10813 -DARDUINO_AVR_ATtiny3217 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.6.8.1\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=8UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy7 "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\variants\\txy7" "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_834720\\sketch\\sketch_sep17a.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE
Generating function prototypes...
"C:\\arduino-1.8.13_AZD5test\\hardware\\tools\\avr/bin/avr-g++" -c -g -Os -Wall -std=gnu++17 -fpermissive -Wno-sized-deallocation -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -mrelax -w -x c++ -E -CC -mmcu=attiny3217 -DF_CPU=16000000L -DCLOCK_SOURCE=0 -DTWI_MORS -DMILLIS_USE_TIMERA0 -DCORE_ATTACH_ALL -DUSE_TIMERD0_PWM -DARDUINO=10813 -DARDUINO_AVR_ATtiny3217 -DARDUINO_ARCH_MEGAAVR "-DMEGATINYCORE=\"2.6.8.1\"" -DMEGATINYCORE_MAJOR=2UL -DMEGATINYCORE_MINOR=6UL -DMEGATINYCORE_PATCH=8UL -DMEGATINYCORE_RELEASED=1 -DARDUINO_attinyxy7 "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore/api/deprecated" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\cores\\megatinycore" "-IC:\\Users\\Spence\\Documents\\Arduino\\hardware\\megaTinyCore\\megaavr\\variants\\txy7" "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_834720\\sketch\\sketch_sep17a.ino.cpp" -o "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_834720\\preproc\\ctags_target_for_gcc_minus_e.cpp" -D [... cut ...]
cmd /C echo. && "C:\\arduino-1.8.13_AZD5test\\hardware\\tools\\avr/bin/avr-nm" --numeric-sort --line-numbers --demangle --print-size --format=s "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_834720/sketch_sep17a.ino.elf" > "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_834720/sketch_sep17a.ino.map"
 
"C:\\arduino-1.8.13_AZD5test\\hardware\\tools\\avr/bin/avr-size" -A "C:\\Users\\Spence\\AppData\\Local\\Temp\\arduino_build_834720/sketch_sep17a.ino.elf"
Sketch uses 374 bytes (1%) of program storage space. Maximum is 32768 bytes.
Global variables use 10 bytes (0%) of dynamic memory, leaving 2038 bytes for local variables. Maximum is 2048 bytes.
```

LaurentR59, 2023-09-17:
Hi

Thanks for your elements. That s really strange anyway.

Here is mine:

PLATFORMIO 6.1.11 HOME 3.4.4 WINDOWS11 64bits 22H2

Latest file from MEGATINY CORE pushed on packages diretory

INI FILE:

[env:ATTINYX7]

;//FOR ATTINY SERIE 1 & 2:  ALL SERIE 1 & 2 PARTS
platform_packages = framework-arduino-megaavr-megatinycore
                    platformio/toolchain-atmelavr@^3.70300.220127
                    platformio/tool-avrdude-megaavr@^3.60300.220118

; 24PINS ATTINY SERIE 1 FAMILY:
platform = atmelmegaavr
framework = arduino

board = ATtiny3217
board_build.variant = txy7 
board_build.mcu = ATTINY3217

build_unflags = -D MILLIS_USE_TIMERB1

;build_flags = -v -D MILLIS_USE_TIMERD0 ;//OK
build_flags = -v -D MILLIS_USE_TIMERA0 ;//KO

sketch file is empty like this:

#include <Arduino.h>

// put function declarations here:
int myFunction(int, int);

void setup() {
  // put your setup code here, to run once:
  int result = myFunction(2, 3);
}

void loop() {
  // put your main code here, to run repeatedly:
}

// put function definitions here:
int myFunction(int x, int y) {
  return x + y;
}

Here is th logs files (part of it):

WITH ERROR MESSAGE:

pio\build\ATTINYX7\FrameworkArduino\api\PluggableUSB.cpp.o C:\Users\LAUREN~1\AppData\Local\Temp\cci8q0Ag.s
COMPILER_PATH=c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../libexec/gcc/avr/7.3.0/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../libexec/gcc/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/
LIBRARY_PATH=c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/avrxmega3/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/avrxmega3/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/
COLLECT_GCC_OPTIONS='-o' '.pio\build\ATTINYX7\FrameworkArduino\UART0.cpp.o' '-c' '-std=gnu++1z' '-fpermissive' '-fno-exceptions' '-fno-threadsafe-statics' '-Wno-sized-deallocation' '-Wno-error=narrowing' '-v'  '-Os' '-w' '-ffunction-sections' '-fdata-sections' '-flto' '-mrelax' '-D' 'PLATFORMIO=60111' '-D' 'ARDUINO_attinyxy7' '-D' 'MILLIS_USE_TIMERD0' '-D' 'UARTBAUD5V' '-D' 'MILLIS_USE_TIMERA0' '-D' 'F_CPU=16000000L' '-D' 'ARDUINO_ARCH_MEGAAVR' '-D' 'ARDUINO=10808' '-D' 'CLOCK_SOURCE=0' '-D' 'MEGATINYCORE="2.6.7"' '-D' 'MEGATINYCORE_MAJOR=Compiling .pio\build\ATTINYX7\FrameworkArduino\new.cpp.o
2UL' '-D' 'MEGATINYCORE_MINOR=6UL' '-D' 'MEGATINYCORE_PATCH=7UL' '-D' 'MEGATINYCORE_RELEASED=1' '-D' 'CORE_ATTACH_ALL' '-D' 'TWI_MORS' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\cores\megatinycore\api\deprecated' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\cores\megatinycore' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\variants\txy7' '-specs=device-specs/specs-ATTINY3217' '-mmcu=avrxmega3'
COMPILER_PATH=c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../libexec/gcc/avr/7.3.0/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../libexec/gcc/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/
LIBRARY_PATH=c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/avrxmega3/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/avrxmega3/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/
COLLECT_GCC_OPTIONS='-o' '.pio\build\ATTINYX7\FrameworkArduino\UART.cpp.o' '-c' '-std=gnu++1z' '-fpermissive' '-fno-exceptions' '-fno-threadsafe-statics' '-Wno-sized-deallocation' '-Wno-error=narrowing' '-v'  '-Os' '-w' '-ffunction-sections' '-fdata-sections' '-flto' '-mrelax' '-D' 'PLATFORMIO=60111' '-D' 'ARDUINO_attinyxy7' '-D' 'MILLIS_USE_TIMERD0' '-D' 'UARTBAUD5V' '-D' 'MILLIS_USE_TIMERA0' '-D' 'F_CPU=16000000L' '-D' 'ARDUINO_ARCH_MEGAAVR' '-D' 'ARDUINO=10808' '-D' 'CLOCK_SOURCE=0' '-D' 'MEGATINYCORE="2.6.7"' '-D' 'MEGATINYCORE_MAJOR=2UL' '-D' 'MEGATINYCORE_MINOR=6UL' '-D' 'MEGATINYCORE_PATCH=7UL' '-D' 'MEGATINYCORE_RELEASED=1' '-D' 'CORE_ATTACH_ALL' '-D' 'TWI_MORS' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\cores\megatinycore\api\deprecated' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\cores\megatinycore' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\variants\txy7' '-specs=device-specs/specs-ATTINY3217' '-mmcu=avrxmega3'
COLLECT_GCC_OPTIONS='-o' '.pio\build\ATTINYX7\FrameworkArduino\api\Print.cpp.o' '-c' '-std=gnu++1z' '-fpermissive' '-fno-exceptions' '-fno-threadsafe-statics' '-Wno-sized-deallocation' '-Wno-error=narrowing' '-v'  '-Os' '-w' '-ffunction-sections' '-fdata-sections' '-flto' '-mrelax' '-D' 'PLATFORMIO=6 [... cut ...] tinycore\cores\megatinycore\api\deprecated' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\cores\megatinycore' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\variants\txy7' '-specs=device-specs/specs-ATTINY3217' '-mmcu=avrxmega3'        
====================================================================== [FAILED] Took 4.13 seconds ======================================================================

LaurentR59, 2023-09-17:
WITH SUCCES On TIMERD0

S_USE_TIMERD0' '-D' 'UARTBAUD5V' '-D' 'MILLIS_USE_TIMERD0' '-D' 'F_CPU=16000000L' '-D' 'ARDUINO_ARCH_MEGAAVR' '-D' 'ARDUINO=10808' '-D' 'CLOCK_SOURCE=0' '-D' 'MEGATINYCORE="2.6.7"' '-D' 'MEGATINYCORE_MAJOR=2UL' '-D' 'MEGATINYCORE_MINOR=6UL' '-D' 'MEGATINYCORE_PATCH=7UL' '-D' 'MEGATINYCORE_RELEASED=1' '-D' 'CORE_ATTACH_ALL' '-D' 'TWI_MORS' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packaCompiling .pio\build\ATTINYX7\FrameworkArduino\new.cpp.o
ges\framework-arduino-megaavr-megatinycore\cores\megatinycore\api\deprecated' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\cores\megatinycore' '-I' 'C:\Users\Laurent & Compiling .pio\build\ATTINYX7\FrameworkArduino\wiring.c.o
Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\variants\txy7' '-specs=device-specs/specs-ATTINY3217' '-mmcu=avrxmega3'
Compiling .pio\build\ATTINYX7\FrameworkArduino\wiring_analog.c.o
COMPILER_PATH=c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../libexec/gcc/avr/7.3.0/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../libexec/gcc/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/
LIBRARY_PATH=c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/avrxmega3/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/avrxmega3/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/
COLLECT_GCC_OPTIONS='-o' '.pio\build\ATTINYX7\FrameworkArduino\UART.cpp.o' '-c' '-std=gnu++1z' '-fpermissive' '-fno-exceptions' '-fno-threadsafe-statics' '-Wno-sized-deallocation' '-Wno-error=narrowing' '-v'  '-Os' '-w' '-ffunction-sections' '-fdata-sections' '-flto' '-mrelax' '-D' 'PLATFORMIO=60111' '-D' 'ARDUINO_attinyxy7' '-D' 'MILLIS_USE_TIMERD0' '-D' 'UARTBAUD5V' '-D' 'MILLIS_USE_TIMERD0' '-D' 'F_CPU=16000000L' '-D' 'ARDUINO_ARCH_MEGAAVR' '-D' 'ARDUINO=10808' '-D' 'CLOCK_SOURCE=0' '-D' 'MEGATINYCORE="2.6.7"' '-D' 'MEGATINYCORE_MAJOR=2UL' '-D' 'MEGATINYCORE_MINOR=6UL' '-D' 'MEGATINYCORE_PATCH=7UL' '-D' 'MEGATINYCORE_RELEASED=1' '-D' 'CORE_ATTACH_ALL' '-D' 'TWI_MORS' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\cores\megatinycore\api\deprecated' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\cores\megatinycore' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\variants\txy7' '-specs=device-specs/specs-ATTINY3217' '-mmcu=avrxmega3'
COMPILER_PATH=c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../libexec/gcc/avr/7.3.0/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../libexec/gcc/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/
LIBRARY_PATH=c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/avrxmega3/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/avrxmega3/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/
COLLECT_GCC_OPTIONS='-o' '.pio\build\ATTINYX7\FrameworkArduino\UART1.cpp.o' '-c' '-std=gnu++1z' '-fpermissive' '-fno-exceptions' '-fno-threadsafe-statics' '-Wno-sized-deallocation' '-Wno-error=narrowing' '-v'  '-Os' '-w' '-ffunction-sections' '-fdata-sections' '-flto' '-mrelax' '-D' 'PLATFORMIO=60111' '-D' 'ARDUINO_attinyxy7' '-D' 'MILLIS_USE_TIMERD0' '-D' 'UARTBAUD5V' '-D' 'MILLIS_USE_TIMERD0' '-D' 'F_CPU=16000000L' '-D' 'ARDUINO_ARCH_MEGAAVR' '-D' 'ARDUINO=10808' '-D' 'CLOCK_SOURCE=0' '-D' 'MEGATINYCORE="2.6.7"' '-D' 'MEGATINYCORE_MAJOR=2UL' '-D' 'MEGATINYCORE_MINOR=6UL' '-D' 'MEGATINYCORE_PATCH=7UL' '-D' 'MEGATINYCORE_RELEASED=1' '-D' 'CORE_ATTACH_ALL' '-D' 'TWI_MORS' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\cores\megatinycore\api\deprecated' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\cores\megatinycore' '-I' 'C:\Users\Laurent & Isabelle\.platformio\packages\framework-arduino-megaavr-megatinycore\variants\txy7' '-specs=device-specs/specs-ATTINY3217' '-mmcu=avrxmega3'
COMPILER_PATH=c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../libexec/gcc/avr/7.3.0/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../libexec/gcc/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/
LIBRARY_PATH=c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/avrxmega3/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/../../../../avr/lib/avrxmega3/;c:/users/laurent & isabelle/.platformio/packages/toolchain-atmelavr/bin/../lib/gcc/avr/7.3.0/;c:/users/laurent & isabell [... cut ...]
I should not be alone under PLATFORIO or VS CODE with that behavior.

About ARDUINO IDE you r right, the menu option do not offer according SERIES 0, 1 or  2 the same default timers avalibility
So some combo are not availaible through this ARDUINO IDE TOOL.
BUT there is nothing to avoid "unsupported" combo oustide this official core getting the same kind of error message

As exempple TIMERA0 can not be set with SERIES 1 on ARDUINO IDE but could by option line in third party IDE. (o require to be limited?)

Regards
Laurent

SpenceKonde, 2023-09-18:
In the failed compile there's a clear issue: 

`'MILLIS_USE_TIMERD0' '-D' 'UARTBAUD5V' '-D' 'MILLIS_USE_TIMERA0`

You have two MILLIS_USE directives! Millis uses one and only one timer at a time. 

Also, good lord, the UARTBAUDxV mechanism was ripped out over a year ago, The UARTBAUD5V is a no-op - the code that it controlled is long gone. Their cal is generally dead on at room temp over the operating voltage range and difference between 3v and 5v adjustments were tiny (and if you weren't at room temp this correction was no good) - but doing that UART baud correction calculation was painful. In involved long division - and it happened to be in a function where even without it the compiler struggles to create decent output because of register pressure. I had to restructure the uncorrected code several times to avoid pathological binaries

LaurentR59, 2023-09-19:
Hi

All done!
Good fine troubleshoot on root cause.
May be not so easy to fix easily? So workaround is to set  with limited value the  timer(s) for MILLIS according CPU modele.

Could it be possible to add an option to bypass this part of compilation step? (hotfix way!) 

Anyway good luck to fix it (again!)

Laurent

SpenceKonde, 2023-09-19:
looking again at the top, I think the problem is the "unbuild MILLIS_USE_TIMERB1" That., if it does what I think works only when MILLIS_USE_TIMERB1 would normally be passed as a build parameter. That's true on the tinyAVR 2-series,. which all have a second TCB1 and don't have a hell-to-use TCD (that's what the 1-series defaults to)

But here's the real problem in "fixing" this thing which is not a bug, but rather a user error in failing to provide correct defines when choosing to use a third party IDE which we do not officially support, and which does not ensure that only valid combinations of defines are passed. If you use such an IDE, it is your responsibility to specifty the correct defines

This would also be a BLOODY NIGHTMARE TO CHANGE. Look how we do it:

Note: Most of the lines in the file are irrelevant and removed - these are only the relevant lines of boards.txt - there are about 2-6 deleted lines per relevant-to-millis-timer lines within the sections relating to it....

```
atxy7.build.millistimer=D0

atxy7.menu.chip.3227.build.millistimer=B1
atxy7.menu.chip.1627.build.millistimer=B1
atxy7.menu.chip.1607.build.millistimer=A0
atxy7.menu.chip.827.build.millistimer=B1
atxy7.menu.chip.807.build.millistimer=A0
atxy7.menu.chip.427.build.millistimer=B1

#----------------------------------------#
# MILLIS TIMER                           #
#________________________________________#
#atxy7.menu.millis.enabled.build.millistimer= *** NOT SPECIFIED, Inherits!**
atxy7.menu.millis.disabled.build.millistimer=NONE
atxy7.menu.millis.timera.build.millistimer=A0
atxy7.menu.millis.timerd.build.millistimer=D0
atxy7.menu.millis.timerb0.build.millistimer=B0
atxy7.menu.millis.timerb1.build.millistimer=B1
atxy7.menu.millis.rtc.build.millistimer=RTC
atxy7.menu.millis.rtcxtal.build.millistimer=RTC_XTAL
atxy7.menu.millis.rtcxosc.build.millistimer=RTC_XOSC
```
So, First, we start off with build.millistimer being TCD0. Then if the chip is chosen buit doesn't have a TCD, we change build.millistimer. 

Finally, we override it entirely with menu.millis. 

But that just gives us 2 letters, we need a whole define, where does that happen?! 
In platform.txt:

```
build.optiondefines=-DF_CPU={build.f_cpu} -DCLOCK_SOURCE={build.clocksource} -DTWI_{build.wire} -DMILLIS_USE_TIMER{build.millistimer} {build.attachmode} {build.pwmflags}

```

So uh, you can probably see that that may not be particularly easy to unpick and the behavior is not trivial. Oh, and that part shown above was for 1/13th of the supported parts. Yeah I am not going to open that can of worms, not for bad input on a third party IDE, where the analogous situation is prevented on the Arduino IDE. 

I think my solution will be better error messages that detect invalid options and #error on them with a message like "You have passed two MILLIS_USE define, only one is permitted. If this occurred with the Arduino IDE please report this promptly." And a #warning "Warning: The UARTBAUD5V/UARTBAUD3V baud rate adjustment were removed in 1.3.0 as it degraded performance for negligible improvement in accuracy." if such invalid removed options were specified.

SpenceKonde, 2023-09-19:
So what's tjhis about the AVRxxDD20? The millis timekeeping infrastructure (if not the innermost guts) has some significant differences (for examople no TCD PWM. default is TCB1 or TCB2 (TCB2 if it has one))

SpenceKonde, 2023-09-22:
https://github.com/SpenceKonde/DxCore/commit/146579b4d289d7ad0109f3dab1dffac5da5e829a adds clear error messages when multiple timers are specified for DxCore, same fix will come to mTC.

closed 2023-09-22 by SpenceKonde