SpenceKonde/DxCore #488: Comparator

simjman, 2023-10-07:
I was playing with the comparator reference voltages and couldn't work out why an interrupt wouldn't fire when using vref_2v500, but would when using vref_4v096. 

I believe the issue lies here:

https://github.com/SpenceKonde/DxCore/blob/2a5b6cc07cdd1c0bb7b11fa37b519c76378d1451/megaavr/libraries/Comparator/src/Comparator.h#L189

and here:

https://github.com/SpenceKonde/DxCore/blob/2a5b6cc07cdd1c0bb7b11fa37b519c76378d1451/megaavr/libraries/Comparator/src/Comparator.h#L191

This is from the DA/DB datasheet for reference:

[image]

SpenceKonde, 2023-10-07:
Yup. I hadn't realized that they changed the order on the Dx/Ex series, I thought it was the same as tiny2 (mTC also gets the identical library). 

I belive it was like that since at least 1.3.0, maybe longer, Well over a year. You're the first to notice.

SpenceKonde, 2023-10-07:
Fixed now

closed 2023-10-07 by SpenceKonde

simjman, 2023-10-10:
That's great, glad we caught it!