SpenceKonde/DxCore #490: Optiboot_dx build fails on various alternative baud rates

tkoumrian, 2023-10-16:
I needed 38400 and 57600 baud builds, and these both fail.  Other rates lower than the default 115200 also fail, but some randomly attempted baud rates succeed (64000, 96000).  The build environment was setup according to the Windows build instructions.  The default 115200 succeeds and the build output is the same as the logfiles in the repo.

I'm not sure what's going on, but it looks like the text section is unexpectedly longer than it should be, although I don't understand why in the normal case it doesn't complain since normally at 115200 baud, the text section is 0x1fc bytes which makes me think there should be an overlap complaint with the .smtarg section at 0x1fa.  In the failed 38400 build below, it seems as though the text section is 0x1fb bytes in length (1 byte shorter than the successful 115200 build) but triggers the overlap error with .spmtarg at 0x1fa.

```
Microsoft Windows [Version 10.0.22621.2428]
(c) Microsoft Corporation. All rights reserved.

c:\Todd\arduino-1.0.6-optiboot_dx\hardware\arduino\bootloaders\optiboot_dx>..\..\..\tools\avr\utils\bin\make OS=windows ENV=arduino 128dx_ser2_alt_all_8sec
..\..\..\tools\avr\utils\bin\make OS=windows ENV=arduino 128dx_ser2_alt_all_8sec
Using Compiler at: c:/Todd/arduino-1.0.6-optiboot_dx/hardware/tools/avr/bin/avr-gcc
c:/Todd/arduino-1.0.6-optiboot_dx/hardware/tools/avr/utils/bin/make -f  makefile optiboot_128dx_ser2_alt_all_8sec.hex UARTTX=F4 TIMEOUT=8 ENTRYCOND_REQUIRE=0x35 LED=A7
Using Compiler at: c:/Todd/arduino-1.0.6-optiboot_dx/hardware/tools/avr/bin/avr-gcc
make[1]: Entering directory `c:/Todd/arduino-1.0.6-optiboot_dx/hardware/arduino/bootloaders/optiboot_dx'
../../../tools/avr/bin/avr-gcc -g -Wall -Os -fno-split-wide-types -mrelax  -DWDTTIME=8  -DENTRYCOND_REQUIRE=0x35 -DLED_START_FLASHES=3  -DLED=A7   -DUARTTX=F4 -DBAUD_RATE=38400        -Wl,-section-start=.text=0 -Wl,--section-start=.spmtarg=0x1fa -Wl,--section-start=.version=0x1fe -Wl,--relax -nostartfiles -nostdlib  -mmcu=avr128da64 -o optiboot_128dx_ser2_alt_all_8sec.elf optiboot_dx.c
c:/todd/arduino-1.0.6-optiboot_dx/hardware/tools/avr/bin/../lib/gcc/avr/7.3.0/../../../../avr/bin/ld.exe: section .spmtarg loaded at [000001fa,000001fd] overlaps section .text loaded at [00000000,000001fb]
collect2.exe: error: ld returned 1 exit status
make[1]: *** [optiboot_128dx_ser2_alt_all_8sec.elf] Error 1
make[1]: Leaving directory `c:/Todd/arduino-1.0.6-optiboot_dx/hardware/arduino/bootloaders/optiboot_dx'
make: *** [128dx_ser2_alt_all_8sec] Error 2

c:\Todd\arduino-1.0.6-optiboot_dx\hardware\arduino\bootloaders\optiboot_dx>
```

SpenceKonde, 2023-10-18:
Hmmm. Per my comments in the source:

```
#define BAUD_SETTING_4 (((4000000) * 64) / (16L * BAUD_RATE))
...
  // Saves SIX BYTES! LDI 0 and an STS to the high byte!
  // Why the compiler wastes a word on the ldi 0 is a mystery to me, since it knows it's got a 0 right there in r1 ready to go...
  #if (BAUD_SETTING_4 < 256)
    MYUART.BAUDL = BAUD_SETTING_4;
  #else
    MYUART.BAUD = BAUD_SETTING_4;
  #endif
```
So that accounts for 3 words of flash. 

And previously we had only 2 words of flash free. So yeah - if the value that we write to the baud register doesn't fit in a byte, our trick to skip loading the high byte (which relies on the fact that, despite the fact that they try to hammer home the concept of a temp register, their description of 16-bit registers is - technically accurate but lacking a critical piece of information, namely that *not all pairings of registers with adjacent addresses, wherein both registers have the same name except for the final character being an L on the first and an H on the second, are accessible as a 16-bit value using that name without the postfix and are described the documentation as 16-bit registers are subject to **Accessing 16-bit registers*** 

That only applies to peripherals that have a "TEMP" register. If they don't, then the 16-bit registers are just 2 8-bit registers standing side by side, and you can write the low one and ignore the high one if the high byte is 0, And since the default clock speed is only 4 MHz and obviously we would never use U2X mode if we didn't have to with the fractional baud generator, as long as the speed is between 62500 baud and 250,000 baud, it will work (the minimum value we can write, for maximum speed, is 64 (each bit should be sampled 16 times, and the sampling should be happening this often in 64ths of a system clock cycle minimum 64/64 `*`)

But as you noticed, we didn't have enough space if we did the write of the high byte there - that's a 1 word overlap, by the way - just 1 word. I was able to get an extra 2 words back by restructuring that part of the code to get a pointer to the USART, and pass that to a block of inline asm that could guarantee what it compiled to, and that it would use std to set them not sts. I updated readme to accurately reflect the amount of remaining flash on the builds we provide. That hack is enabled only when we are on a 128k part and we are using the alternate pin, and the baud rate is slow enough to need 2 bytes. 

Anyway, with the changes, your build comes in with 1 word free (see, we *had* to get onto 512b, because the *compiler* needs to know the size of the bootloader to produce a correct binary, so not only do people have to reburn bootloader to get the space back (if we did 1024 then managed to hit 512), but it wouldn't be possible to have one set of tools submenu options which could upload to both, nor is there even a theoretical way to do that without needing to amend the bootloader (there goes fitting it in 512b!) and the upload tool (we would need to have a way for the computer to ask the device which bootloader it had to even understand that that's why things won't work... So it was clear that we had to get below 512b ASAP before it got more painful to fix, and since then a few issues ate away what little space we had.But hey, I got not two but FOUR bytes!

This is the binary I got - untested.
[optiboot_128dx_ser2_alt_all_8sec_38400.hex.txt](https://github.com/SpenceKonde/DxCore/files/12981572/optiboot_128dx_ser2_alt_all_8sec_38400.hex.txt)

`*`  that's how modern AVRs have the same maximum USART baud as classic AVRs - but the modern AVRs really do work all the way up to the maximum, whereas classic AVR's UART scrapes by with a barely-passing-grade at 16 MHz and 115200 baud (to be fair, that is an almost pessimal baud rate for classic AVR. To be less fair, "The most common baud rates are pessimal for a classic AVR's UART at the most common and convenient system clocks!"), limiting practical speed using power-of-two clock at (14400/28800/57600/115200/230400/460800/921600) baud rates, some of the worst baud rates around for a classic AVR, end up effectively limiting the parts to more like F_CPU/64 than F_CPU/8 (the classic AVR could do 1 mbaud at 8 MHz system clock, though it would struggle to keep feeding bytes to it that fast, but it couldn't do 115200. You can combine that with the janky internal oscillator and differences between specimens to contrive situations in which you have like 4 chips and each device can talk to 1-3 of the others, and in some cases the communication is one-directional (the other one seeing only garbage), and one pairing of them can only communicate when it's sunny because the member of that pair near the window warms up in the sun and the internal oscillator speed gets closer to 8 MHz). The really insidious thing about this is that two  classic AVRs will have no trouble communicating no matter what the baud rate they try to use - as long as their clocks are within 3-4% of each other and provided the two AVRs specify the same baud rate and use the same algorithm to calculate what they write to the register, they'll be able to talk. They won't actually be communicating at that baud rate - but they'll both be equally wrong in the same direction, so they can communicate just fine! This has caught people with working installations which they attempt to add a modern AVR to and it can't talk to the other parts (the obvious workaround is to nudge the baud rate up or down on the modern AVR until it works)

(your nicer serial adapters let you specify anything their clock divider can manage; less nice ones may be hardwired to only the common ones, or have a fixed number of slots that you can change (CP2102) with vendor tools. Don't get the CP2102, lacks desirable attributes, has too many undesirable ones)

SpenceKonde, 2023-10-18:
Please let me know if this has resolved things - you should be able to build with the new source like you were doing.

tkoumrian, 2023-10-22:
Great, thanks.  I can now build with custom parameters and the two lower baud rates have tested out as working.  Will optiboot_dx also work for these baud rates on DB parts with smaller memories or would that require a different special case fix?

tkoumrian, 2023-10-29:
Closing since the fix was tested: it builds and produces working bootloaders at the required baud rates.  Thanks.

closed 2023-10-29 by tkoumrian