SpenceKonde/DxCore #497: DD series: Error in digitalPinToAnalogInput macro in pins_arduino.h

nurazur, 2023-11-01:
When using the bootloader, the MVIO Fuse is evaluated as:
`#define IS_MVIO_ENABLED() ((FUSE.SYSCFG1 & 0x01) == 0)`
However the bits defining MVIO in FUSE.SYSCFG1 are bits 3 and 4, where bit 4 is 0 in case MVIO is set for dual supply configuration, and 1 when set to single supply configuration. 
so the correct MVIO evaluation macro is this:
`#define IS_MVIO_ENABLED() ((FUSE.SYSCFG1 & 0x10) == 0)`

Tested for AVR64DD28 and AVR64DD32.  The error is also in AVR64DD20 and AVR64DD14 pin definitions, but I can't test it because I do not have such parts available.

WestfW, 2023-11-02:
Should probably be:
```
#define IS_MVIO_ENABLED() ((FUSE.SYSCFG1 & MVSYSCFG_SINGLE_gc) == 0)
```
Or even
```
#define IS_MVIO_ENABLED() ((FUSE.SYSCFG1 & FUSE_MVSYSCFG_gm) == MVSYSCFG_DUAL_gc)
```

nurazur, 2023-11-02:
Because I do not want to touch the DxCore core files, here is a workaround that works for me:

In each file of the project where the ADC is used, 
add these lines on top of the file:

```
#ifdef  IS_MVIO_ENABLED
#undef  IS_MVIO_ENABLED
#define IS_MVIO_ENABLED() ((FUSE.SYSCFG1 & FUSE_MVSYSCFG_gm) == MVSYSCFG_DUAL_gc)
#endif
```

SpenceKonde, 2023-12-05:
This issue has been fixed in github.

commit 2184d923 (2024-01-04): Multiple corrections

Fix #497 for EB and DD (EB ofc doesn't even have MVIO!)

Fix spurrious warning that cfouldbe generated while using EEPROM.h correctly. Fix an incomplete constraint in a snippet of asm.

Updated boards.txt.
closed 2023-12-05 by SpenceKonde