SpenceKonde/DxCore #508: EVENT AND COMPARATOR ISSUES WITH EA and EB parts

LaurentR59, 2024-01-05:
Hi

It look like using EVENT and/or COMPARATOR with latest Ex CPU parts should not provide good result. at all ( unable to COMPILE)

Stangly no issue with TINY2 and MEGATINYCORE with such usages. (close to similar as EA EB parts on many point) 

So suggestly bug file detected for these new parts on event and comparator files to macth with Ex parts

(latest files from DXCORE github page taken on 06012024 and currently used) ( don t care about 1.5.6 log indicator, Platformio uses only latest published version here not listed of latest files in use at compile process)

Processing CPU_SELECTED (framework: arduino; platform: atmelmegaavr; board: AVR64EA32)
------------------------------------------------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/atmelmegaavr/AVR64EA32.html
PLATFORM: Atmel megaAVR (1.8.0) > AVR64EA32
HARDWARE: AVR64EA32 20MHz, 6KB RAM, 64KB Flash
PACKAGES: 
 - framework-arduino-megaavr @ 1.8.8 
 - framework-arduino-megaavr-dxcore @ 1.5.6 
 - framework-arduino-megaavr-megacorex @ 1.1.2 
 - framework-arduino-megaavr-megatinycore @ 2.6.7 
 - tool-avrdude @ 1.70100.0 (7.1.0) 
 - tool-avrdude-megaavr @ 3.60300.220118 (6.3.0) 
 - toolchain-atmelavr @ 3.70300.220127 (7.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 19 compatible libraries
Scanning dependencies...
Dependency Graph
|-- AP_DCC_Library @ 1.2.1
Building in release mode
Compiling .pio\build\CPU_SELECTED\src\main.cpp.o
Compiling .pio\build\CPU_SELECTED\libc3b\Comparator\Comparator.cpp.o
Compiling .pio\build\CPU_SELECTED\libc3b\Comparator\ComparatorISR.cpp.o
Compiling .pio\build\CPU_SELECTED\lib74b\Event\Event.cpp.o
In file included from C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Comparator\src\Comparator.h:4:0,
                 from C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Comparator\src\Comparator.cpp:2:
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\cores\dxcore/Arduino.h:637:1: error: expected initializer before '}' token
 } // extern "C"
 ^
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Comparator\src\Comparator.cpp:73:36: error: redefinition of 'AnalogComparator Comparator0'
     AnalogComparator    Comparator0(1, AC1, PORTD.PIN2CTRL, PORTD.PIN3CTRL, PORTD.PIN4CTRL, PORTD.PIN6CTRL, PORTC.PIN3CTRL,/* EA-series has no MVIO */            PORTD.PIN5CTRL, PORTD.PIN0CTRL, PORTD.PIN7CTRL, PORTC.PIN2CTRL);
                                    ^
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Comparator\src\Comparator.cpp:69:25: note: 'AnalogComparator Comparator0' previously declared here
       AnalogComparator  Comparator0(0, AC0, PORTD.PIN2CTRL, AC_NULL_REG,    AC_NULL_REG,    PORTD.PIN6CTRL, PORTC.PIN3CTRL,/* EA-series has no MVIO */            PORTD.PIN3CTRL, PORTD.PIN0CTRL, PORTD.PIN7CTRL, PORTC.PIN2CTRL);
                         ^~~~~~~~~~~
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Comparator\src\Comparator.cpp: In member function 'void AnalogComparator::init()':
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Comparator\src\Comparator.cpp:476:47: error: 'in4' is not a member of 'comparator::in_p'
       } else if (input_p == comparator::in_p::in4) {
                                               ^~~
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Comparator\src\Comparator.cpp:476:47: note: suggested alternative: 'in3'
       } else if (input_p == comparator::in_p::in4) {
                                               ^~~
                                               in3
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Comparator\src\Comparator.cpp: In member function 'void AnalogComparator::stop(bool)':
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Comparator\src\Comparator.cpp:599:49: error: 'in4' is not a member of 'comparator::in_p'
         } else if (input_p == comparator::in_p::in4) {
                                                 ^~~
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Comparator\src\Comparator.cpp:599:49: note: suggested alternative: 'in3'
         } else if (input_p == comparator::in_p::in4) {
                                                 ^~~
                                                 in3
In file included from C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Event\src\Event.h:4:0,
                 from C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Event\src\Event.cpp:2:
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\cores\dxcore/Arduino.h:637:1: error: expected initializer before '}' token       
 } // extern "C"
 ^
In file included from C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Comparator\src/Comparator.h:4:0,
                 from C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Comparator\src\ComparatorISR.cpp:1:
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\cores\dxcore/Arduino.h:637:1: error: expected initializer before '}' token       
[... 59 lines cut ...]
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Event\src\Event_parts.h:55:7: note:   'tcb2_capt'
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Event\src\Event.cpp: In static member function 'static event::user::user_t Event::user_from_peripheral(USART_t&)':
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Event\src\Event.cpp:1328:27: error: 'usart0_irda' is not a member of 'event::user'
       return event::user::usart0_irda;
                           ^~~~~~~~~~~
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Event\src\Event.cpp:1332:27: error: 'usart1_irda' is not a member of 'event::user'
       return event::user::usart1_irda;
                           ^~~~~~~~~~~
C:\Users\lroek\.platformio\packages\framework-arduino-megaavr-dxcore\libraries\Event\src\Event.cpp:1337:27: error: 'usart2_irda' is not a member of 'event::user'
       return event::user::usart2_irda;
                           ^~~~~~~~~~~
*** [.pio\build\CPU_SELECTED\libc3b\Comparator\Comparator.cpp.o] Error 1
*** [.pio\build\CPU_SELECTED\libc3b\Comparator\ComparatorISR.cpp.o] Error 1
*** [.pio\build\CPU_SELECTED\src\main.cpp.o] Error 1
*** [.pio\build\CPU_SELECTED\lib74b\Event\Event.cpp.o] Error 1
============================================================= [FAILED] Took 1.69 seconds =============================================================

Environment    Status    Duration
-------------  --------  ------------
CPU_SELECTED   FAILED    00:00:01.687

Any idea bout possible fixe on such case?

Regards
Laurent.

SpenceKonde, 2024-01-06:
WTF?!? I thought I fixed this AND HAD THE TESTS AUTOMATICALLY CHECK FOR IT!

pcfreak1201, 2024-01-06:
Hmm, that looks a little strange:
```
C:\Users\lroek.platformio\packages\framework-arduino-megaavr-dxcore\cores\dxcore/Arduino.h:637:1: error: expected initializer before '}' token
} // extern "C"
```

... and it is the first error in your list ...

SpenceKonde, 2024-01-06:
And arduino.h doesn't have that line there in the current version - you're on an old version encountering a bug I already fixed.

LaurentR59, 2024-01-07:
Hi

I apologize for my mistake.
After a full refresh of all files in PlatformIO, this action have  fixed so now all compile process work like a charm!

Thanks anyway.

Closed issue.

SpenceKonde, 2024-01-09:
Great!

closed 2024-01-09 by SpenceKonde