SpenceKonde/DxCore #513: Wrong timer for millis() on 14-pin DD part

iboguslavsky, 2024-01-14:
I found myself troubleshooting non-working millis() and delay() on AVR16DD14. The counter stays at 0 and is not advancing. 

millis() is set to use MILLIS_USE_TIMERB2, which this part doesn’t appear to have (it only has 2 x TIMER B as per datasheet). While the ioavr64dd14.h  does include it, reading TCB2 registers, like CTRLA, produces 0x0, and writing to it doesn’t seem to affect that. This leads me to believe it truly doesn’t exist on this part. 

My next step is to use TCB0/1 for millis(), but I’m struggling with finding where MILLIS_USE_TIMERx is defined. Setting a millistimer build option in platformio.ini doesn’t seem to affect the  TCB2 selection.

SpenceKonde, 2024-01-19:
Yeah I'm not surprised it doesn't work, platformio doesn't emulate the wacky boards.txt parser completely I don't think.

In platform.txt we have
```
build.optiondefines=-DF_CPU={build.f_cpu} -DCLOCK_SOURCE={build.clocksource} -DTWI_{build.wiremode} -DMILLIS_USE_TIMER{build.millistimer} {build.attachmode} {build.flmapopts}
```
This was done to make build.extra_flags less of a godawful copy-pasted error prone mess. 
But the rule is that we use TCB2, or the highest numbered TCB if there isn't a TCB2

Over in boards.txt, we have (with stuff in the middle cut out)

```
avrdd.build.highestcb=B2
```
Overridden by chip selection...
```
avrdd.menu.chip.avr64dd20.build.mcu=avr64dd20
avrdd.menu.chip.avr64dd20.upload.maximum_size=65536
avrdd.menu.chip.avr64dd20.upload.maximum_data_size=8192
avrdd.menu.chip.avr64dd20.build.highestcb=B1
avrdd.menu.chip.avr64dd20.build.variant=20pin-ddseries
avrdd.menu.chip.avr64dd20.build.defaultswaps=-DMUX_DEFAULT_USART1=2
```
But here's where we actually use it:
```
#----------------------------------------#
# millis/micros timing source            #
#________________________________________#
avrdd.menu.millis.tcbhighest=Default (recommended)
avrdd.menu.millis.disabled=Disabled (delay() only)
avrdd.menu.millis.tcb0=TCB0
avrdd.menu.millis.tcb1=TCB1 (default for 14/20 pins)
avrdd.menu.millis.tcb2=TCB2 (default for 28/32 pins)
avrdd.menu.millis.tca0=TCA0
avrdd.menu.millis.tcbhighest.build.millistimer={build.highestcb}
avrdd.menu.millis.tcb0.build.millistimer=B0
avrdd.menu.millis.tcb1.build.millistimer=B1
avrdd.menu.millis.tcb2.build.millistimer=B2
avrdd.menu.millis.tca0.build.millistimer=A0
avrdd.menu.millis.disabled.build.millistimer=NONE
```

I don't know nothin about platformio though.

SpenceKonde, 2024-07-29:
Closed. It has been 6 months, nobody has chimed in offering to help me out with this and I sure don't know anything about third party IDEs.

closed 2024-07-29 by SpenceKonde

ontje, 2026-09-19:
Just for the reference, here is how to make it work in Platformio IDE (as per https://github.com/platformio/platform-atmelmegaavr/issues/65)

You need to set board_hardware.millistimer to your selected timer as well, MILLIS_USE_TIMER flag is not sufficient.

Either do so in your

platformio.ini
```
[env]
platform = atmelmegaavr
framework = arduino

[env:AVR16DD14]
board = AVR16DD14
board_hardware.millistimer = B1
```

or in the boards/AVR16DD14.json you need to create anyway
```
{
  "build": {
    ...
    "variant": "14pin-ddseries"
  },
  "hardware": {
    "oscillator": "internal",
    "millistimer": "B1"
  },
  ...
}
```