SpenceKonde/DxCore #520: Board Manager SSH Certificate Expired

MiWeiss, 2024-02-18:
Following the instructions [here](https://github.com/SpenceKonde/DxCore/blob/master/Installation.md#manual-installation), I am facing an error after step 1 as the network connection cannot be safely established.

[image]

Manually checking, it indeed seems that the currently deployed certificate has expired.

MiWeiss, 2024-02-20:
Note: As a short-term workaround, I prepared [https://github.com/SpenceKonde/DxCore/files/14349040/index-copy.json](https://github.com/SpenceKonde/DxCore/files/14349040/index-copy.json). This should not be used for long though, as it won't be updated

andrewwootton, 2024-02-22:
I'm getting an "Error downloading" from the Arduino IDE with the additional boards manager set with http://drazzy.com/package_drazzy.com_index.json
Suspect its the same issue

closed 2024-07-28 by SpenceKonde