SpenceKonde/DxCore #525: program avr128da28 chip using arduino uno as a programmer

hoka83, 2024-04-23:
hello does jtag2updi support avr128da28 to be programmed by arduino uno? , i want to program avr128da28 chip using arduino uno as a programmer, can anyone provide me steps to do that

pcfreak1201, 2024-04-25:
Hi! Spence doesn't support jtag2updi any more. You can try to use avrdude directly. There should be support for the avr12da28 since version 7.2.
You should give SerialUPDI a try: https://github.com/SpenceKonde/DxCore/blob/master/README.md#updi-programming-hardware

hoka83, 2024-04-25:
> Hi! Spence doesn't support jtag2updi any more. You can try to use avrdude directly. There should be support for the avr12da28 since version 7.2. You should give SerialUPDI a try: https://github.com/SpenceKonde/DxCore/blob/master/README.md#updi-programming-hardware

i watched this video on youtube which Program an AVR DA MCU Using an Arduino Nano but in my case i have AVR128DA28 chip and arduino uno , i connected the pins as the same he did and it gave me error while burning the boatloader

i am just connecting arduino uno as a programmer for AVR128DA chip using updi and it gives me this error:

avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE
avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE
avrdude: jtagmkII_read_byte(): bad response to read memory command: RSP_ILLEGAL_MCU_STATE
avr_read(): error reading address 0x0000
read operation not supported for memory "signature"
avrdude: error reading signature data for part "AVR128DA28", rc=-2
avrdude: error reading signature data, rc=-2
avrdude: jtagmkII_program_disable(): bad response to leave progmode command: RSP_ILLEGAL_MCU_STATE
Error while burning bootloader. 
[image]
[image]

[image]

nomakewan, 2024-07-29:
Looks like you have voltage connected to 3.3V instead of 5V even though the logic level outputs on your Uno are 5V? Any particular reason you did that?

We also can't diagnose any potential connection issues to your AVR128DB28 because your photo does not show up that chip, there are no labels on where the wires are going into the green PCB, and you have not provided a schematic of that board.