SpenceKonde/DxCore #526: error when burning the bootloader for avr128DA28 and arduino uno as updi

hoka83, 2024-04-24:
[image]
[image]

hoka83, 2024-04-24:
why it gives me this error? 

avrdude: jtagmkII_initialize(): Cannot locate "flash" and "boot" memories in description
avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE
avrdude: jtagmkII_program_enable(): bad response to enter progmode command: RSP_ILLEGAL_MCU_STATE
avrdude: jtagmkII_read_byte(): bad response to read memory command: RSP_ILLEGAL_MCU_STATE
avr_read(): error reading address 0x0000
read operation not supported for memory "signature"
avrdude: error reading signature data for part "AVR128DA28", rc=-2
avrdude: error reading signature data, rc=-2
avrdude: jtagmkII_program_disable(): bad response to leave progmode command: RSP_ILLEGAL_MCU_STATE
Error while burning bootloader.

mikrocoder, 2024-05-17:
Hi,

Why are you photographing your monitor?
Can't you take screenshots?

About your problem.
The selected com port is wrong.
The Arduino Uno is not an AVR128DA28.
Do you already have the bootloader on it?

SpenceKonde, 2024-07-28:
They have no communication whatsoever between the arduino uno acting as UPDI device and the UPDI device. Wiring incorrect.

nomakewan, 2024-07-29:
Agreed. Suggest closing this as a duplicate of #525 honestly.

SpenceKonde, 2024-09-27:
Though, it should be noted that there were also buggy versions of jtag2updi

closed 2024-09-27 by SpenceKonde