SpenceKonde/DxCore #528: Is there a -r option hidden in the DxCore settings?

mikrocoder, 2024-05-17:
Hi,

I am currently testing the avr-gcc 8.5.0 toolchain. I always have the problem that the compilation stops because of a mrelax error. Somewhere the compiler option -r is set according to the error message. But with -v and -Wl,-v I don't see an option -r set anywhere.

Now I ask you specifically. Are you aware that there might be an option -r hidden somewhere in your DxCore settings that I don't see?

Thanks.

SpenceKonde, 2024-07-28:
Please post the text of the error you get when using my core with a toolchain other than the known working one which we ship with, and I will comment on it,

The specifics of the error message *do actually contain meaning* to those versed in the requisite dark and forbidden knowledge.

If you're asking whether -mrelax is being used, yes, of course it is, the bloat without it sucked.

SpenceKonde, 2024-08-15:
No response from OP. 

Yes, -mrelax is used because it saves 10-20% in binary size. You are the first person to speak ill of it.

closed 2024-08-15 by SpenceKonde

mikrocoder, 2024-08-16:
I didn't say anything bad about it. What makes you think that? I had a problem with the compile option. Nothing more. But it's done anyway.

By the way, about “no response”. [Controller Images](https://github.com/SpenceKonde/DxCore/discussions/519)
Maybe you should also answer if you are already calling for support. No response since February.