SpenceKonde/DxCore #535: What is GCSE optimisation?

technoblogy, 2024-06-16:
The README says:

> We now have a tools submenu that allows you to select either -Os or -O3 with or without GCSE.

What is GCSE? The GCC compiler pages don't mention it, and Google doesn't seem to know.

nomakewan, 2024-06-23:
https://gcc.gnu.org/news/gcse.html

SpenceKonde, 2024-07-29:
I'm trying to comment on here without saying something that would be controversial, so all the ways this could have been a funny comment, sorry, couldn't come up with a way that I wasn't concerned would offend someone. 

But yeah, toggling GCSE is when you do when you have no ideas on how to save space, don't need that much space, and haven't recently tried GCSE toggle.

In my tests, about 50% of the time it saved flash, and the rest of the time it cost flash. 

Can you tell me exactly where that line is or submit a PR though? We no longer offer an option to select the level of optimization. You can still toggle GCSE, but you can't change the overall optimization level, because the dirty trick I used to prevent up to fifteen unused copies of the serial interrupt handlers kicking around breaks when speed optimized compilation is used, because it will attempt (utterly unsuccessfully) to undo all the things I did with great care in hand optimized asm and make it bloated and slower like it was before. Except because it doesn't do that successfully, the compilation fails. 

As I am unaware of any way around this problem, and due to the magnitude (and frankly, the amount of effort it took alone would make me fall on my sword before removing it. It was a great deal of effort, and - now that it's in - does save like 500 bytes or something for each serial port after the first.  

But it cost the optimization options. If the O2 and O3 optimization options were useful on AVR that would be a big deal. Thankfully the optimizer is such shit that we're really losing any functionality :-/

closed 2024-07-29 by SpenceKonde

technoblogy, 2024-07-29:
Thanks! I would have been happy with something like:

"GCSE stands for Global Common Subexpression Elimination, and sometimes saves flash, but sometimes costs flash."

> Can you tell me exactly where that line is or submit a PR though?

It was in 1.5.6 but is no longer in 1.5.11.

P.S. I was sorry not to get one of your funny comments!

SpenceKonde, 2024-07-29:
Oh well that's a relief! I'm glad I haven't been claiming to support O2 and
O3 all this time.

-- 
____________
Spence Konde
Azzy’S Electronics

New products! Check them out at tindie.com/stores/DrAzzy
GitHub: github.com/SpenceKonde
ATTinyCore <https://github.com/SpenceKonde/ATTinyCore>: Arduino support for
all pre-2016 tinyAVR with >2k flash!
megaTinyCore <https://github.com/SpenceKonde/megaTinyCore>: Arduino support
for all post-2016 tinyAVR parts!
DxCore <https://github.com/SpenceKonde/DxCore>: Arduino support for the AVR
Dx-series parts, the latest and greatest from Microchip!
Contact: ***@***.***