SpenceKonde/DxCore #536: AVR128DA48 and AVR128DB48 Curiosity Nano boards don't work on 1.5.8 or later

technoblogy, 2024-06-16:
I have an application (uLisp) that used to work on the AVR128DA48 Curiosity Nano or AVR128DB48 Curiosity Nano with DxCore versions up to and including 1.5.6. However, it doesn't work on 1.5.8, or the latest, 1.5.11.

It uses the default settings on the **Tools** menu, apart from:

**Write to flash from app: "Above 96K (AVR128 only)"**

I use **PROGMEM**, but not **PROGMEM_MAPPED**.

I use **Serial**, and cater for the differences on these boards with, on the AVR128DA48 Curiosity Nano:

    #define Serial Serial1

or, on the AVR128DB48 Curiosity Nano:

    #define Serial Serial3

I noticed that the DxCore versions it stopped working on are the ones with the new **How to set FLMAP** option, but I've tried every option on that menu and none of them solve the issue.

I'm using Arduino IDE 1.8.19 on MacOS 10.13.

Any suggestions?

Thanks, David

SpenceKonde, 2024-07-29:
We already provide a facility to support arbitrary renumbering of the serial ports by having a variant file define that, or a tools submenu can redefine it (by adding a define to the command line invoked to run the compiler). 

However, that does inevitably run into a conflict if someone else tries to do the same thing a different way.

See line 1090-1100 of Arduino.h.

technoblogy, 2024-07-29:
Thanks for your reply. My point is that I had an application that used to work up to 1.5.6 and now it doesn't work, and it's not obvious how I should fix it.

> See line 1090-1100 of Arduino.h.

Those lines are identical in the `Arduino.h` of core 1.5.6 that does work, and 1.5.11 that doesn't work, so I don't think that can be the explanation.

Any other suggestions?

technoblogy, 2024-10-01:
I hope this is a clearer explanation of the problem:

This test works on the AVR128DA48 Curiosity Nano board with DxCore 1.5.6:
````text
#define Serial Serial1
  
void setup () {
  Serial.begin(9600);
}

void loop () {
  Serial.println("Hello");
  delay(1000);
}
````
However, DxCore 1.5.11 breaks it, and I can't figure out how to get it working again.

I'm using the following **Board** options:

**Board:" AVR DA-series (no bootloader)"**
**Chip: "AVR128DA48"**

everything else at defaults.

I'm hoping for a solution that's easy to explain to other users, rather than "revert to 1.5.6".

Thanks, David