SpenceKonde/DxCore #541: PWM off TCD on 14-pin DD series

fijam, 2024-08-04:
DxCore 1.5.11
AVR32DD14

This works: 

```
void setup() {
 PORTMUX.TCAROUTEA = PORTMUX_TCA0_PORTD_gc;
}

void loop() {
  analogWrite(PIN_PD5, 128);
}
```

This doesn't work:

```
void setup() {
 PORTMUX.TCDROUTEA = PORTMUX_TCD0_ALT4_gc;
}

void loop() {
  analogWrite(PIN_PD5, 128);
}
```

WeirdConstructor, 2024-08-17:
Wow, thanks a lot! This helped a lot, thanks to this PWM works on my AVR32DD14 now.

fijam, 2025-02-28:
This bug is now six months old. Is there anything I can do to help find it?

SpenceKonde, 2025-06-06:
nevermind, misread issue. previous statement holds.

SpenceKonde, 2025-09-24:
Any prior findings are no longer valid; PWM code was refactored drastically, and this should fix any outstanding issues with PWM.

closed 2025-09-24 by SpenceKonde