SpenceKonde/DxCore #543: AVR32EA64 uploads once only

Taragorm, 2024-08-15:
A couple of issues. Firstly, I had to update Avrdude to a later version to get support for the AVR32EA64 - the provided one in DxCore doesn't work.

That done, I compiled, and managed to get a download, though I had to remove  { program.verify } from the command tail as that seemed to break things.

So now it downloads (atmelice_udpi):

`Uploading 'TENS' to 'AVR EA-series (no bootloader)' using 'Atmel-ICE'
Uploader started for board AVR EA-series (no bootloader)
Uploader will use programmer name: atmelice_updi
C:\Users\Simon\AppData\Local\arduino15\packages\DxCore\tools\avrdude\6.3.0-arduino17or18\bin\avrdude "-CC:\Users\Simon\AppData\Local\arduino15\packages\DxCore\hardware\megaavr\1.5.11/avrdude.conf" -v -pavr64ea32 -catmelice_updi -Pusb "-Ufuse5:w:0b11001101:m" "-Ufuse6:w:0x04:m" "-Ufuse7:w:0x00:m" "-Ufuse8:w:0x00:m" "-Uflash:w:C:\T\VMBuilds\TENS\DxCore_avrea\Release/TENS.ino.hex:i"
avrdude: Version 7.3
         Copyright the AVRDUDE authors;
         see https://github.com/avrdudes/avrdude/blob/main/AUTHORS
         System wide configuration file is C:\Users\Simon\AppData\Local\arduino15\packages\DxCore\hardware\megaavr\1.5.11\avrdude.conf
         Using port            : usb
         Using programmer      : atmelice_updi
         AVR Part              : AVR64EA32
         Programming modes     : UPDI, SPM
         Programmer Type       : JTAGICE3_UPDI
         Description           : Atmel-ICE (ARM/AVR) in UPDI mode
         ICE HW version        : 0
         ICE FW version        : 1.42 (rel. 161)
         Serial number         : J41800100289
         Vtarget               : 3.31 V
         PDI/UPDI clk          : 500 kHz
avrdude: partial Family_ID returned: "AVR "
avrdude: silicon revision: 2.1
avrdude: AVR device initialized and ready to accept instructions
avrdude: device signature = 0x1e961f (probably avr64ea32)
avrdude: Note: flash memory has been specified, an erase cycle will be performed.
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: processing -U fuse5:w:0b11001101:m
avrdude: reading input file 0b11001101 for fuse5/syscfg0
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse5/syscfg0 ...
avrdude: 1 byte of fuse5/syscfg0 written
avrdude: verifying fuse5/syscfg0 memory against 0b1100 1101                                                                                
avrdude: 1 byte of fuse5/syscfg0 verified
avrdude: processing -U fuse6:w:0x04:m
avrdude: reading input file 0x04 for fuse6/syscfg1
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse6/syscfg1 ...
avrdude: 1 byte of fuse6/syscfg1 written
avrdude: verifying fuse6/syscfg1 memory against 0x04
avrdude: 1 byte of fuse6/syscfg1 verified
avrdude: processing -U fuse7:w:0x00:m
avrdude: reading input file 0x00 for fuse7/codesize
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse7/codesize ...
avrdude: 1 byte of fuse7/codesize written
avrdude: verifying fuse7/codesize memory against 0x00
avrdude: 1 byte of fuse7/codesize verified
avrdude: processing -U fuse8:w:0x00:m
avrdude: reading input file 0x00 for fuse8/bootsize
         with 1 byte in 1 section within [0, 0]
avrdude: writing 1 byte fuse8/bootsize ...
avrdude: 1 byte of fuse8/bootsize written
avrdude: verifying fuse8/bootsize memory against 0x00
avrdude: 1 byte of fuse8/bootsize verified
avrdude: processing -U flash:w:C:\T\VMBuilds\TENS\DxCore_avrea\Release/TENS.ino.hex:i
avrdude: reading input file C:\T\VMBuilds\TENS\DxCore_avrea\Release/TENS.ino.hex for flash
         with 4998 bytes in 1 section within [0, 0x1385]
         using 40 pages and 122 pad bytes
avrdude: writing 4998 bytes flash ...
Writing | ################################################## | 100% 0.59s
avrdude: 4998 bytes of flash written
avrdude: verifying flash memory against C:\T\VMBuilds\TENS\DxCore_avrea\Release/TENS.ino.hex
Reading | ################################################## | 100% 0.18s
avrdude: 4998 bytes of flash verified
avrdude: bad response to leave progmode command: 0xa0
avrdude done.  Thank you.
	The upload process has finished.
`

The issue seems to be: 

```
1 byte of fuse5/syscfg0 written
avrdude: verifying fuse5/syscfg0 memory against 0b1100 1101                                                                                
avrdude: 1 byte of fuse5/syscfg0 verified

```

The settings:

[image]

It looks to me like fuse5 is getting set wrong, in that bit#4 is UPDI enable, and that's getting reset, so you can't program the device again, at least not without HV programming (which I don't have support for)

Bit #2 is also suspicious, as that's "not used" in the datasheet.

I swapped the IC to verify this, fortunately I have a few spare.

Also: Anyone know how to do a HV reset on one of these parts? :)

Taragorm, 2024-08-16:
I *think* the issue comes from here: [boards.txt#1609]

`avrea.bootloader.SYSCFG0=0b1100{bootloader.resetpinbits}0{bootloader.eesavebit}`

should be
`avrea.bootloader.SYSCFG0=0b110{bootloader.resetpinbits}00{bootloader.eesavebit}`

If time permits, I'll risk another cpu later today

Taragorm, 2024-08-16:
I can confirm it works with the change above

SpenceKonde, 2025-06-06:
Yes we are aware of the issue with serialupdi that we use, and fixing it is the top priority once I can compile sketches for Dx-series again with the dev version of the core (went a long time doing work on the core without ever having compiletested, and need to go back and fix everything I broke)

SpenceKonde, 2025-08-10:
I figured out the cause of this problem by chance looking at the boards.txt - copy-paste error shifted the bits written to the syscfg0 fuse, which could in turn result in the UPDI pin being disabled when performing a normal upload. As I have never been able to upload to these, I hadn't realized the problem existed.

commit 30742a74 (2025-08-10): Fix #543
closed 2025-08-10 by SpenceKonde