SpenceKonde/DxCore #544: EA Timers

Taragorm, 2024-08-16:
A couple of timer related issues (not sure if I should open two issues here, or not)

1] We don't seem to have an option for using the RTC for millis(). Unclear if this is a deliberate omission, as we obviously have an RTC, and other parts have used this.

2] ioavr64ea32.h [and likely others] don't define TCA1 or TCB3. This is Microchip supplied, I guess, but maybe in need of a patch. I downloaded the latest, but it still seems to be broken. What's the usual way to handle this? Without the #defines you can't use the timers as a millis() source.

Taragorm, 2024-08-17:
:(
Even if you patch the header so TCA1 and TCB3 are mapped as per the datasheet, it doesn't seem to work. Almost as if the headers are right and the timers either don't work, or don't exist.

Bit of a problem as I need those timers.

I'm just going to park this unless anyone requests further testing, and redesign around an ATMega4809 part, which fortunately, I can do.

SpenceKonde, 2024-09-08:
```
#define TCA0                  (*(TCA_t *) 0x0A00) /* 16-bit Timer/Counter Type A */
#define TCB0                  (*(TCB_t *) 0x0B00) /* 16-bit Timer Type B */
#define TCB1                  (*(TCB_t *) 0x0B10) /* 16-bit Timer Type B */
#define TCB2                  (*(TCB_t *) 0x0B20) /* 16-bit Timer Type B */
#define SYSCFG             (*(SYSCFG_t *) 0x0F00) /* System Configuration Registers */
#define NVMCTRL           (*(NVMCTRL_t *) 0x1000) /* Non-volatile Memory Controller */
#define LOCK                 (*(LOCK_t *) 0x1040) /* Lockbits */
#define FUSE                 (*(FUSE_t *) 0x1050) /* Fuses */
#define USERROW           (*(USERROW_t *) 0x1080) /* User Row */
#define SIGROW             (*(SIGROW_t *) 0x1100) /* Signature row */
```

vs 
```
#define TWI0                  (*(TWI_t *) 0x0900) /* Two-Wire Interface */
#define SPI0                  (*(SPI_t *) 0x0940) /* Serial Peripheral Interface */
#define TCA0                  (*(TCA_t *) 0x0A00) /* 16-bit Timer/Counter Type A */
#define TCA1                  (*(TCA_t *) 0x0A40) /* 16-bit Timer/Counter Type A */
#define TCB0                  (*(TCB_t *) 0x0B00) /* 16-bit Timer Type B */
#define TCB1                  (*(TCB_t *) 0x0B10) /* 16-bit Timer Type B */
#define TCB2                  (*(TCB_t *) 0x0B20) /* 16-bit Timer Type B */
#define TCB3                  (*(TCB_t *) 0x0B30) /* 16-bit Timer Type B */
````

OOOFFF!!! did they cut the TCA1 and TCBs? If so, that was omne of the most appea;omg things about them.

CreativeRobotics, 2025-02-27:
Does anyoone know what the state of play is with these devices seeing as they are now being sold to distributors. I Just designed something around the AVR64EA32 device based on the datasheet and errata, but now it doesn't recognise TCA1 as being defined, but the AVR64EA48 does have the TCA1 defined.

The datasheet describes all devices as having TCA1 and there is no errata that mentions a missing TCA ... is this a hardware or software problem?

CreativeRobotics, 2025-02-28:
> :( Even if you patch the header so TCA1 and TCB3 are mapped as per the datasheet, it doesn't seem to work. Almost as if the headers are right and the timers either don't work, or don't exist.
>
> Bit [...]

Soooo, I tried compiling for the 64EA48 part which DOES have the timer defined, then used Microchip studio to flash the code to my 64EA32 part figuring that there isn't going to be any difference apart from the extra pins.

I used the code below to take over both timers and configure them for PWM output and it works, PWM output from both timers - this is confirmation that these parts DO have the TCA1 timer so this must be an issue with incorrect headers. I don't know where to look on Microchips site for the latest headers so I can't check if they have fixed it.

`
void configurePWM(){
  pinMode(PIN_PD0, OUTPUT);
  pinMode(PIN_PA4, OUTPUT);

  PORTMUX.TCAROUTEA = 0x23; //default TCA0 port D0,1,2 TCA1 port A4,5,6
  takeOverTCA0();
  takeOverTCA1();
  //configure both timers for PWM output on WO0
  TCA0.SINGLE.CTRLB   = (TCA_SINGLE_CMP0EN_bm | TCA_SINGLE_WGMODE_SINGLESLOPE_gc); // Single slope PWM mode, PWM on WO0
  TCA0.SINGLE.PER     = 0xFFF;               // Count all the way up to 0xFFF (4095)
  TCA0.SINGLE.CMP1    = 0x01;                    // 
  TCA0.SINGLE.CTRLA   = TCA_SINGLE_ENABLE_bm; // Enable the timer with no prescaler //(presc << 1) | TCA_SINGLE_ENABLE_bm;

  TCA1.SINGLE.CTRLB   = (TCA_SINGLE_CMP0EN_bm | TCA_SINGLE_WGMODE_SINGLESLOPE_gc); // Single slope PWM mode, PWM on WO0
  TCA1.SINGLE.PER     = 0xFFF;               // Count all the way up to 0xFFF (4095)
  TCA1.SINGLE.CMP1    = 0x02;                    // 
  TCA1.SINGLE.CTRLA   = TCA_SINGLE_ENABLE_bm; // Enable the timer with no prescaler //(presc << 1) | TCA_SINGLE_ENABLE_bm;
}

//buffered PWM duty cycle writing
void setTCA0_0_Duty(uint16_t duty) {
  //TCA0.SINGLE.CMP1 = duty;
  TCA0.SINGLE.CMP0BUF = duty;
}
void setTCA0_1_Duty(uint16_t duty) {
  //TCA0.SINGLE.CMP1 = duty;
  TCA0.SINGLE.CMP1BUF = duty;
}

void setTCA1_0_Duty(uint16_t duty) {
  //TCA1.SINGLE.CMP1 = duty;
  TCA1.SINGLE.CMP0BUF = duty;
}
void setTCA1_1_Duty(uint16_t duty) {
  //TCA1.SINGLE.CMP1 = duty;
  TCA1.SINGLE.CMP1BUF = duty;
}`

SpenceKonde, 2025-05-25:
I will look into this, there is likely a bug with pin detection. I need to solve the programming issue before anything  though. See, I can't hardware test anything on Ex until I can fix the programming problem. 

But I think I got the last of the fires on mTC extinguished (warmup)...

Some bugs will be fixed by default when new toolchain version is released, because microchip had some garbage hiding in their I/O headers which they recently noticed and cleaned off.

SpenceKonde, 2025-06-06:
It's either pin detection issue, or we may need to update to the latest ATPACKs, which needsto be done anyway to pick up allthe new parts and several fixes to Ex and DU series io headers.