SpenceKonde/DxCore #546: 64DA32/64DD and PWM

TheSWilde, 2024-08-30:
Hey there Guys

Im struggel with this 2 Days now.  Am I stupid or blind? 
Want to have 4 PWM Outputs (with LEDs on each) on Port A, D,.. whatever.
All i get ist 1 working PWM output, the other 3 LEDs just flickering a bit.

Looks like the last "analogWrite" stops the other 3 from spit out a beautiful PWM-Wave. But why? TCA should have 6(?) Outputs?
If i put e.g. analogWrite(PinG, i) on the end, the green LED will work, all others just flicker.

If I remove PORTMUX.TCA... I will get 2 blinking LED (PA0 and 1) and 2 Working PWM (2 and 3).

If I move each LED a pin higher (PA0 -> PA2, PA1->PA3. etc.) I get 3 working PWM (PA2,3 and 4) and one strange PWM on PA5, the LED will never complete turn off and is "out off sync".

Also tested on a 64DD32, same thing. 

So, where is my error? what am i doing wrong? 

```
#define PinW  PIN_PA0
#define PinR  PIN_PA1
#define PinG  PIN_PA2
#define PinB  PIN_PA3

void setup() { 
  PORTMUX.TCAROUTEA = PORTMUX_TCA0_PORTA_gc;
  pinMode(PinW, OUTPUT); 
  pinMode(PinR, OUTPUT);
  pinMode(PinG, OUTPUT);
  pinMode(PinB, OUTPUT);
}

void loop() {
  for (int i=0;i<=255;i++){ 
  analogWrite(PinW, i);
  analogWrite(PinR, i);
  analogWrite(PinG, i);
  analogWrite(PinB, i);
  delay(10);
 }
}
```

SpenceKonde, 2024-09-04:
We by default set the portmux to point to whichever port appears to be the most useful for PWM/least useful for other things. Since PORTA is super-crowded all the time, we put it on a different port, The specific port is listed in the part specific documentation pages. 

PA2 and PA3, when TCA0 isn't pointed at PORTA will have PWM through some of the TCBs. (this is a waste of a TCB and gives mediocre PWM, but w/e, you do get PWM there. And with the default PORTD portmux, there will be TCD0 PWM on PA4-7. Combined with the fact that PA0/PA1 are usually used for console if not for a crystal, and especially on the DA-series and DB's until rev B where only the first TCD portmux option works..... well.... you can see why I decided to put the default somewhere else (the support for detecting when you've changed PORTMUX came much later)

There may be something up with this, but I'm almost certain the timer assignments that are occurring are not what you expect. Have you read the part specific docs? I know I wrote a bunch about how timers are allocated there.

TheSWilde, 2024-09-08:
If i understand your well written docs right, TCA0 should point to PortD by CoreDefault.
Which means to me, TCA0 should be able to output PWM to Pins PD0-5, right? But it doesnt.
If i programm 
analogWrite(PD0, 100);
followed by
analogWrite(PD1, 100);
PWM on PD0 will be stopped and PD1 works like aspected.

analogWrite(PD2, 100);
kills PWM on PD1 and PD2 will now output PWM.

Maybe i missunterstood something you write in your docs, as i'm not a native English-speaker.

SpenceKonde, 2024-09-08:
Won't have access tpo test until 14th

SpenceKonde, 2025-09-24:
PWM pin checking completely reimplemented.