SpenceKonde/DxCore #550: AVR64DD32 via UPDI, I can burn the bootloader but uploading a sketch fails?

fxframes, 2024-09-25:
The title pretty much sums it up.
It's UPDI so only 3 pins if burning the bootloader works why can't I upload a sketch right after?
The sketch is as simple as it gets and compiles ok. I even switched down to 57600 baud, same error. 
Screenshots attached.
Thanks.

[image]
[image]

WestfW, 2024-09-25:
Did you write the fuses as well?

fxframes, 2024-09-26:
Thanks for the quick reply.

This is my first time using UPDI.
I went with the defaults from the Tools setup except that I'm using an external 16M oscillator. I didn't realise I was burning fuses but you're right. Please find the image for the Tools setup below and after that both text logs. Thanks for any insights.

[image]

`/Users/**/Library/Arduino15/packages/megaTinyCore/tools/python3/3.7.2-post1/python3 -u /Users/**/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.11/tools/prog.py -t uart -u /dev/cu.usbserial-310 -b 230400 -d avr64dd32 --fuses 0:0b00000000 1:0x00 2:0x00 5:0b11011000 6:0b00001100 7:0x00 8:0x01 -f /Users/**/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.11/bootloaders/hex/optiboot_64dd_ser0_extr.hex -a write -v 
SerialUPDI
UPDI programming for Arduino using a serial adapter
Based on pymcuprog, with significant modifications
By Quentin Bolsee and Spence Konde
Version 1.3.0.3 - Jul 2023
Using serial port /dev/cu.usbserial-310 at 230400 baud.
Target: avr64dd32
Set fuses: ['0:0b00000000', '1:0x00', '2:0x00', '5:0b11011000', '6:0b00001100', '7:0x00', '8:0x01']
Action: write
File: /Users/**/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.11/bootloaders/hex/optiboot_64dd_ser0_extr.hex
pymcuprog.programmer - INFO - Setting up programming session for 'avr64dd32'
pymcuprog.deviceinfo.deviceinfo - INFO - Looking for device avr64dd32
pymcuprog.serialupdi.physical - INFO - Opening port '/dev/cu.usbserial-310' at '115200' baud
pymcuprog.serialupdi.link - INFO - STCS 08 to 0x03
pymcuprog.serialupdi.link - INFO - STCS 06 to 0x02
pymcuprog.serialupdi.link - INFO - LDCS from 0x00
pymcuprog.serialupdi.link - INFO - UPDI init OK
pymcuprog.serialupdi.link - INFO - STCS 06 to 0x02
pymcuprog.serialupdi.link - INFO - Setting UPDI clock to 8 MHz
pymcuprog.serialupdi.link - INFO - STCS 02 to 0x09
pymcuprog.serialupdi.physical - INFO - Switching to '230400' baud
pymcuprog.serialupdi.application - INFO - SIB: 'AVR     P:2D:1-3M2 (A3.KV00K.0)'
pymcuprog.serialupdi.application - INFO - Device family ID: 'AVR'
pymcuprog.serialupdi.application - INFO - NVM interface: 'P:2'
pymcuprog.serialupdi.application - INFO - Debug interface: 'D:1'
pymcuprog.serialupdi.application - INFO - PDI oscillator: '3M2'
pymcuprog.serialupdi.application - INFO - Extra info: '(A3.KV00K.0)'
pymcuprog.serialupdi.application - INFO - Using 24-bit UPDI
pymcuprog.serialupdi.link - INFO - STCS 08 to 0x03
pymcuprog.serialupdi.link - INFO - STCS 06 to 0x02
pymcuprog.serialupdi.link - INFO - LDCS from 0x00
pymcuprog.serialupdi.link - INFO - UPDI init OK
pymcuprog.serialupdi.link - INFO - LDCS from 0x00
pymcuprog.serialupdi.application - INFO - PDI revision = 0x03
pymcuprog.serialupdi.link - INFO - LDCS from 0x0B
pymcuprog.serialupdi.link - INFO - LDCS from 0x0B
pymcuprog.serialupdi.application - INFO - Entering NVM programming mode
pymcuprog.serialupdi.link - INFO - LDCS from 0x07
pymcuprog.serialupdi.application - INFO - Apply reset
pymcuprog.serialupdi.link - INFO - STCS 59 to 0x08
pymcuprog.serialupdi.application - INFO - Release reset
pymcuprog.serialupdi.link - INFO - STCS 00 to 0x08
pymcuprog.serialupdi.link - INFO - LDCS from 0x0B
pymcuprog.serialupdi.link - INFO - LDCS from 0x0B
pymcuprog.nvm - INFO - No specific initializer for this provider
Pinging device...
pymcuprog.programmer - INFO - Reading device ID...
pymcuprog.serialupdi.application - INFO - SIB: 'AVR     P:2D:1-3M2 (A3.KV00K.0)'
pymcuprog.serialupdi.application - INFO - Device family ID: 'AVR'
pymcuprog.serialupdi.application - INFO - NVM interface: 'P:2'
pymcuprog.serialupdi.application - INFO - Debug interface: 'D:1'
pymcuprog.serialupdi.application - INFO - PDI oscillator: '3M2'
pymcuprog.serialupdi.application - INFO - Extra info: '(A3.KV00K.0)'
pymcuprog.serialupdi.application - INFO - Using 24-bit UPDI
pymcuprog.serialupdi.link - INFO - STCS 08 to 0x03
pymcuprog.serialupdi.link - INFO - STCS 06 to 0x02
pymcuprog.serialupdi.link - INFO - LDCS from 0x00
pymcuprog.serialupdi.link - INFO - UPDI init OK
pymcuprog.serialupdi.link - INFO - LDCS from 0x00
pymcuprog.serialupdi.application - INFO - PDI revision = 0x03
pymcuprog.serialupdi.link - INFO - LDCS from 0x0B
pymcuprog.serialupdi.application - INFO - Device ID = '1E961A' rev 'A3'
[... 125 lines cut ...]
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "/Users/**/Library/Arduino15/packages/DxCore/hardware/megaavr/1.5.11/avrdude.conf"
         User configuration file is "/Users/**/.avrduderc"
         User configuration file does not exist or is not a regular file, skipping

         Using Port                    : /dev/cu.usbserial-310
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x20
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x30
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x20

avrdude done.  Thank you.

An error occurred while uploading the sketch
`

nomakewan, 2024-09-27:
You don't write the sketch using UPDI if you're trying to flash a sketch using optiboot. You write the sketch using serial. UPDI is for burning the bootloader (and burning the program, if you are not using optiboot). Once you've burned the bootloader, connect to TX0/RX0 and your sketch upload should work fine.

fxframes, 2024-10-24:
> UPDI is for burning the bootloader (and burning the program, if you are not using optiboot).

Thanks!

closed 2024-10-24 by fxframes